zephyrproject-rtos/zephyr #17314: doc: add tutorial for using mbed TLS

dbkinder, 2019-07-03:
As reported by a developer:

I have seen that other IoT OS use tinydtls for authentication over CoAP and IEEE 802.15.4. On the other hand,  the Zephyr RTOS I have seen (in documentation 1.9) that uses mbed TLS but I confused on how to set it in IoT end devices. **Specifically, I was wondering if there is any tutorial to start (e.g in reel boards). ** Also, I would like to know if the DTLS is based on pre-shared keys or in any elliptic curve cryptography (ECC) algorithm?

Thank you for your time

--
Nikos Karamolegkos
R & D Engineer at ICS-FORTH
Telecommunications and Networks Lab (TNL)

d3zd3z, 2019-12-03:
There is an example in `samples/net/google_iot_mqtt`, although this just uses the socket interface.

I fear that an example would quickly grow out of date, and that security issues in it would not be addressed, and I think this is better served by making sure there is a good working example in the tree.

As far as the question. DTLS in mbed TLS can be configured, and can use pre-shared keys based on ECJPAKE, or it can use various EC curves, or even RSA.

carlescufi, 2021-03-11:
As per @d3zd3z, close this as won't do.

closed 2021-03-11 by carlescufi