zephyrproject-rtos/zephyr #17323: failing network tests with code coverage enabled

andrewboie, 2019-07-03:
```
$ sanitycheck --coverage -p qemu_x86 -T tests/net/ 
JOBS: 12
Cleaning output directory /home/apboie/projects/zephyr1/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
48 tests selected, 20148 tests discarded due to filters
total complete:    0/  48   0%  failed:    0

qemu_x86                  tests/net/all/net.build                            FAILED: build_error
	see: sanity-out/qemu_x86/tests/net/all/net.build/build.log

total complete:    1/  48   2%  failed:    1

qemu_x86                  tests/net/iface/net.iface.no_userspace_allowed     FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/iface/net.iface.no_userspace_allowed/handler.log

total complete:    2/  48   4%  failed:    2

qemu_x86                  tests/net/iface/net.iface.with_userspace           FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/iface/net.iface.with_userspace/handler.log

total complete:   20/  48  41%  failed:    3

qemu_x86                  tests/net/mld/net.mld                              FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/mld/net.mld/handler.log

total complete:   21/  48  43%  failed:    4

qemu_x86                  tests/net/socket/getaddrinfo/net.socket            FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/socket/getaddrinfo/net.socket/handler.log

total complete:   26/  48  54%  failed:    5

qemu_x86                  tests/net/socket/getnameinfo/net.socket            FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/socket/getnameinfo/net.socket/handler.log

total complete:   27/  48  56%  failed:    6

qemu_x86                  tests/net/mgmt/net.management                      FAILED: failed
	see: sanity-out/qemu_x86/tests/net/mgmt/net.management/handler.log

total complete:   30/  48  62%  failed:    7

qemu_x86                  tests/net/socket/net_mgmt/net.socket.mgmt          FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/socket/net_mgmt/net.socket.mgmt/handler.log

total complete:   34/  48  70%  failed:    8

qemu_x86                  tests/net/socket/register/net.socket.register.tls  FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/socket/register/net.socket.register.tls/handler.log

total complete:   36/  48  75%  failed:    9

qemu_x86                  tests/net/socket/select/net.socket                 FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/socket/select/net.socket/handler.log

total complete:   37/  48  77%  failed:   10

qemu_x86                  tests/net/socket/tcp/net.socket.tcp                FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/socket/tcp/net.socket.tcp/handler.log

total complete:   38/  48  79%  failed:   11

qemu_x86                  tests/net/udp/net.udp                              FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/udp/net.udp/handler.log

total complete:   39/  48  81%  failed:   12

qemu_x86                  tests/net/utils/net.util                           FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/utils/net.util/handler.log

total complete:   48/  48  100%  failed:   13
```

Enabling CONFIG_COVERAGE exposes a bunch of test failures with networking tests under tests/net/. We need these be able to build and run successfully to assess code coverage of the network stack.

At least one (tests/net/all/net.build) is related to #15831
Others may be stack size related, CONFIG_COVERAGE requires considerably more stack space to function correctly.
One is baffling, appears to be crashing in early boot with tests/net/iface/ under both scenarios.

Reproduce with:

```
sanitycheck --coverage -p qemu_x86 -T tests/net/ 
```

aescolar, 2019-07-04:
CC @jukkar

jukkar, 2019-07-04:
The  #17344 fixes three tests from the above list. Increasing stack sizes helped for tests in tests/net/udp, tests/net/mgmt and tests/net/mld directories. Increasing stack size did not help for the other tests, or I might have increased wrong values. I attached gdb to the failing tests and they fail before ztest main() is called.

andrewboie, 2019-08-03:
Many of these are fixed by #17989

andrewboie, 2019-08-03:
With #17989 applied, we are down to three failing tests:

```
$ sanitycheck -p qemu_x86 --coverage -T tests/net/
JOBS: 6
Cleaning output directory /home/apboie/projects/zephyr3/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
52 tests selected, 20372 tests discarded due to filters
total complete:   13/  52  25%  failed:    0

qemu_x86                  tests/net/lib/dns_addremove/net.dns                FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/lib/dns_addremove/net.dns/handler.log

total complete:   14/  52  26%  failed:    1

qemu_x86                  tests/net/lib/dns_addremove/net.dns.no_ipv4        FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/lib/dns_addremove/net.dns.no_ipv4/handler.log

total complete:   42/  52  80%  failed:    2

qemu_x86                  tests/net/socket/udp/net.socket.udp                FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/socket/udp/net.socket.udp/handler.log

total complete:   52/  52  100%  failed:    3
Generating coverage files...
HTML report generated: /home/apboie/projects/zephyr3/zephyr/sanity-out/coverage/index.html
49 of 52 tests passed with 0 warnings in 594 seconds
```

camicarballo, 2019-08-08:
Down to 1 failing test with #17786 applied

```
$ ./scripts/sanitycheck --coverage -p qemu_x86 -T tests/net
JOBS: 4
Cleaning output directory /home/camelot/zephyrproject/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
52 tests selected, 21149 tests discarded due to filters
total complete:   43/  52  82%  failed:    0

qemu_x86                  tests/net/socket/udp/net.socket.udp                FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/socket/udp/net.socket.udp/handler.log

total complete:   52/  52  100%  failed:    1
Generating coverage files...
HTML report generated: /home/camelot/zephyrproject/zephyr/sanity-out/coverage/index.html
51 of 52 tests passed with 0 warnings in 580 seconds
```

jukkar, 2020-05-06:
Tried this and the dns tests were still failing. After increasing the main stack size, the failures disappeared.

PR #25013 (2020-05-06): tests: net: dns: Increase stack size for coverage support
Increase stack size so that coverage information can be generated.

Fixes #17323
commit f95c349c (2020-05-06): tests: net: dns: Increase stack size for coverage support

Increase stack size so that coverage information can be generated.

Fixes #17323
closed 2020-05-06 by jukkar