zephyrproject-rtos/zephyr #17324: failing bluetooth tests with code coverage enabled

andrewboie, 2019-07-03:
```$ sanitycheck --coverage -p qemu_x86 -T tests/bluetooth/ 
JOBS: 12
Cleaning output directory /home/apboie/projects/zephyr1/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
16 tests selected, 9708 tests discarded due to filters
total complete:    4/  16  25%  failed:    0

qemu_x86                  tests/bluetooth/mesh_shell/bluetooth.mesh_shell    FAILED: build_error
	see: sanity-out/qemu_x86/tests/bluetooth/mesh_shell/bluetooth.mesh_shell/build.log

total complete:    5/  16  31%  failed:    1

qemu_x86                  tests/bluetooth/hci_prop_evt/bluetooth.hci_prop_evt FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/bluetooth/hci_prop_evt/bluetooth.hci_prop_evt/handler.log

total complete:    9/  16  56%  failed:    2

qemu_x86                  tests/bluetooth/shell/bluetooth.shell              FAILED: build_error
	see: sanity-out/qemu_x86/tests/bluetooth/shell/bluetooth.shell/build.log

total complete:   10/  16  62%  failed:    3

qemu_x86                  tests/bluetooth/shell/bluetooth.shell_br           FAILED: build_error
	see: sanity-out/qemu_x86/tests/bluetooth/shell/bluetooth.shell_br/build.log

total complete:   16/  16  100%  failed:    4
```

3 of these appear to be due to  #15831

Reproduce with:

```
sanitycheck --coverage -p qemu_x86 -T tests/bluetooth/ 
```

andrewboie, 2019-07-17:
This is in much better shape now, with current master I'm only seeing one test fail:

```
qemu_x86                  tests/bluetooth/hci_prop_evt/bluetooth.hci_prop_evt FAILED: timeout
	see: sanity-out/qemu_x86/tests/bluetooth/hci_prop_evt/bluetooth.hci_prop_evt/handler.log
```

ioannisg, 2019-07-30:
Still, this needs some fixing. And an assignee @andrewboie

andrewboie, 2019-08-13:
setting this to low, this is an alternate configuration for code coverage metrics, the test does not fail when run under its original configuration.

nashif, 2020-03-11:
Closing, can't reproduce.

closed 2020-03-11 by nashif