zephyrproject-rtos/zephyr #17336: insufficient code coverage for include/sys/math_extras_impl.h

andrewboie, 2019-07-04:
The inline functions declared here are not sufficiently tested by test cases in tests/kernel/ or tests/lib/.

I can find no test case at all which proves that these work. Need to add a test under tests/lib/ which exercises every function, with and without builtins, and prove that they work.

rubidijum, 2023-04-22:
These are actually covered in `/tests/unit_tests`. 
Here is the lcov report:
[image]

I tested with both builtin and portable versions, although there is a small bug in `tests/unit/math_extras/portable.c`:

```
#define VNAME(N) test_portable_##N
#include "tests.inc"

ZTEST(math_extras_portable, test_portable_u32_add) {
	run_portable_u32_add();
}

...
```

Where i had to change `#define VNAME(N) test_portable_##N` to `#define VNAME(N) run_portable_##N`

nashif, 2024-02-08:
well, I would say we have many things that do not have good code coverage, we do not need to have an issue about each.,

closed 2024-02-08 by nashif