zephyrproject-rtos/zephyr #17337: arm mpu sub region alignment

wentongwu, 2019-07-04:
for systems that have mpu sub region support and also mpu don't require power of two alignment, users want to use sub region size granularity to do alignment instead of power of two to avoid wasting much memory, for example code size is about 512K + 1Byte, if power of two alignment enabled, another almost 512K will be wasted. So for that case, we should disable CONFIG_MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT, and if configure CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE to mpu sub region size, the size of thread stack will be very large because of thread stack align with CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE, so my suggestion is to add one more Kconfig(like CONFIG_ARM_MPU_SUB_REGION_MIN_ALIGN_AND_SIZE) to configure mpu sub region size.

```
/* Set alignment to CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE
 * to make linker section alignment comply with MPU granularity.
 */
#if defined(CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE)
_region_min_align = CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE;
#else
/* If building without MPU support, use default 4-byte alignment. */
_region_min_align = 4;
#endif

#if defined(CONFIG_MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT)
#define MPU_ALIGN(region_size) \
    . = ALIGN(_region_min_align); \
    . = ALIGN( 1 << LOG2CEIL(region_size))
#else
#define MPU_ALIGN(region_size) \
    . = ALIGN(_region_min_align)
#endif

```

wentongwu, 2019-07-04:
@ioannisg  could you please share some idea for this problem? thanks

ioannisg, 2019-07-04:
I am trying to understand this, @wentongwu thanks for looking at options for improvement.

>for systems that have mpu sub region support and also mpu don't require power of two alignment, 
 
Which systems are those?
- ARMv7-M MPU supports the option for sub-region disabling, but requires power-of-two-sized (and aligned-with-size) MPU regions.
- ARMv8-M MPU does not require power-of-two alignment of MPU regions, but does not support sub-regions.

wentongwu, 2019-07-04:
> I am trying to understand this, @wentongwu thanks for looking at options for improvement.
>
>
> Which systems are those?
>
>     * ARMv7-M MPU supports the option for sub-region disabling, but requires [...]

the system is ARMv7-M, and what I mean users want to use sub region size granularity(that's power of two alignment) to do alignment to avoid large memory hole(for above example)which can be seen in the map file, sorry I have wrong description, I mean we should disable CONFIG_MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT in linker script, but when configure mpu, the size will be power of two alignment. For above example, if 64K granularity to do alignment in linker script, only another almost 64K will be wasted. And for the mpu configuration, 512k -> region1 and 64k -> region2 have same mpu attribute. @ioannisg

ioannisg, 2019-07-04:
Alright, thanks for explaining. Now, I also confused; what do you mean by sub-region?
Are you referring to the ARMv7-M MPU sub-regions? If so, these ones are fixed to 1/8 of the MPU region - how the sub-region size could be used in a Kconfig?

wentongwu, 2019-07-04:
Ok, it is, but for example somehow we can now the size of .text section, 512k+4B, with current code and CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE enabled, the size of .text section in RAM (suppose only RAM) will be 1M.
```
#define MPU_ALIGN(region_size) \
    . = ALIGN(_region_min_align); \
    . = ALIGN( 1 << LOG2CEIL(region_size))
```
But if with below code, the size of .text section will be 512k+64k(MPU_SUB_REGION_SIZE), that saved memory.

```
#define MPU_ALIGN(region_size) \
    . = ALIGN(_region_min_align); \
    . = ALIGN( MPU_SUB_REGION_SIZE)
```

For MPU configuration, 512k will be in MPU region1 and 64k will be in MPU region2 as one sub region of region2 which disable all the other sub regions. And the disabled sub regions of region2 can also be part of regionN(N>2) and will be functional in regionN.
If SUB_GEGION_SIZE confused, can we use USER_ALIGN_SIZE or some other name? 
The basic idea is to use mpu sub region and mpu overlap characteristics to avoid wasting too much RAM in some cases.

ioannisg, 2019-07-04:
Alright, now I understand what you want to do. So, this means that for a data region (e.g. a user thread stack, app mem partition, etc) you might need to waste 2 MPU regions at run-time, to make this work. We cannot afford this, sorry. We need to use as few regions as possible.

ioannisg, 2019-07-04:
But now I started thinking more of it: could we use sub-region feature to shrink the MPU regions?
Example:
Assume a desired thread stack size of 1200 bytes.
We still align this with 2048, but the user thread stack can be 2048 * (5 / 8) = 1280 bytes.

wentongwu, 2019-07-04:
> Alright, now I understand what you want to do. So, this means that for a data region (e.g. a user thread stack, app mem partition, etc) you might need to waste 2 MPU regions at run-time, to make this work. We cannot afford this, sorry. We need to use as few regions as possible.

Yes, you are right. But we just use this in the linker script to avoid memory waste in linking time. For stack and memory partition, we still use CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE.

ioannisg, 2019-07-04:
Alright. So we are saving space in code, not SRAM, right?

wentongwu, 2019-07-08:
> But now I started thinking more of it: could we use sub-region feature to shrink the MPU regions?
> Example:
> Assume a desired thread stack size of 1200 bytes.
> We still align this with 2048, but the user thread stack can be 2048 * (5 / 8) = 1280 bytes.
> So we save space, but still need to respect alignment...
Sorry for the late response. 
Yes, we still need to respect alignment, but I didn't get the idea about shrink the MPU regions. AFAIK, we can just enable/disable sub-region and all the sub-regions have the same attributes as the whole MPU region.

wentongwu, 2019-07-08:
@ioannisg I created a example for this alignment proposal, hope that can show my idea more. Thanks.

PR #17390 (2019-09-20): linker: add custom align size to reduce alignment memory wasting 
```
1) When enable CONFIG_CUSTOM_SECTION_ALIGN, it need less alignment memory
for image rom region. But it will consume one more MPU region. For example somehow
we can know the size of .text and .rodata section, 512k+4B, with current code and
CONFIG_MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT enabled, the size of .text
and .rodata section in RAM (suppose only RAM) will be 1M.
But if with CONFIG_CUSTOM_SECTION_ALIGN(for example 64k), the size of .text and 
.rodata section will be 512k+64k(MPU_SUB_REGION_SIZE), that saved memory. 
For MPU [... cut]
commit 09af98ab (2019-09-20): scripts: gen_relocate_app.py: add custom align size for relocation

add custom align size for code relocation to reduce alignment memory
wasting.

Fixes: #17337.
commit da31c817 (2019-09-20): linker: add custom align size to reduce alignment memory wasting

when enable CONFIG_CUSTOM_SECTION_ALIGN, it need less alignment
memory for image rom region. But that needs carefully configure
MPU region and sub-regions(ARMv7-M) to cover this feature.

Fixes: #17337.
PR #19308 (2019-09-25): linker: add custom align size to reduce alignment memory wasting
1) When enable CONFIG_CUSTOM_SECTION_ALIGN, it need less alignment memory
for image rom region. But it will consume one more MPU region. For example somehow
we can know the size of .text and .rodata section, 512k+4B, with current code and
CONFIG_MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT enabled, the size of .text
and .rodata section in RAM (suppose only RAM) will be 1M.
But if with CONFIG_CUSTOM_SECTION_ALIGN(for example 64k), the size of .text and 
.rodata section will be 512k+64k(MPU_SUB_REGION_SIZE), that saved memory. But that need [... cut]
closed 2019-09-17 by wentongwu