zephyrproject-rtos/zephyr #17338: kernel objects address check in elf_helper.py

wentongwu, 2019-07-04:
when code relocation enabled, there will be some other regions to hold kernel objects(like k_mutex, k_sem, etc), but in elf_helper.py, it will do the address check like below code, only in SRAM region, so kernel objects in other regions will be discarded. There is a method which is to add symbols in these regions to mark the start and end of the memory containing kernel objects, and then add more check here in elf_helper.py.

But I want to know if delete below check logic, is there any other side effect? Because, run time we will check if there is hash key in the hash table, even there is no below check, other intend and unintended non-kernel object address will also be dropped when these addresses being passed to kernel space.

```
497             _, user_ram_allowed = kobjects[ko.type_obj.name]
498             if (not user_ram_allowed and
499                     (addr < kram_start or addr >= kram_end) and
500                     (addr < krom_start or addr >= krom_end)):
501 
502                 self.debug_die(die,
503                                "object '%s' found in invalid location %s"
504                                % (name, hex(addr)))
505                 continue
```

wentongwu, 2019-07-04:
@andrewboie @ioannisg could you please take a look this question? Thanks

andrewboie, 2019-07-04:
this check is a legacy of CONFIG_APPLICATION_MEMORY.
You should be able to change this to check that the kernel object is outside the app_smem area.

PR #17360 (2019-07-08): scripts: elf_helper: correct kernel object address check range
kernel object should not located in app_smem area.

Fixes: #17338
commit 859ca420 (2019-07-08): scripts: elf_helper: correct kernel object address check range

kernel object should not located in app_smem area.

Fixes: #17338.
commit c89ee554 (2019-07-09): scripts: elf_helper: correct kernel object address check range

kernel object should not located in app_smem area.

Fixes: #17338.
commit 212938a6 (2019-07-09): scripts: elf_helper: correct kernel object address check range

kernel object should not located in app_smem area.

Fixes: #17338.
closed 2019-07-08 by andrewboie