zephyrproject-rtos/zephyr #17340: Bluetooth Mesh: Unable to receive messages when RPL is full.

xiaoliang314, 2019-07-04:
In Mesh communication, the system reported an error:
```
[01:17:32.160,000] <err> bt_mesh_transport: RPL is full!
[01:17:32.160,000] <wrn> bt_mesh_transport: Replay: src 0x0002 dst 0xc002 seq 0x0060f0
```
My Mesh node can't receive this message, In the Mesh protocol stack, messages are blocked when the relay list is full.
https://github.com/zephyrproject-rtos/zephyr/blob/9e57338efdb63a068c904b57eb16fe33f6b59cd3/subsys/bluetooth/host/mesh/transport.c#L900-L904
I think that whether relaying is a node's choice is not an obligation. It should not affect the core functionality of the node -- receiving messages.

xiaoliang314, 2019-07-04:
Is my understanding error!

jhedberg, 2019-07-04:
Yep, I guess you confused between "relay" and "replay" :) As per specification the implementation is obliged to discard any message destined for its elements if it cannot perform replay protection on it (e.g. if the RPL is full).

jhedberg, 2019-07-04:
One more note: the correct action is for you to increase the value of `CONFIG_BT_MESH_CRPL` to match the maximum number of unique unicast addresses your node may be receiving messages from.

xiaoliang314, 2019-07-04:
@jhedberg Thank you for your answer, I did confuse them. What I don't understand is, under what circumstances will the replay protection list be full? I used the light control model, and after switching multiple switches, some elements could not be controlled. These seem to be normal behaviors. Is the replay protection list full and related to the number of nodes in the network?

jhedberg, 2019-07-04:
@xiaoliang314 as I mentioned in my previous comment there is a build-time configuration option Called `CONFIG_BT_MESH_CRPL` using which you set the size of the RPL. It's a configurable value since it depends on the type of network and the number of nodes. So you should set it to a value which makes sense for your node. The default value for it is 10 (IIRC), i.e. if you don't set it explicitly for your application then your node will be able to receive messages from at most 10 other nodes (or more accurately from 10 other *elements* since a node can consist of multiple elements)

jhedberg, 2019-07-04:
>  What I don't understand is, under what circumstances will the replay protection list be full

It becomes full when a node has received messages from as many unique unicast addresses (i.e. other nodes/elements) as the size of the RPL is. After that the node can continue receiving messages from these nodes, but it will not be able to receive messages from any new nodes.

xiaoliang314, 2019-07-04:
@jhedberg  thinks, This is a satisfactory answer.

closed 2019-07-04 by xiaoliang314