zephyrproject-rtos/zephyr #17353: Configuring with POSIX_API disables NET_SOCKETS_POSIX_NAMES

SteveOss, 2019-07-04:
**Describe the bug**

If POSIX support is enabled with CONFIG_POSIX_API=y then use of POSIX compliant
network interface names with CONFIG_NET_SOCKETS_POSIX_NAMES is disabled. This seems illogical to me.

**Expected behavior**
Should be able to set both CONFIG_POSIX_API and CONFIG_NET_SOCKETS_POSIX_NAMES for maximum POSIX compliance.

**Impact**
Breaks existing build

**Screenshots or console output**

warning: NET_SOCKETS_POSIX_NAMES (defined at subsys/net/lib/sockets/Kconfig:16) was assigned the
value 'y' but got the value 'n'. You can check symbol information (including dependencies) in the
'menuconfig' interface (see the Application Development Primer section of the manual), or in the
Kconfig reference at

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.10.1
 - Commit 1.14 LTS branch

**Additional context**
Am pretty sure this used to work on 1.14, looks like has been broken on the LTS branch
at some point.

aescolar, 2019-07-04:
CC: @pfalcon @jukkar

jukkar, 2019-07-04:
Yes, this is a known issue. Can you try #17346 if it helps?

SteveOss, 2019-07-04:
Will give it a try, but to me looks like a core configuration dependency issue. For NET_SOCKETS_POSIX_NAMES the dependency is:

!POSIX_API && NET_SOCKETS && NETWORKING

This just looks broken to me, to enable socket POSIX names, selecting POSIX APIs must be false ???
IMHO should be:

POSIX_API && NET_SOCKETS && NETWORKING or potentially just:
 
NET_SOCKETS && NETWORKING

pfalcon, 2019-07-04:
> This just looks broken to me, to enable socket POSIX names, selecting POSIX APIs must be false ???

No, you should just use POSIX APIs, period.

However, if you know what you're doing, and would like to munge with Zephyr's adhoc proprietary APIs, while making them to pretend to be POSIX-like, you may select a very special option NET_SOCKETS_POSIX_NAMES.

The current behavior works as intended. If it looks strange, then the strange thing is the use of NET_SOCKETS_POSIX_NAMES, not that it conflicts with POSIX_API.

SteveOss, 2019-07-05:
The intent is to be able to compile code using standard POSIX and standard BSD networks calls e.g.

pthread_mutex_t mutex;
int sock = socket (AF_INET, SOCK_STREAM, 0);
pthread_mutex_init (&mutex, NULL);

Currently the catch 22 is that I have to enable POSIX_API to use the pthread stuff and enable
NET_SOCKETS_POSIX_NAMES to use the standard BSD socket calls. However enabling POSIX_API
disables NET_SOCKETS_POSIX_NAMES. For anyone attempting to port an existing application
to Zephyr this is the behaviour they want, the current configuration support appears to support either
POSIX or standard BSD network APIs, IMHO it should be possible to support both or am I missing
something ?

SteveOss, 2019-07-05:
Oops have just read above log and see you guys already have the issue in hand.

pfalcon, 2019-07-05:
> The intent is to be able to compile code using standard POSIX and standard BSD networks

"Standard BSD networks" are part of "standard POSIX".

> Currently the catch 22

The catch 22 is that Zephyr has severely broken POSIX subsystem, though some people are fixing it (while others are opposing streamlined progress in that direction).

> Currently the catch 22 is that I have to enable POSIX_API to use the pthread stuff and enable
NET_SOCKETS_POSIX_NAMES to use the standard BSD socket calls. 

No, POSIX_API enables all POSIX APIs. If you had to enable POSIX_API and NET_SOCKETS_POSIX_NAMES in some previous Zephyr version, that's the bug, which is being fixed.

> For anyone attempting to port an existing application
to Zephyr this is the behaviour they want,

I'm one of the guys who work on porting existing apps to Zephyr, and speak from experience of doing that more than once (~3 porting projects by now).

> the current configuration support appears to support either
POSIX or standard BSD network APIs, IMHO it should be possible to support both or am I missing
something ?

Once again, POSIX API does include BSD Socket API. However, the current POSIX support in Zephyr is severely limited and broken, and needs fixing to work as expected by "people who work on porting existing applications". There's WIP in that direction, which unfortunately, gets too much bikeshedding and sidetracking. Reversing it is however not an option (from my humble opinion as a POSIX subsys maintainer in Zephyr). Instead, interested parties should back and vote up the pending patches (https://github.com/zephyrproject-rtos/zephyr/pulls?q=is%3Aopen+is%3Apr+label%3A%22area%3A+POSIX%22).

SteveOss, 2019-07-05:
Am totally in support of implementing the correct fixes so that Zephyr has a quality POSIX API. Can
always macro my way round my current issues. No offence intended.

closed 2019-08-30 by ioannisg

pfalcon, 2019-08-30:
Ok, this got closed automatically as a result of merging #18736. I wanted to post comments here before that, but got lost in errands with that PR.

So, since the earlier comments on this PR, there were a lot of discussion where POSIX subsystem goes, and how. To clarify it, https://github.com/zephyrproject-rtos/zephyr/issues/17706 was posted. And fortunately, we went forward with the rest of pending POSIX features for 2.0, instead of being lost in small-steps, as subject of this ticket.

To summarize, the current situation is:

1. There's an adhoc, BSD Sockets-**like** API in Zephyr, whose functions start with zsock_ (zsock_socket, zsock_recv, etc.). As a **workaround**, this API provides CONFIG_NET_SOCKETS_POSIX_NAMES option. Entire this API is a thing in itself. You can use it if you want to send results of a+b over socket. There's no guarantee that it will integrate with any other API (like, wider POSIX APIs), actually, it won't.

2. There's a POSIX subsystem in Zephyr, which offers gradually growing implementation of POSIX APIs, including BSD Socket API. It's enabled with CONFIG_POSIX_API, and intended to provide well-integrated set of de facto and de jure APIs used in the computing. That's forward-looking goal at least, we're not exactly there. But 2.0 is important milestone, and the PR which closes this, #18736, demonstrates that it's possible to build and run (small) applications having the 100% the same source code for both Linux and Zephyr.

P.2 is the intended way to use BSD Sockets in Zephyr. Parties with special needs may use p.1, but again, there's no guarantee that p.1 integrates well with any other API.

expploitt, 2021-02-23:
Hello, I have some questions about zephyr's configuration flags. I'm trying to use the socket BSD, so I used this config

CONFIG_NETWORKING=y
CONFIG_NET_SOCKETS=y
CONFIG_PTHREAD_IPC=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y

But in my compilation's output, i see implicit reference to usleep() and sleep().

I see that I have to use CONFIG_POSIX_API to implement that methods. But them the socket code don't compile. What does the reason is?

Thanks!

PR #18736 (2019-08-30): samples: net: sockets: Allow to build and test with POSIX subsys
With CONFIG_POSIX_API enabled, these samples now build under Zephyr
with exactly the same source as e.g. Linux (or in general, other POSIX
systems). However, building without CONFIG_POSIX_API (i.e. with
CONFIG_NET_SOCKETS_POSIX_NAMES aux option) is retained for now.

Add testcase definitions to build these samples with CONFIG_POSIX_API
in CI.

Fixes: #17353
commit 5eb974d8 (2019-08-30): samples: net: sockets: Allow to build and test with POSIX subsys

With CONFIG_POSIX_API enabled, these samples now build under Zephyr
with exactly the same source as e.g. Linux (or in general, other POSIX
systems). However, building without CONFIG_POSIX_API (i.e. with
CONFIG_NET_SOCKETS_POSIX_NAMES aux option) is retained for now.

Add testcase definitions to build these samples with CONFIG_POSIX_API
in CI.

Fixes: #17353