zephyrproject-rtos/zephyr #17363: SPI driver does not reset master mode fault on STM32

mrfuchs, 2019-07-05:
**Describe the bug**

The current implementation of the STM32 SPI driver does not reset MODF (master mode fault). This leads to the situation where, after starting a new transmission, the SPI chip select signal is active but the function never returns because the interrupts is never triggered.

https://github.com/zephyrproject-rtos/zephyr/blob/eb57607c7d761f076d9f4abd03b22e933a2898b2/drivers/spi/spi_ll_stm32.c#L52-L69

**To Reproduce**

Set up an SPI master with CS control. Build for an STM32 board/target.

```
static const spi_cs_control cs_ctrl = {
  .gpio_dev = device_get_binding(DT_ST_STM32_SPI_1_CS_GPIOS_CONTROLLER),
  .gpio_pin = DT_ST_STM32_SPI_1_CS_GPIOS_PIN,
  .delay = 0
};

static const spi_config spi_cfg = {
  .frequency = spi_clock_freq,
  .operation = SPI_OP_MODE_MASTER | SPI_TRANSFER_MSB | SPI_WORD_SET(8) | SPI_LINES_SINGLE,
  .cs        = &cs_ctrl
};
```

Continuously send data over SPI (in a thread/loop). Externally pull low the CS signal on your board. The SPI communication stops working.

**Expected behavior**

MODF does not lock up SPI communication. MODF, if being set by the controller, is reset by the driver.

**Additional context**

Fix:

```diff
diff --git a/drivers/spi/spi_ll_stm32.c b/drivers/spi/spi_ll_stm32.c
index 67ecb32336..ec8ec80c40 100644
--- a/drivers/spi/spi_ll_stm32.c
+++ b/drivers/spi/spi_ll_stm32.c
@@ -62,6 +62,11 @@ static int spi_stm32_get_err(SPI_TypeDef *spi)
 			LL_SPI_ClearFlag_OVR(spi);
 		}
 
+		/* MODF error must be explicitly cleared */
+		if (LL_SPI_IsActiveFlag_MODF(spi)) {
+			LL_SPI_ClearFlag_MODF(spi);
+		}
+
 		return -EIO;
 	}
```

erwango, 2019-07-05:
@mrfuchs, thanks for reporting, and since you propose the solution fixing the issue, feel free to push it as well.

yboutreux, 2019-07-05:
@mrfuchs feel free to tag me inside of a comment of your PR so I can follow the progress of the PR.
If you are not willing to push this fix please notice me and I'll do it, it will be useful to me for adding SPI to a STM32 board :)

mrfuchs, 2019-07-05:
@yboutreux Yes, please do so. Thanks!

arnop2, 2019-07-23:
@mrfuchs 
It seems that there are 2 different issues in your report.
The first is that the MODF flag is never cleared. Your patch should fixes it.
The second is that MODF error is generated...
I would like to be sure that the use case is valid and that MODF error IRQ is not spurious (meaning a bad configuration in the driver)... 
Please could you detail your setup?
- Are you in slave, master or multi master mode?
- do you use the SS pin? for which purpose?

mrfuchs, 2019-07-25:
> Please could you detail your setup?
>     * Are you in slave, master or multi master mode?
>     * do you use the SS pin? for which purpose?

SPI module is configured as master with `spi_config.cs` pointing to a `spi_cs_control` structure, as seen in my first comment above

There are two STM32 controllers on my board, which are connected via SPI. They are working in a single master, single slave configuration. The STM32F4 is acting as the SPI master and running Zephyr. Usually, these two controllers start up synchronously, but in the event when the slave controller resets itself (e.g. due to a fault condition), the master sometimes stopped SPI communication afterwards - probably because of spikes on the chip select line caused by the slave controller during its start-up sequence.

closed 2019-07-31 by galak