zephyrproject-rtos/zephyr #17366: Regression: sanitycheck coverage generation defaults will error out for POSIX arch targets

aescolar, 2019-07-05:
**Describe the bug**
After #17271 (8047a6f), running sanitycheck with the `-C`/`--coverage` option for POSIX arch based boards (e.g. `native_posix`) will result in a lot of errors instead of a coverage report due to `gcov` version incompatibilities.

**To Reproduce**
Steps to reproduce the behavior, for example:
sanitycheck -p native_posix -T tests/kernel -C

**Expected behavior**
Things working as before the offending commit for POSIX arch boards (`native_posix` and `nrf52_bsim`)

**Impact**
Confused users of POSIX arch boards due to their workflows broken.

**Environment:**
 - OS: Linux (e.g. Ubuntu 18.04)
 - Toolchain: SDK 0.10.1 + host gcc 7.3.0-27ubuntu1~18.04
 - Commit SHA or Version used: Any after 8047a6f

-------
Related to #17365

aescolar, 2019-07-05:
@andrewboie Will you take this?

andrewboie, 2019-07-08:
> @andrewboie Will you take this?

Why are there two bugs filed on this?
If #17359 is not an acceptable solution to this, no I will not take this.

aescolar, 2019-07-08:
@andrewboie : Well, #17359 does not fix the regression in itself, though it reduces the effect for new users.
Would it be a reasonable solution, that would make both qemu_x86 and native_posix users happy, to have the default  gcov tool in sanitycheck be selected based on the selected platform? (if type native => gcov, else, the current default)

PR #17359 (2019-07-09): doc: update coverage documentation
- The sanitycheck defaults have changed.

- Sanitycheck attempts to select the right gcov binary based on selected platforms

Fixes: #17366 
Fixes: #17365
closed 2019-07-09 by aescolar