zephyrproject-rtos/zephyr #17372: [Zephyr 1.14.0] Sanitiy check will not parse multiple CONF_FILE extra_confs correctly

gon1332, 2019-07-05:
**Describe the bug**
I have a sample.yaml file that has a test case where more than one prj.conf are needed:
```
...
tests:
  a_test_case:
    build_only: true
    extra_args: CONF_FILE="prj_a.conf boards/myboard.conf"
...
``` 

AFAIC, this fails because it parses the `extra_args` by delimiting on spaces, so it ends up with these args for the cmake:

* `-DCONF_FILE=prj_a.conf`
* `-Dboards/myboard.conf`

After fooling around, and knowing that cmake usually uses `;` as a delimiter I concluded in rewriting the above offending line from:
```
extra_args: CONF_FILE="prj_a.conf boards/myboard.conf"
```
to
```
extra_args: CONF_FILE="prj_a.conf;boards/myboard.conf"
```

This is OK. The tests run as expected. I haven't got the time to search and fix the python script so I leave it here, and we will see.

**To Reproduce**
Add a testcase like the above and run sanity_check on that. See the results.

**Expected behavior**
It should succeed.

**Impact**
Annoyance only, since I've found a quick workaround.

**Environment (please complete the following information):**
 - OS: GNU/Linux
 - Toolchain: Zephyr SDK
 - tags/v1.14.0

marc-hb, 2019-07-08:
> this fails

An error message wouldn't hurt.

> so it ends up with these args for the cmake:

You can see the complete cmake command in `sanity-out/Makefile`. How does it look?
Can you please look again at `sanity-out/Makefile` after doing this?
```
extra_args: CONF_FILE="prj_a.conf"
extra_args: CONF_FILE="boards/myboard.conf"
```
Finally, there is a (too large?) number of places/options for config files implemented by `cmake/kconfig.cmake`, search for `merge_config_files` there. Maybe there's some other way to implement what you want?

cc @ulfalizer

marc-hb, 2019-07-08:
BTW I think I've been using this in the past and it worked:
```
sanitycheck --extra-args=CMAKE_CFLAGS='-g3 -Dfu=bar'
```
This doesn't solve your problem, this is just some more background.

gon1332, 2019-07-08:
> An error message wouldn't hurt.

The first failing message:
```
~/D/C/ioannis-mcuboot ❯❯❯ $ZEPHYR_BASE/scripts/sanitycheck --testcase-root $MCUBOOT_DIR/boot/zephyr --tag build --build-only --platform ctsppmb01 --verbose --inline-logs --outdir /tmp/sanity
JOBS: 8
Cleaning output directory /tmp/sanity
Reading test case configuration files under /mplampla/ioannis-mcuboot/mcuboot/boot/zephyr...
Reading platform configuration files under /mplampla/ioannis-mcuboot/3rd-party/zephyr/boards...
Reading platform configuration files under /mplampla/ioannis-mcuboot/3rd-party/zephyr/scripts/sanity_chk/boards...
Building testcase defconfigs...
Filtering test cases...
2 tests selected, 330 tests discarded due to filters
1/2 ctsppmb01                 test_build_ota                                     FAILED: build_error
--------------------../../../../../tmp/sanity/ctsppmb01/test_build_ota/build.log--------------------
```

The build.log file contents. Focus on `File not found`:
```
~/D/C/ioannis-mcuboot ❯❯❯ cat /tmp/sanity/ctsppmb01/test_build_ota/build.log                                                                                                                                                              ✘ 1 
Zephyr version: 1.14.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4") 
-- Selected BOARD ctsppmb01
-- Found west: /home/gon1332/.local/bin/west (found suitable version "0.5.8", minimum required is "0.5.6")
-- Loading /mplampla/ioannis-mcuboot/3rd-party/zephyr/boards/arm/ctsppmb01/ctsppmb01.dts as base
-- Overlaying /mplampla/ioannis-mcuboot/3rd-party/zephyr/dts/common/common.dts
-- Overlaying /mplampla/ioannis-mcuboot/mcuboot/boot/zephyr/dts.overlay
CMake Error at /mplampla/ioannis-mcuboot/3rd-party/zephyr/cmake/kconfig.cmake:134 (message):
  File not found:
  /mplampla/ioannis-mcuboot/mcuboot/boot/zephyr/-Dboards/ctsppmb01.conf
Call Stack (most recent call first):
  /mplampla/ioannis-mcuboot/3rd-party/zephyr/cmake/app/boilerplate.cmake:399 (include)
  CMakeLists.txt:55 (include)

-- Configuring incomplete, errors occurred!
```

The `sanity-out` Makefile looks like this:
```
~/D/C/ioannis-mcuboot ❯❯❯ cat /tmp/sanity/Makefile

# Generated by /mplampla/ioannis-mcuboot/3rd-party/zephyr/scripts/sanitycheck which is expected
# to spawn qemu or testbinary, renode, etc. 

...

.PHONY: myboard/test_build_ota
myboard/test_build_ota:
	@echo sanity_test_building myboard/test_build_ota >&2
	cmake  \
		-G"Unix Makefiles"\
		-H/mplampla/ioannis-mcuboot/mcuboot/boot/zephyr/.\
		-B/tmp/sanity/myboard/test_build_ota\
		-DEXTRA_CFLAGS="-Werror   -Wno-deprecated-declarations"\
		-DEXTRA_AFLAGS=-Wa,--fatal-warnings\
		-DEXTRA_LDFLAGS="-Wl,--fatal-warnings"\
		-DCONF_FILE="prj_ota.conf -Dboards/myboard.conf" -DBOARD=myboard\
		>/tmp/sanity/myboard/test_build_ota/build.log 2>&1
	$(MAKE) -C /tmp/sanity/myboard/test_build_ota\
		VERBOSE=1 \
		>>/tmp/sanity/myboard/test_build_ota/build.log 2>&1
	@echo sanity_test_finished myboard/test_build_ota >&2

all: ... myboard/test_build_ota
```

So you can see the inserted `-D` prefix in the wrong place.

> BTW I think I've been using this in the past and it worked:
> 
> ```
> sanitycheck --extra-args=CMAKE_CFLAGS='-g3 -Dfu=bar'
> ```
> 
> This doesn't solve your problem, this is just some more background.

Yes, this is working, but it isn't useful in automated tests.

marc-hb, 2019-07-08:
Thanks for the additional info. I took the Python debugger for a quick ride and I think the parsing issue happens here:
```
class SanityConfigParser:

    def _cast_value(self, value, typestr):

        elif typestr.startswith("list") and isinstance(value, str):
            vs = v.split() # <<====
 
```

It seems sanitycheck doesn't trust YAML and re-implements its own parsing/"type system". So for instance it tells the YAML parser in `scripts/sanity_chk/sanitycheck-tc-schema.yaml` that `extra_args` has for YAML type `str` whereas the "real" type hidden from YAML is `list` as found in `testcase_valid_keys`. If sanitycheck trusted YAML lists I think one could probably just do:
```
extra_args:
    - 'CONF_FILE=-Dprf.conf -Dboard.conf'
    - 'CMAKE_CFLAGS=-Dfubar1=barfu1 -Dfubar2=barfu2'
```

carlescufi, 2019-08-25:
@marc-hb @nashif any news on this one?

dleach02, 2020-07-23:
Given that the "fix" was to use ';' can we close this bug?

marc-hb, 2020-07-23:
> Given that the "fix" was to use ';' can we close this bug?

I'm pretty sure the double quotes mean it's a workaround :-) The more serious fix is to somehow (dunno how) get rid the parsing code that incompletely duplicates the YAML parser. White space matters.

I'm afraid the main problem is why does a sanitycheck configuration parser bug impacts Zephyr release and quality metrics. I bet that's where the desire to close it comes from?

Rather than closing it and potentially hiding it from someone affected by it or maybe even from someone - who knows - willing to fix it, how about github label(s) that exclude irrelevant stuff like this from Zephyr quality metrics?

In this particular case it could be even simpler: just remove the "bug" label. Switching to plain and full YAML syntax can be considered as a "feature request" IMHO.

PS: don't get me wrong: if some other sanitycheck bug reports test failures as PASS then it should definitely NOT be excluded from Zephyr metrics!

nashif, 2020-07-25:
IMO this can be closed, using ";" is the right solution, moving everything to yaml syntax  is an optimization.

closed 2020-07-25 by nashif