zephyrproject-rtos/zephyr #17375: Add VREF, TEMPSENSOR, VBAT internal channels to the stm32 adc driver

StefJar, 2019-07-05:
The current stm32 adc driver is missing the internal channels. Would be great to have at least VREF avaible to improve the reference of the measurement.

cybertale, 2019-07-10:
Valuable suggestion, I'm currently busy on the dma driver of stm32 in Zephyr. Feel free to try to add it yourself or wait for me or some others to add this support later.

StefJar, 2019-09-23:
hey ho
@cybertale did do have any progress on that?

cybertale, 2019-09-28:
@StefJar Sorry, the DMA driver is still not finished. :disappointed_relieved:

StefJar, 2019-09-30:
I wrote a simple ADC1 DMA demo driver for the STM32F4 MCU. It works fine for my purposes. I put it on github: https://github.com/StefJar/zephyr_stm32_adc1_dma_driver
Maybe @cybertale it can be used as a blueprint for a async AD1 driver using DMA and internal channels ;)

cybertale, 2019-09-30:
@StefJar Thanks your code! The DMA driver is very close to be merged. I also wanted to add DMA support to the ADC driver, because in zephyr, interrupt is handled first by the OS then our thread, so it becomes a bit slower, thus measuring a sequence of ADC channels is not possible since we have to retrieve ADC result every end of conversion (irq thread in zephyr is slower as I mentioned but ADC conversion can be very fast). After the DMA driver is merged, the first thing I would do in zephyr would be adding DMA support for the ADC driver.
But I think I can help to add the support for internal channel since the DMA PR needs review now.

martinjaeger, 2020-01-06:
Is there any progress on this, as the DMA PR is merged now?

We are currently working on ADC + DMA for the STM32G4 and also need to read out the internal channels.

StefJar, 2020-01-07:
as far as I know the DMA driver got updated, but the ADC driver is not updated yet.

cybertale, 2020-01-09:
So sorry guys, I'm busying on my graduation project these several months, what's worse is that my private local server just got attacked several days ago by ransomware and I lost 4TB data, I'm busying on recovering my files now. Have to working on this later.

ikerperezdelpalomar, 2020-08-18:
Hello. I was wondering how are you interested in approach this task. Are you interested on being able to use macros like `LL_ADC_CHANNEL_TEMPSENSOR` and `LL_ADC_CHANNEL_VREFINT` to enable each of them? That would cause that if TEMPSENSOR is enabled VBAT can't be used and vice versa, other choice would be to actively switch between them, but that would involve more logic.

erwango, 2021-06-04:
@ikerperezdelpalomar Can you check if this issue could be closed, following mege of https://github.com/zephyrproject-rtos/zephyr/pull/34673 ?
Txs

StefJar, 2021-06-04:
this issue was partially fixed through https://github.com/zephyrproject-rtos/zephyr/pull/34673.
@erwango it works for me with that stage. Lets close it for now

closed 2021-06-04 by StefJar

ycsin, 2022-01-12:
@StefJar #34673 solve the TEMPSENSOR part of this issue, but are you aware of anything that can read the VBAT?

ycsin, 2022-01-12:
2a96cc03478a348f301649870fa1d5b88005f70b added support for internal voltage reference source but left out the VBAT, I modified the code to include VBAT:
```c
	if (__LL_ADC_CHANNEL_TO_DECIMAL_NB(ADC_CHANNEL_TEMPSENSOR) == channel_cfg->channel_id) {
		adc_stm32_set_common_path(dev, LL_ADC_PATH_INTERNAL_TEMPSENSOR);
	} else if (__LL_ADC_CHANNEL_TO_DECIMAL_NB(ADC_CHANNEL_VREFINT) == channel_cfg->channel_id) {
		adc_stm32_set_common_path(dev, LL_ADC_PATH_INTERNAL_VREFINT);
	} else if (__LL_ADC_CHANNEL_TO_DECIMAL_NB(ADC_CHANNEL_VBAT) == channel_cfg->channel_id) {
		adc_stm32_set_common_path(dev, LL_ADC_PATH_INTERNAL_VBAT);
	}
```

But the ADC value I got doesn't match the voltage on the VBAT.

erwango, 2022-01-12:
@ycsin Please open a new issue.