zephyrproject-rtos/zephyr #17379: Wrong hex file generated for MCUboot

Laczen, 2019-07-06:
**Describe the bug**
When building with `CONFIG_BOOTLOADER_MCUBOOT=y` a empty header of image header size is generated in zephyr.hex. This empty header is not required and results in a failure of
imgtool.py to generate a correct hex file (another header will be prepended to the image).

**To Reproduce**
Steps to reproduce the behavior:
1. west build -s zephyr/samples/hello_world/ -b nrf52_pca10040
2. zephyr.hex (image starts at C200):
```
:10C000000000000000000000000000000000000030
:10C010000000000000000000000000000000000020
:10C020000000000000000000000000000000000010
:10C030000000000000000000000000000000000000
:10C0400000000000000000000000000000000000F0
:10C0500000000000000000000000000000000000E0
:10C0600000000000000000000000000000000000D0
:10C0700000000000000000000000000000000000C0
:10C0800000000000000000000000000000000000B0
:10C0900000000000000000000000000000000000A0
:10C0A0000000000000000000000000000000000090
:10C0B0000000000000000000000000000000000080
:10C0C0000000000000000000000000000000000070
:10C0D0000000000000000000000000000000000060
:10C0E0000000000000000000000000000000000050
:10C0F0000000000000000000000000000000000040
:10C10000000000000000000000000000000000002F
:10C11000000000000000000000000000000000001F
:10C12000000000000000000000000000000000000F
:10C1300000000000000000000000000000000000FF
:10C1400000000000000000000000000000000000EF
:10C1500000000000000000000000000000000000DF
:10C1600000000000000000000000000000000000CF
:10C1700000000000000000000000000000000000BF
:10C1800000000000000000000000000000000000AF
:10C19000000000000000000000000000000000009F
:10C1A000000000000000000000000000000000008F
:10C1B000000000000000000000000000000000007F
:10C1C000000000000000000000000000000000006F
:10C1D000000000000000000000000000000000005F
:10C1E000000000000000000000000000000000004F
:10C1F000000000000000000000000000000000003F
:10C20000500500207DD4000075E800000DD300002B
:10C210000DD300000DD300000DD300000DD300009E
:10C220000DD300000DD300000DD3000069CF000036
:10C230000DD300000DD3000011CF000065E8000011
:10C240003DD400003DD400003DD400003DD40000AA
:10C250003DD400003DD400003DD400003DD400009A
:10C260003DD400003DD400003DD400003DD400008A
:10C270003DD400003DD400003DD400003DD400007A
:10C280003DD400003DD400003DD400003DD400006A
:10C290003DD400003DD400003DD400003DD400005A
:10C2A0003DD400003DD400003DD400003DD400004A
:10C2B0003DD400003DD400003DD400003DD400003A
:10C2C0003DD400003DD400003DD400003DD400002A
...
```
**Expected behavior**
The header that is added should not be there zephyr.hex:
```
:10C20000500500207DD4000075E800000DD300002B
:10C210000DD300000DD300000DD300000DD300009E
:10C220000DD300000DD300000DD3000069CF000036
:10C230000DD300000DD3000011CF000065E8000011
:10C240003DD400003DD400003DD400003DD40000AA
:10C250003DD400003DD400003DD400003DD400009A
:10C260003DD400003DD400003DD400003DD400008A
:10C270003DD400003DD400003DD400003DD400007A
:10C280003DD400003DD400003DD400003DD400006A
:10C290003DD400003DD400003DD400003DD400005A
:10C2A0003DD400003DD400003DD400003DD400004A
:10C2B0003DD400003DD400003DD400003DD400003A
:10C2C0003DD400003DD400003DD400003DD400002A
...
```

**Impact**
imgtool.py cannot generate images using the zephyr hex file, so at the moment only bin input is allowed. Other bootloaders might have the same problem with the generated hex file.

ioannisg, 2019-07-07:
Can you bisect to determine the commit to blame, in case this was not always wrong?

Laczen, 2019-07-07:
@ioannisg I can't because I don't understand what is causing the zeros to be prepended.

ioannisg, 2019-07-07:
@nvlsianpu could you investigate this please?

nvlsianpu, 2019-07-08:
Zeros at hex begin are fine - Build behaves like that since mcuboot was ported for zephyr. 
0x200 bytes are reserved for image header which is added by the imgtool.py.

Be aware of this change: https://github.com/JuulLabs-OSS/mcuboot/commit/44588eff41a3f90145394cd739ffd3e993087396#diff-3b31e4a6417a485b1075e37618453979 was done in imgtool.py some time ago.

In the mean time will check whether this work as expected.

ioannisg, 2019-07-08:
Thanks @nvlsianpu

nvlsianpu, 2019-07-08:
[TEXT_SECTION_OFFSET](https://github.com/zephyrproject-rtos/zephyr/blob/master/Kconfig.zephyr#L123) sets the offset - it is not available via menuconfig (I was not aware that it was hidden) but can be modified via adding to `west build` a configuration file option `-- -DCONF_FILE=*.conf`

Laczen, 2019-07-08:
Hi @nvlsianpu, I don't understand why the 0's are added, when a hex file is read by imgtool it will add an extra header (only for hex files). Would it not be simpler/easier for MCUboot config to set:
```
CONFIG_FLASH_LOAD_OFFSET=DT_FLASH_IMAGE0_OFFSET + 0x200
CONFIG_FLASH_LOAD_SIZE=DT_FLASH_IMAGE0_SIZE - 0x200
```
(Not exactly this code of course, using $(dt_hex_val, ...))

In that case it will not add anything to the generated hex file, and imgtool can do the required padding. And it is easy to extend to ram load images in future: zephyr is compiled with the correct ram location and imgtool can add the required header. Zephyr just generates the required hex file without any padding at all, and the same for the binary files.

nvlsianpu, 2019-07-08:
Originally imgtool.py had been able to operate only on *.bin files., and then was extended to support *.hex file as well. TEXT_SECTION_OFFSET might be different than 0x200 for other SoC.
Probably unused part of this area need to be zeroed in order to meet image SHA.

>CONFIG_FLASH_LOAD_OFFSET=DT_FLASH_IMAGE0_OFFSET + 0x200
CONFIG_FLASH_LOAD_SIZE=DT_FLASH_IMAGE0_SIZE - 0x200

That`s possible

I would not change this until it is required for feature you are developing (ques ram load images...).
Just need to be carefully while changing that.

Laczen, 2019-07-08:
@nvlsianpu, I know this offset might be different for other SOC, at the moment it is quite fixed in zephyr.

I will not use anything from MCUboot to support any feature I am developing. I will create a proper separate Kconfig. When I see things that are strange, or could be improved for mcuboot I am just reporting them. The way I look at mcuboot/zephyr the zephyr images generated should be as standard as possible, what is really needed is to set the load/exe offset OK (taking into account the header size and slot slot size), and that is all. It is then up to imgtool to generate a good image from this.

After this there might still be a required to offset the hex to load it into the correct slot (slot1) using nrfjprog but this I do not consider a task of imgtool.

But to get back on topic, is the generated hex file OK to be processed by imgtool ?

nvlsianpu, 2019-07-08:
yes it is ok.

Laczen, 2019-07-08:
If the generated hex file is OK, I will close the issue.

closed 2019-07-08 by Laczen