zephyrproject-rtos/zephyr #17381: DTS compatible property processing assumes specific driver exists

pabigot, 2019-07-06:
By experimentation I believe the semantics of the `compatible` property in Zephyr devicetree files is:

* The first entry must match a YAML binding
* Binding defines are generated for all entries

At least one driver (ssd16xx) is taking advantage of this to share a driver implementation among multiple displays, by providing a compatible like:

    compatible = "solomon,ssd16xxfb", "gd,gde0213b1";

then keying off the generated `DT_GD_GDE0213B1_0` to determine the specific parameters for the driver instantiation.

This is inconsistent with the devicetree specification, which says:

> The property value consists of a concatenated list of null terminated strings, from most specific to most general. They allow a device to express its compatibility with a family of similar devices, potentially allowing a single device driver to match against several devices.

For consistency with the specification--and with most other multi-entry compatible properties currently in Zephyr, which do go from specific to generic--the devicetree entry should be:

    compatible = "gd,gde0213b1", "solomon,ssd16xxfb";

However, that does not work: no bindings are generated for the node.  There is no diagnostic: the node is simply dropped.

The code appears to use only the first compatible when determining the binding to use for generation.  It must instead iterate through all compatibles in order, use the first one that's available, and should emit a diagnostic if it doesn't find any.

pabigot, 2019-07-06:
@jfischer-phytec-iot FYI.

It doesn't appear to be easy to fix this in the current DTS scripting, so a solution may not be available until the in-work rewrite is complete.

ulfalizer, 2019-07-08:
The rewrite iterates through the compatibles and uses the first binding it finds. Hadn't noticed the old code didn't until now.

Would need to turn `get_binding()` in `extract/globals.py` into a loop in the old code to fix it there.

jfischer-phytec-iot, 2019-07-10:
With #17398, the driver will not depend on it anymore. But anyway could be useful and nice to have.

pabigot, 2019-07-10:
The ssd16xx driver might not depend on it.  The incoming il03xx driver definitely will.

galak, 2019-07-30:
I think this should be fixed with the new generation code, if I swap the compat order in `boards/arm/reel_board/reel_board.dts` we generate defines for both compats.

pabigot, 2019-07-30:
This worked correctly when I tested the new code before merge, so we'll call it good.

closed 2019-07-30 by pabigot