zephyrproject-rtos/zephyr #17410: k_work should have a user_data field

GregBurns, 2019-07-08:
Typically APIs that have callbacks allow for a user_data parameter (usually `void*`) to be passed to the callback function to provide contextual information.

Proposed solution is to append a _user_data_ field to `struct k_work`

I considered wrapping `struct k_work` inside my own struct appending a user_data field and casting to the wrapper struct type inside the callback but this will not work with delayed work items.

andyross, 2019-07-08:
The intent of the API is indeed that the struct k_work gets embedded in whatever data structure you are managing, and you use CONTAINER_OF() on the argument to recover the pointer from the k_work* argument.

This is... admittedly kinda weird.  But it's robust (even with k_delayed_work structs) and provides all the context a called function is going to need.  The advantage to this scheme is that you don't need dynamic allocation, nor a seperate pointer field to  track the relationship between the disjoint structs.  It's a common pattern in zephyr (c.f. data structures like slist/dlist/rbtree).

GregBurns, 2019-07-08:
Of course it works because _k_work_ is the first field in _k_delayed_work_. If this is expected common usage it would be good to show this in the example code.

closed 2019-07-08 by GregBurns

GregBurns, 2019-07-09:
I didn’t notice this in the documentation earlier, but it now looks like my issue is a non-issue.

The single argument that is passed to a handler function can be ignored if it is not required. If the handler function requires additional information about the work it is to perform, the work item can be embedded in a larger data structure. The handler function can then use the argument value to compute the address of the enclosing data structure, and thereby obtain access to the additional information it needs.