zephyrproject-rtos/zephyr #17443: Kconfig: move arch-specific stack sizes to arch trees?

gnuless, 2019-07-09:
There are a few items in the `kernel/Kconfig` like

> config IDLE_STACK_SIZE
>         int "Size of stack for idle thread"
>         default 2048 if COVERAGE_GCOV
>         default 1024 if XTENSA
>         default 512 if RISCV32
>         default 320 if ARC || (ARM && CPU_HAS_FPU)
>         default 256

I don't think we should be special-casing like this in the arch-independent parts of the kernel if we can avoid it.  These should be hidden, and the `arch/*/Kconfig` should set the defaults.

ioannisg, 2019-07-10:
@gnuless this is ok for me, in principle, but we need to think about the overhead of having the same Kconfigs defined in multiple times. A case-by-case approach looks more reasonable to me.

The particular case you posted here can go the the /arch/*/Kconfig, sure, and needs also cleanup (e..g why having FPU should increase the size of the idle stack??)

Finally, strictly speaking, this is an enhancement, not really code bug, imho...

ioannisg, 2019-07-10:
CC @andrewboie

gnuless, 2019-07-10:
@ioannisg What exactly do you mean by "overhead"?  The overhead of maintaining the options in separate places, or the overhead of Kconfig processing at build/config time?

ioannisg, 2019-07-10:
> 
> 
> @ioannisg What exactly do you mean by "overhead"? The overhead of maintaining the options in separate places, or the overhead of Kconfig processing at build/config time?

the first :)

closed 2019-09-11 by gnuless