zephyrproject-rtos/zephyr #17454: Bluetooth: Add provisioner support

tsvehagen, 2019-07-10:
## Introduction

Currently, Zephyr support being provisioned over both PB-GATT and PB-ADV but not being a provisioner. There's a lot of things in common for the provisioner and the device and most of the needed code is already implemented.

The protocol can be found in chapter 5 of the Mesh Profile Specification.

https://www.bluetooth.com/specifications/mesh-specifications/

### Problem description

As it stands right now, some other device, typically a phone or computer, must act as a provisioner. The provisioner will hold all the DevKey's and will be the only one that can access the configuration server model for devices.

I can see multiple reasons for wanting to provision new devices from some sort of hub device. For example you might want to provision/configure devices via some web interface or you might want to do it remotely.

### Proposed change

I propose to add support for provisioning, starting with PB-ADV, by supporting saving multiple DevKey's and implement the missing parts of the provisioning protocol.

## Detailed RFC

I see three main changes that has to be made.

1) Support for multiple DevKey's

The bt_mesh_net structure currently only has one dev_key so this should probably be made into a list of DevKey's instead. This is similar to the how the AppKey's are handled but instead of a key being mapped to an AppId it will be mapped to a mesh address, ie the primary address of a device.

2) Add callback for BEACON_TYPE_UNPROVISIONED

The BEACON_TYPE_UNPROVISIONED is currently ignored but since it carries the device UUID which is used in the provisioning process, the application must be able to get this value somehow.

3) Implement missing provisioning functionality

The API function bt_mesh_provision is currently only used for provisioning the local device so some other API is needed to provision remote devices. This function must have access to the device MAC and UUID.

As stated in the Introduction, there's a lot of things in common for the provisioner and device. One thing that differs though is the initiation of the messages and for example generation of the PROV_DATA message.

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

1)

- Introduce a struct bt_mesh_dev_key that contains a mesh address and the device key for that address.
- In struct bt_mesh_net, change u8_t dev_key[16] to an array of struct bt_mesh_dev_key.
- Add functions bt_mesh_dev_key_[alloc|del|find] to manage the device keys. Similiar to bt_mesh_app_key.
- Change all uses of dev_key to use bt_mesh_dev_key_find first
- Change the setting bt/mesh/Net to only contain primary_addr
- Add the setting bt/mesh/DevKey/{addr} to contain all device keys

2)

- Add a unprovisioned_device callback in the bt_mesh_prov structure

3)

In prov.c

- Add a PROVISIONER flag to prov_link.flags
- Add a bt_mesh_pb_adv_open_link function that sets the PROVISIONER flag and kick things off
- Change link_ack to set LINK_ACTIVE and send the Invite message if PROVISIONER flag is set
- Change prov_capabilities to send Start and PubKey messages
- Change prov_input_complete to send Confirm message

The functions prov_pub_key, prov_confirm and prov_random are used for both devices and provisioner but the behaviour is a little different. This could be handled by checking the PROVISIONER flag.

- Change prov_complete to calculate and add the DevKey for the newly provisioned device

### Dependencies

If the settings structure is changed, devices with mesh already configured will have to be re-configured.

### Concerns and Unresolved Questions

- When capabilities are received from device, how to select the preferred OOB method? Use bt_mesh_prov struct?

- There should probably be some config for enabling provisioner support but how should this relate to the current BT_MESH_PROV for example?

- Should DevKey's be mapped just to a mesh address or must the UUID be included?

- Does it feel messy to mix the provisioning code with the device code in prov.c?

## Alternatives

One alternative is to not remove the dev_key from the bt_mesh_net struct but rather add a list that only contains keys for other devices while the local dev_key is left as is. IMHO it seems more consistent to keep them all in the same list.

aescolar, 2019-07-10:
CC @jhedberg

jhedberg, 2019-07-10:
Generally I agree with with most of what's being proposed here - thanks for creating a formal issue to track it!

Some comments:

> The API function bt_mesh_provision is currently only used for provisioning the local device so some other API is needed to provision remote devices. This function must have access to the device MAC and UUID.

The MAC is not needed (or useful), at least not for PB-ADV, since it's perfectly valid for a device to send each PDU using a unique Non-resolvable private address. For PB-GATT (once we get to it) we might need to cache this, but even then we could do it all based on UUID and do a new scan once the app requests to provision a given UUID.

> When capabilities are received from device, how to select the preferred OOB method? Use bt_mesh_prov struct?

Yes, that seems reasonable.

> There should probably be some config for enabling provisioner support but how should this relate to the current BT_MESH_PROV for example?

Not sure what you mean by "the current BT_MESH_PROV", but provisioner support should definitely be behind a new Kconfig (and possibly some sub-options as well).

> Should DevKey's be mapped just to a mesh address or must the UUID be included?

Most provisioners will also want to be configuration clients, so there may be more info we want to track. You also need to decide what makes sense to take care of in the stack and what should be left for the application. In addition to a provisioned node's primary element's address a provisioner also needs to track how many elements it has - this is critical since that way you know which addresses have been assigned so that you don't go assigning the same address to another node.

> Does it feel messy to mix the provisioning code with the device code in prov.c?

I'd be open to look at what it'd take to split this into three c-files: one for the common provisioning protocol and two for each of the two possible roles (provisioner and provisionee). Whether do go ahead with this or not depends mostly on how messy a single c-file becomes with all the #ifdefs you'll need to add to make the support compile-time configurable.

I'd also suggest fourth optional step to your list of things to do: support exporting the provisioner database so that multiple provisioners can collaborate by sharing information on provisioned nodes. There's a standard coming from the SIG at some point - if you have access to the Mesh Working Group you might want to look it up.

jhedberg, 2019-07-10:
One other things: our message sending APIs need to be able to specify whether to use the local device key or the remote device key. E.g. configuration server uses the local key, but configuration client uses the remote key.

tsvehagen, 2019-07-11:
> The MAC is not needed (or useful), at least not for PB-ADV, since it's perfectly valid for a device to send each PDU using a unique non-resolvable private address. For PB-GATT (once we get to it) we [...]

Yea that's right, I changed the issue to just UUID

> Not sure what you mean by "the current BT_MESH_PROV", but provisioner support should definitely be behind a new Kconfig (and possibly some sub-options as well).

He, not really sure myself but I think one thing might be the naming of the configs. Now there is BT_MESH_PROV, BT_MESH_PB_ADV and BT_MESH_PB_GATT and they all relate to being able to be provisioned. Does it make sense to have something like

BT_MESH_DEVICE_PROV, BT_MESH_DEVICE_PB_ADV, BT_MESH_DEVICE_PB_GATT
and
BT_MESH_PROVISIONER, BT_MESH_PROVISIONER_PB_ADV, BT_MESH_PROVISIONER_PB_GATT

> Most provisioners will also want to be configuration clients, so there may be more info we want to track. You also need to decide what makes sense to take care of in the stack and what should be [...]

Good point. So after the provisioning is done it is necessary to get the device composition and then save the number of elements for the device. It feels like this should also be saved together with the DevKey for that device so maybe instead of settings bt/mesh/DevKey/{addr} it should be bt/mesh/Device/{addr} where both DevKey and number of elements is saved.

This raises two new questions:
1. How to get the device composition after provisioning?
2. What to do if provisioning is ok but reading device composition fails?

Possible solutions:

1.
It seems resonable to use the existing cfg_cli API. A call to bt_mesh_cfg_comp_data_get will block until answer or timeout. The API call to do provisioning could be made into a blocking call that first does provisioning and then uses the cfg_cli API before returning.

Another option would maybe be to provide a function which should be set as the complete callback in bt_mesh_prov.

2.
We could simply treat the node as unprovisioned and not save it even though it actually has access to the network. It won't have any AppKey though.

> I'd be open to look at what it'd take to split this into three c-files: one for the common provisioning protocol and two for each of the two possible roles (provisioner and provisionee).

Yea that's what I was thinking as well but I think I'll try and add it to prov.c and see what it looks like.

> I'd also suggest fourth optional step to your list of things to do: support exporting the provisioner database so that multiple provisioners can collaborate by sharing information on provisioned [...]

I don't have access to that but it sounds interesting... :)

> One other thing: our message sending APIs need to be able to specify whether to use the local device key or the remote device key. E.g. configuration server uses the local key, but configuration client uses the remote key.

What API do you mean? The bt_mesh_model_send?

jhedberg, 2019-07-12:
> BT_MESH_DEVICE_PROV, BT_MESH_DEVICE_PB_ADV, BT_MESH_DEVICE_PB_GATT and BT_MESH_PROVISIONER, BT_MESH_PROVISIONER_PB_ADV, BT_MESH_PROVISIONER_PB_GATT

Yes, something like that, but please try to keep existing option names as-is to maintain backward compatibility with apps external to the tree (that may be referencing the existing options).

> So after the provisioning is done it is necessary to get the device composition and then save the number of elements for the device

That's not quite true. You're right that the number of elements information is part of the composition data, however you also get this number early on during provisioning through the Provisioning Capabilities PDU (see section 5.4.1.2). It would be bad if the provisioner only found out *after* provisioning how many unicast addresses the provisioned node actually consumed. The spec is intentionally designed to make it possible to have the provisioner and configuration client as different devices, which would mean that the provisioner must hand over device keys and unicast addresses to the configuration client. In practice however I haven't seen or heard anyone creating such a split yet, i.e. in practice almost all provisioners will also be configuration clients. Either way, we shouldn't necessarily enforce this marriage in Zephyr - i.e. it should be possible to build an app that only does one but not the other.

> How to get the device composition after provisioning?

We already have a Configuration Client API (which I think you referred to earlier), so this would be the method, I think. Meaning, it would be left to the application. We probably want to create a sample provisioner app under samples/bluetooth which shows this. We can later consider moving something to be part of the stack (perhaps as some sort of library) but I'd first like to see how this all ends up looking.

> What to do if provisioning is ok but reading device composition fails?

I'd say that's app-specific, but a sane thing would be to show an error to the user and ask them to reset the node into non-provisioned mode again (hopefully it has some button or similar to do this).

> What API do you mean? The bt_mesh_model_send?

Yes.

tsvehagen, 2019-07-12:
> You're right that the number of elements information is part of the composition data, however you also get this number early on during provisioning through the Provisioning Capabilities PDU (see section 5.4.1.2).

Ah I thought it was a bit weird that it wouldn't be available until after handing out an address but I totally missed that it was already available during provisioning. Like you say then, the configuration client things can be all up to the application.

Regarding the key used when sending, I was thinking that it would be solved by using bt_mesh_dev_key_find(ctx->addr) but I see the problem as even when you send to a remote address with cfg_srv you should still use your local DevKey. Any suggestion for handling this? Maybe adding a flag to bt_mesh_msg_ctx?

jhedberg, 2019-07-12:
> Regarding the key used when sending, I was thinking that it would be solved by using bt_mesh_dev_key_find(ctx->addr) but I see the problem as even when you send to a remote address with cfg_srv you [...]

We should do it the way we currently deal with the device key, i.e. a special key index value. Both App and Net Key indices are 12 bits in the spec so we can start counting down from 0xffff for any special values we need, e.g. currently we have the following in `include/bluetooth/mesh/access.h`:
```
#define BT_MESH_KEY_UNUSED        0xffff                                        
#define BT_MESH_KEY_DEV           0xfffe  
```

So we'd need a `BT_MESH_KEY_DEV_REMOTE` and `BT_MESH_KEY_DEV_LOCAL`. You could keep `BT_MESH_KEY_DEV` around for backward compatibility but make it a synonym with `BT_MESH_KEY_DEV_LOCAL`.

One note: I'm off on summer vacation today, so I won't be able to provide much (if any) feedback on this issue for the next several weeks.

tsvehagen, 2019-07-12:
> So we'd need a `BT_MESH_KEY_DEV_REMOTE` and `BT_MESH_KEY_DEV_LOCAL`.

Sounds good.

> One note: I'm off on summer vacation today, so I won't be able to provide much (if any) feedback on this issue for the next several weeks.

I think I have what I need to try and get something working. I hope you have a nice vacation!

tsvehagen, 2019-07-23:
I have created a pull request so I think the discussion can continue there instead.

https://github.com/zephyrproject-rtos/zephyr/pull/17729

trond-snekvik, 2019-11-14:
Resolved by #17729 :tada:

closed 2019-11-14 by trond-snekvik