zephyrproject-rtos/zephyr #17460: sample: gui/lvgl

chtavares592, 2019-07-10:
**Describe the bug**
Hello everyone, Someone could help me?
I was trying to run the sample/gui/lvgl on nRF52840 with adafruit Adafruit 2.2 inch TFT Display.
I built the image and flashed on the board, but the display just blinks and didn't show the message "hello world" and the count number. The screen stays white and that's it.
I already check if connections are right.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. west build -b nrf52840_pca10056 ..
3. west flash
4. The code runs right but the screen just blink

**Environment:**
 - OS: Debian Linux
 - Board: nRF52840
 - Display: Adafruit 2.2 inch TFT Display
 - Commit 1baae96af6573fb20de37dc9a76729b3d8b524f0

**Image the display running*
[image]

jfischer-phytec-iot, 2019-07-10:
@chtavares592 Does `samples/display/ili9340` work for you?

chtavares592, 2019-07-10:
@jfischer-phytec-iot didn't work too, happens the same error

chtavares592, 2019-07-10:
@jfischer-phytec-iot yes, now is running the samples/display/ili9340 ... 
But the sample/gui/lvgl nothing happens

chtavares592, 2019-07-10:
Now I used the `nrf52840_pca10056.conf` and `nrf52840_pca10056.overlay` of `samples/display/ili9340` at the `sample/gui/lvg` and everything works.

Before the pins were connected thus

nrf52840Pin | Adafruit TFTPin
-- | --
P0.27 | SCK
P0.31 | D/C
P0.30 | RST
P0.26 | MOSI
P0.29 | MISO
P0.4 | CS

Now this

nRF52840Pin | Adafruit TFTPin
-- | --
P1.15 | SCK
P1.13 | MOSI
P1.14 | MISO
P1.12 | CS
P1.11 | D/C

ioannisg, 2019-07-12:
@chtavares592 this issue was not manually closed, but #17473 was merged, so please check if this bug-report is now obsolete

closed 2019-07-12 by chtavares592