zephyrproject-rtos/zephyr #17480: holyiot_yj16019 cannot compile IEEE 802.15.4 L2

jukkar, 2019-07-11:
**Describe the bug**
After applying #17477 the sanitychecker is complaining IEEE 802.15.4 compilation issues for holyiot_yj16019 board.

**To Reproduce**
* Apply #17477 
* `sanitycheck -p holyiot_yj16019 -s tests/net/ieee802154/l2/net.ieee802154.l2`

**Expected behavior**
Tests pass

**Additional context**
Some of the errors look like this, there are lot of them.
```
In file included from zephyr/drivers/ieee802154/ieee802154_nrf5.c:40:
zephyr/ext/hal/nordic/nrfx/mdk/nrf52840.h:71:3: error: redeclaration of enumerator 'Reset_IRQn'
   Reset_IRQn                = -15,              /*!< -15  Reset Vector, invoked on Power up and warm reset                     */
   ^~~~~~~~~~
In file included from zephyr/ext/hal/nordic/nrfx/mdk/nrf.h:88,
                 from zephyr/soc/arm/nordic_nrf/nrf52/soc.h:17,
                 from zephyr/include/arch/arm/cortex_m/cmsis.h:17,
                 from zephyr/include/arch/arm/cortex_m/mpu/arm_mpu_v7m.h:10,
                 from zephyr/include/arch/arm/cortex_m/mpu/arm_mpu.h:13,
                 from zephyr/include/arch/arm/arch.h:240,
                 from zephyr/include/arch/cpu.h:17,
                 from zephyr/include/kernel_includes.h:34,
                 from zephyr/include/kernel.h:17,
                 from zephyr/include/logging/log_msg.h:9,
                 from zephyr/include/logging/log_core.h:9,
                 from zephyr/include/logging/log.h:11,
                 from zephyr/drivers/ieee802154/ieee802154_nrf5.c:16:
zephyr/ext/hal/nordic/nrfx/mdk/nrf52.h:71:3: note: previous definition of 'Reset_IRQn' was here
   Reset_IRQn                = -15,              /*!< -15  Reset Vector, invoked on Power up and warm reset                     */
   ^~~~~~~~~~
```

tbursztyka, 2019-07-11:
@carlescufi Would you have someone to look at it?

rlubos, 2019-07-12:
I've identified the problem, holyiot_yj16019 has nRF52832, which does not support 802.15.4. So the capabilities of the board are incorrect. Will fix.

PR #17501 (2019-07-12): boards: holyiot_yj16019: Fix radio capabilities
holyiot_yj16019 board uses nRF52832 SoC, which does not have 802.15.4
radio. Therefore, 802.15.4 should not be listed in it's capabilities.

Fixes #17480.
closed 2019-07-12 by ioannisg