zephyrproject-rtos/zephyr #17485: sanitycheck: Over-zealous checking for binary sections

pfalcon, 2019-07-11:
Steps to reproduce:

~~~
./scripts/sanitycheck --no-clean --enable-slow --inline-logs -j2  -s tests/bluetooth/shell/bluetooth.shell -p qemu_cortex_m3
~~~

Output:

~~~
Building testcase defconfigs...
Filtering test cases...
1 tests selected, 121412 tests discarded due to filters
total complete:    1/   1  100%  failed:    0
FAILED: qemu_cortex_m3/tests/bluetooth/shell/bluetooth.shell has unrecognized binary sections: ['_settings_handlers_area']
0 of 1 tests passed with 0 warnings in 20 seconds
~~~

Tested with master 74dbd835c36a4e3cb.

This issue seems to affect only qemu_cortex_m3. Initially spotted via Linaro CI: https://ci.linaro.org/view/lite-iot-ci/job/zephyr-upstream/4070/PLATFORM=qemu_cortex_m3,ZEPHYR_TOOLCHAIN_VARIANT=zephyr,label=docker-xenial-amd64-13/console . The issue started within last 2 weeks (our CI was broken due to SDK 0.10.1 during this window, so it's hard to specify more exact date).

pfalcon, 2019-07-11:
@galak: FYI

pfalcon, 2019-07-11:
Setting priority as high, as this breaks (full) CI.

aescolar, 2019-07-11:
CC @Laczen , @jhedberg @Vudentz @nvlsianpu

Laczen, 2019-07-11:
@pfalcon, @aescolar, this was reported before by @andrewboie.

The problem is the "_bt_settings_area" has been replaced by "_settings_handlers_area". As I am not a user of sanitycheck (it just takes to long on my development computer) I have not been able to test if the following correction is OK:

In line 911 of scripts/sanitycheck to replace "_bt_settings_area" to "_settings_handlers_area".

PR #17500 (2019-07-12): sanitycheck: update sections whitelist
In c20ff1150f717d79374997ed3b429b2449d2c3b8
and 5f19c8160ade6d5ef5ef5e440e86cf4468210542
(PR #16897)
_bt_settings_area was replaced with _settings_handlers_area

Update sanitycheck sections whitelist accordingly

Fixes #17485
closed 2019-07-12 by aescolar