zephyrproject-rtos/zephyr #17486: nRF52: SPIM: Errata work-around status?

tautologyclub, 2019-07-11:
We're having LOTS of trouble with SPIM3 on a nrf52840 rev 1 (Particle Boron). Looked through the errata: 

https://infocenter.nordicsemi.com/index.jsp?topic=%2Ferrata_nRF52840_Rev1%2FERR%2FnRF52840%2FRev1%2Flatest%2Ferr_840_new.html&cp=3_0_1_2_1

and there are multiple seemingly potentially catastrophic issues with SPIM3, namely 174, 193, 195 and 198.

What is that status of these erratas wrt Zephyr drivers? For example, 193 prohibits triggering TASKS_SUPEND before being sure all bytes are shifted out. Are these work-arounds something that the application developer has to account for or have they been implemented in the driver?

aescolar, 2019-07-11:
CC @anangl

anangl, 2019-07-12:
Workarounds for anomalies 195 and 198 are implemented in the `nrfx_spim.c` driver that is used by the `spi_nrfx_spim.c` shim. The first workaround is always active, the second one needs to be activated (it is not by default, see [nrfx_config_nrf52840.h](https://github.com/zephyrproject-rtos/zephyr/blob/master/ext/hal/nordic/nrfx_config_nrf52840.h#L1932)) by setting `NRFX_SPIM3_NRF52840_ANOMALY_198_WORKAROUND_ENABLED` to `1`. You can do this for instance by adding:
```
zephyr_compile_definitions(NRFX_SPIM3_NRF52840_ANOMALY_198_WORKAROUND_ENABLED=1)
```
to your application's `CMakeLists.txt`.

Anomalies 174 and 193 do not affect the driver as it does not use PPI nor the SUSPEND task.

tautologyclub, 2019-07-12:
@anangl Thanks. I for sure though 193 would be relevant since I have been told that the SPI/I2C masters will get auto-disabled after each transfer but this is done by some method other than SUSPEND then, I gather?

anangl, 2019-07-12:
> @anangl Thanks. I for sure though 193 would be relevant since I have been told that the SPI/I2C masters will get auto-disabled after each transfer but this is done by some method other than SUSPEND then, I gather?

The SUSPEND task is for suspending SPI transactions (a given transfer is simply interrupted, then continued after the RESUME task is triggered - the driver does not make use of this functionality), not for disabling the peripheral (this is done by setting the ENABLE register to 0).

tautologyclub, 2019-07-12:
Gotcha.

closed 2019-07-12 by tautologyclub

bwasim, 2021-04-30:
> Workarounds for anomalies 195 and 198 are implemented in the `nrfx_spim.c` driver that is used by the `spi_nrfx_spim.c` shim. The first workaround is always active, the second one needs to be [...]

@anangl , Sorry to bring this up on such an old issue, but does enabling ANAMOLY_198_WORKAROUND have any side effects on the rest of the system ? This work around writes to registers which aren't document in the any guides, and so we aren't sure what happens under the hood and if it has any impact on other stuff.