zephyrproject-rtos/zephyr #17487: v1.14-branch: SDK 0.10.1 support?

pfalcon, 2019-07-11:
Current master requires (forces) use of SDK 0.10.1, while current v1.14-branch vice-versa, is unable to work with it (due to lack of cmake/toolchain/zephyr/0.10.1 or whatever).

This complicates various triaging and CI tasks.

Is there interest for the latest LTS release to support the latest SDK? If not, d'oh.

pfalcon, 2019-07-11:
cc @galak

pfalcon, 2019-07-11:
Specifically issues happen with Zephyr.js project - that was barely updated to build against 1.14, we definitely don't want to switch it to master yet. But would be nice to use the current SDK (I don't even know if our (Linaro's) CI supports multiple SDK versions, likely we just switch to the latest.

pfalcon, 2019-07-17:
@ioannisg, @galak, Any feedback?

andrewboie, 2019-07-17:
I already have a backport PR which does this #17470

pfalcon, 2019-07-17:
@andrewboie : Thanks!

nashif, 2019-08-19:
this was already merged.

closed 2019-08-19 by nashif