zephyrproject-rtos/zephyr #17488: CDC_ACM USB on nRF device fails after suspend

aolowin, 2019-07-11:
**Describe the bug**
After a suspend and resume the USB serial port is no longer responsive. Possibly related to #16599.

**To Reproduce**
Steps to reproduce the behavior:
1. build samples/subsys/usb/cdc_acm for the nrf52840_pca10056 board
2. enable Logging and set USB_CDC_ACM_LOG_LEVEL_DBG=y
3. set CDC_ACM_IAD=y to allow Windows to correctly use the device
4. plug into Windows 10 PC, device is recognized and works
5. enter sleep mode on PC, device is suspended
6. after device resumes communication fails.
7. replugging the USB port returns it to correct operation

**Expected behavior**
When the PC resumes I expect the USB serial port to be operating.

**Impact**
Serious concern. We are developing a therapy product where the PCs will often be suspended. Forcing the user to replug the device every time will be a huge inconvenience.

**Screenshots or console output**
Comments in the log file below are preceeded by a \#
```
[00:00:53.647,460] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
--- 27 messages dropped ---
[00:00:53.647,827] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x0

## initial open terminal program
[00:00:53.648,254] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 115200 0 0 8
[00:00:53.648,651] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:16:02.571,105] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:16:02.571,472] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:16:02.571,868] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:16:02.572,235] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:16:02.572,784] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 115200 0 0 8
[00:16:02.573,089] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:16:02.573,455] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x3
[00:16:02.573,516] <dbg> cdc_acm_echo.main: DTR set
[00:16:02.574,005] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 115200 0 0 8
[00:16:02.574,432] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:16:02.578,643] <dbg> usb_cdc_acm.cdc_acm_int_in: CDC_IntIN EP[81]
[00:16:02.586,669] <dbg> usb_cdc_acm.cdc_acm_int_in: CDC_IntIN EP[81]
[00:16:03.599,578] <inf> cdc_acm_echo: Baudrate detected: 115200
[00:16:03.599,639] <dbg> usb_cdc_acm.cdc_acm_fifo_read: dev 0x20003f50 size 64 rx_ringbuf space 1023
[00:16:03.599,639] <dbg> cdc_acm_echo.interrupt_handler: tty fifo -> ringbuf 0 bytes
[00:16:03.599,670] <dbg> cdc_acm_echo.interrupt_handler: Ring buffer empty, disable TX IRQ

## send one byte
[00:16:47.832,885] <dbg> usb_cdc_acm.cdc_acm_read_cb: ep 1 size 1 dev_data 0x20003d8c rx_ringbuf space 1023
[00:16:47.832,946] <dbg> usb_cdc_acm.cdc_acm_fifo_read: dev 0x20003f50 size 64 rx_ringbuf space 1022
[00:16:47.832,946] <dbg> cdc_acm_echo.interrupt_handler: tty fifo -> ringbuf 1 bytes
[00:16:47.832,977] <dbg> usb_cdc_acm.cdc_acm_fifo_fill: dev_data 0x20003d8c len 1 tx_ringbuf space 1023
[00:16:47.832,977] <dbg> cdc_acm_echo.interrupt_handler: ringbuf -> tty fifo 1 bytes
[00:16:47.833,007] <dbg> usb_cdc_acm.tx_work_handler: Got 1 bytes from ringbuffer send to ep 82
[00:16:47.833,160] <dbg> usb_cdc_acm.cdc_acm_write_cb: ep 82: written 1 bytes dev_data 0x20003d8c
[00:16:47.833,160] <dbg> usb_cdc_acm.cdc_acm_write_cb: tx_ringbuf is empty
[00:16:47.833,190] <dbg> cdc_acm_echo.interrupt_handler: Ring buffer empty, disable TX IRQ

## close terminal program
[00:16:59.832,702] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x2

## open terminal program
[00:20:02.908,538] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:20:02.908,874] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:20:02.909,362] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:20:02.909,698] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:20:02.910,247] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 115200 0 0 8
[00:20:02.910,522] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:20:02.910,949] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x3
[00:20:02.911,407] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 115200 0 0 8
[00:20:02.911,712] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[... 87 lines cut ...]
[00:44:42.439,361] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8
[00:44:42.439,849] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x3
[00:44:42.440,307] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_LINE_CODING 115200 0 0 8
[00:44:42.440,643] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_GET_LINE_CODING 115200 0 0 8

## send one byte
[00:45:21.992,706] <dbg> usb_cdc_acm.cdc_acm_read_cb: ep 1 size 1 dev_data 0x20003d8c rx_ringbuf space 1023
[00:45:21.992,797] <dbg> usb_cdc_acm.cdc_acm_fifo_read: dev 0x20003f50 size 64 rx_ringbuf space 1022
[00:45:21.992,797] <dbg> cdc_acm_echo.interrupt_handler: tty fifo -> ringbuf 1 bytes
[00:45:21.992,828] <dbg> usb_cdc_acm.cdc_acm_fifo_fill: dev_data 0x20003d8c len 1 tx_ringbuf space 1023
[00:45:21.992,828] <dbg> cdc_acm_echo.interrupt_handler: ringbuf -> tty fifo 1 bytes
[00:45:21.992,858] <dbg> usb_cdc_acm.tx_work_handler: Got 1 bytes from ringbuffer send to ep 82
[00:45:21.992,980] <dbg> usb_cdc_acm.cdc_acm_write_cb: ep 82: written 1 bytes dev_data 0x20003d8c
[00:45:21.993,011] <dbg> usb_cdc_acm.cdc_acm_write_cb: tx_ringbuf is empty
[00:45:21.993,041] <dbg> cdc_acm_echo.interrupt_handler: Ring buffer empty, disable TX IRQ

## close terminal program
[00:45:56.242,187] <dbg> usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x2

```

**Environment (please complete the following information):**
 - OS: Linux (build), Windows 10 Pro (test)
 - Toolchain: zephyr-sdk-0.10.1
 - Commit SHA or Version used:  1.14.99,  f1b349aa72476802dac8e49b410ae34aaf24df4b

jfischer-phytec-iot, 2019-07-11:
@aolowin Please set USB_DEVICE_LOG_LEVEL and USB_CDC_ACM_LOG_LEVEL to Info, USB_DRIVER_LOG_LEVEL to Debug, and replace the console output above with a new one.

jfischer-phytec-iot, 2019-07-11:
@aolowin Can you test #17492, please?

ioannisg, 2019-07-12:
@anangl please, follow up the progress with this issue, as well :)

aolowin, 2019-07-12:
I've done the same test with #17492. It's better, in that the port doesn't lock up, but no data is transferred after a suspend. Log follows:

```
### Test with PR #17492

## initial open terminal program
[00:12:26.325,897] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read done: 7
--- 105 messages dropped ---
[00:12:26.325,958] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 7
[00:12:26.325,988] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0
[00:12:26.326,446] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:33 rt:161 v:0 i:0 l:7
[00:12:26.326,477] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:12:26.326,538] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 7
[00:12:26.326,599] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl write complete
[00:12:26.327,087] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:34 rt:33 v:3 i:0 l:0
[00:12:26.327,087] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:12:26.327,148] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0
[00:12:26.327,209] <dbg> cdc_acm_echo.main: DTR set
[00:12:26.327,209] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x81, len 10
[00:12:26.327,453] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:32 rt:33 v:0 i:0 l:7
[00:12:26.327,453] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:12:26.327,575] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read request
[00:12:26.327,636] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read done: 7
[00:12:26.327,667] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 7
[00:12:26.327,728] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0
[00:12:26.328,216] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:33 rt:161 v:0 i:0 l:7
[00:12:26.328,216] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:12:26.328,277] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 7
[00:12:26.328,338] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl write complete
[00:12:26.330,566] <dbg> usb_nrfx.usbd_event_transfer_data: write complete, ep 0x81
[00:12:26.330,627] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x81, len 10
[00:12:26.338,562] <dbg> usb_nrfx.usbd_event_transfer_data: write complete, ep 0x81
[00:12:27.351,562] <inf> cdc_acm_echo: Baudrate detected: 115200
[00:12:27.351,593] <dbg> cdc_acm_echo.interrupt_handler: tty fifo -> ringbuf 0 bytes
[00:12:27.351,623] <dbg> cdc_acm_echo.interrupt_handler: Ring buffer empty, disable TX IRQ

## send one byte
[00:14:00.095,977] <dbg> usb_nrfx.usbd_event_transfer_data: read request, ep 0x01
[00:14:00.096,069] <dbg> usb_nrfx.usbd_event_transfer_data: read complete, ep 0x01, len 1
[00:14:00.096,191] <dbg> cdc_acm_echo.interrupt_handler: tty fifo -> ringbuf 1 bytes
[00:14:00.096,221] <dbg> cdc_acm_echo.interrupt_handler: ringbuf -> tty fifo 1 bytes
[00:14:00.096,252] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x82, len 1
[00:14:00.096,343] <dbg> usb_nrfx.usbd_event_transfer_data: write complete, ep 0x82
[00:14:00.096,405] <dbg> cdc_acm_echo.interrupt_handler: Ring buffer empty, disable TX IRQ

## close terminal program
[00:14:40.289,642] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:34 rt:33 v:2 i:0 l:0
[00:14:40.289,642] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:14:40.289,672] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0

## suspend (Windows 10 sleep)
[00:15:18.466,186] <dbg> usb_nrfx.usbd_event_handler: SUSPEND state detected
[00:15:18.466,247] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Suspend state

## resume
[00:15:36.938,079] <dbg> usb_nrfx.usbd_event_handler: RESUMING from suspend
[00:15:36.938,110] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB resume
[00:15:36.987,304] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:32 rt:33 v:0 i:0 l:7
[00:15:36.987,335] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:15:36.987,396] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read request
[00:15:36.987,487] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read done: 7
[00:15:36.987,518] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 7
[00:15:36.987,579] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0
[00:15:36.988,128] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:33 rt:161 v:0 i:0 l:7
[00:15:36.988,128] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:15:36.988,189] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 7
[00:15:36.988,250] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl write complete

## open terminal program
[00:16:37.837,127] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:33 rt:161 v:0 i:0 l:7
[... 138 lines cut ...]
[00:24:12.056,427] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:32 rt:33 v:0 i:0 l:7
[00:24:12.056,427] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:24:12.056,518] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read request
[00:24:12.056,610] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read done: 7
[00:24:12.056,640] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 7
[00:24:12.056,701] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0
[00:24:12.057,128] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:33 rt:161 v:0 i:0 l:7
[00:24:12.057,128] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:24:12.057,189] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 7
[00:24:12.057,281] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl write complete

## send one byte
[... cut ...]
[00:24:47.996,398] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x82, len 1
[00:24:47.996,459] <dbg> usb_nrfx.usbd_event_transfer_data: write complete, ep 0x82
[00:24:47.996,551] <dbg> cdc_acm_echo.interrupt_handler: Ring buffer empty, disable TX IRQ

## close terminal program
[00:25:19.872,955] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:34 rt:33 v:2 i:0 l:0
[00:25:19.872,955] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:25:19.872,985] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0
```

jfischer-phytec-iot, 2019-07-13:
@aolowin I do not use Windows OS and do not have one running native. I can not reproduce your issue, I tested cdc_acm sample with the patch from #17492 on a ThinkPad running Linux OS.

aolowin, 2019-07-15:
@jfischer-phytec-iot it looks like with #17492 the following message was missing after a suspend:
<dbg> usb_nrfx.usbd_event_transfer_data: read complete, ep 0x01, len 1
any suggestions as to where to start looking or any extra debug I should enable?

jfischer-phytec-iot, 2019-07-15:
@carlescufi Could you verify it with #17492, please?

jfischer-phytec-iot, 2019-07-15:
@aolowin I will try to organize a suitable device and test is.

anangl, 2019-07-17:
I tried to replicate this but I didn't manage to. I used a Windows 10 machine (native one) and the current master (caa47e6c973c1010fe8f915c89bd83c9d6477036), without #17492. I see some significant differences in the logs I achieved:
```
## suspend (Windows 10 sleep)
[00:03:10.541,656] <dbg> usb_nrfx.usbd_event_handler: SUSPEND state detected
[00:03:10.541,687] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Suspend state
[00:03:10.541,687] <dbg> usb_nrfx.usb_dc_ep_disable: EP disable: 0x81
[00:03:10.541,687] <dbg> usb_nrfx.usb_dc_ep_disable: EP disable: 0x01
[00:03:10.541,717] <dbg> usb_nrfx.usb_dc_ep_disable: EP disable: 0x82
[00:03:14.402,954] <dbg> usb_nrfx.usbd_event_handler: RESUMING from suspend
[00:03:14.402,984] <dbg> usb_nrfx.hf_clock_enable: HF clock stop success (0)
[00:03:14.402,984] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Removed

## resume
[00:07:30.276,306] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB detected
[00:07:30.276,336] <dbg> usb_nrfx.hf_clock_enable: HF clock start success (-68)
[00:07:30.277,374] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Powered
[00:07:30.280,456] <dbg> usb_nrfx.usbd_event_handler: SUSPEND state detected
[00:07:30.280,487] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Suspend state
[00:07:36.574,096] <dbg> usb_nrfx.usbd_event_handler: RESUMING from suspend
[00:07:36.574,157] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB resume
[00:07:36.574,157] <dbg> usb_nrfx.usbd_work_handler: USBD reset event
[00:07:36.674,774] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:6 rt:128 v:256 i:0 l:64
[00:07:36.674,774] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:07:36.674,804] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 18
[00:07:36.674,896] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl write complete
[00:07:36.675,140] <dbg> usb_nrfx.usbd_work_handler: USBD reset event
[00:07:36.773,223] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:6 rt:128 v:256 i:0 l:18
[00:07:36.773,254] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:07:36.773,284] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 18
[00:07:36.773,376] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl write complete
[00:07:36.773,529] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:9 rt:0 v:1 i:0 l:0
[00:07:36.773,529] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:07:36.773,590] <dbg> usb_nrfx.usb_dc_ep_enable: EP enable: 0x81
[00:07:36.773,590] <dbg> usb_nrfx.usb_dc_ep_enable: EP enable: 0x82
[00:07:36.773,620] <dbg> usb_nrfx.usb_dc_ep_enable: EP enable: 0x01
[00:07:36.773,651] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0
[00:07:36.774,078] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:32 rt:33 v:0 i:0 l:7
[00:07:36.774,078] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:07:36.774,169] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read request
[00:07:36.774,230] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read done: 7
[00:07:36.774,261] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 7
[00:07:36.774,322] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0
[00:07:36.774,444] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:33 rt:161 v:0 i:0 l:7
[00:07:36.774,444] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:07:36.774,505] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 7
[00:07:36.774,627] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl write complete
```
It was a USB 3.0 port if that makes a difference. I'll check if with 2.0 this will look the same.

anangl, 2019-07-17:
@aolowin I manage to replicate this. I had to enable the "USB Wake Support" option in my machine's setup and now I can observe the behavior you described.

anangl, 2019-07-18:
I reproduced this issue also with Windows 7. Just needed to supply a proper `.inf` file so that Windows can recognize the device as a CDC one (I used the one from nRF5 SDK and changed VID and PID in the cdc_acm sample). Then I was able force the USB suspend by just disabling the device in the Device Manager and the USB resume by re-enabling it. After such cycle, the device behaves as @aolowin described in the initial comment.
@jfischer-phytec-iot Applying #17492 doesn't help much. As @aolowin already mentioned, although the port does not lock up after the terminal program is closed and re-opened, it is still impossible to get anything back after sending a character. On the first one sent I only see this in the logs:
```
[00:15:09.796,173] <dbg> usb_nrfx.usbd_event_transfer_data: read request, ep 0x01
[00:15:09.796,234] <dbg> usb_nrfx.usbd_event_transfer_data: read complete, ep 0x01, len 1
```
Then on the next character sent (no matter if the terminal program is closed and re-opened in between or not) I see another `read request, ep 0x01` message and that's all. No further messages on consecutive attempts to send.

I'll investigate this further.

jfischer-phytec-iot, 2019-07-18:
@anangl curious, maybe an other issue in cdc_acm class, but why it pops only on Windows OS?

anangl, 2019-07-31:
@aolowin @jfischer-phytec-iot I think we've got a couple of things causing this issue. First one is, as @jfischer-phytec-iot rightly points out in #17492, the endpoints should not be disabled when the device enters the Suspended state. I think that also USB transfers should not be canceled in this case, see https://github.com/zephyrproject-rtos/zephyr/pull/17492#pullrequestreview-268918115. And last but not least, the current CDC_ACM class implementation sends data [only when its stored device status is `USB_DC_CONFIGURED`](https://github.com/zephyrproject-rtos/zephyr/blob/5c96de6194a4bb14cca4d0b7b87234d229762b79/subsys/usb/class/cdc_acm.c#L550). And this stored status is [updated on (almost) every notification from the stack](https://github.com/zephyrproject-rtos/zephyr/blob/5c96de6194a4bb14cca4d0b7b87234d229762b79/subsys/usb/class/cdc_acm.c#L390), so after a suspend/resume cycle will stay at `USB_DC_RESUME`, preventing anything from being sent until the device is configured again.
After removing the disabling of endpoints and cancellation of transfers, and providing a dirty hack that made the CDC_ACM class seeing the device as configured after the bus activity is resumed, I was able to make the sample work properly after a suspend/resume cycle. But I'm not sure how the proper solution for this issue should look like. I'd like hear an opinion from @finikorg first. I believe that this is an issue in the USB device stack, not an nRF platform specific problem.

PR #18573 (2019-08-23): usb: cdc_acm: fix suspend handling in CDC ACM class
usb: cdc_acm: update state after RESUME event 
Update state after RESUME event.
Use LOG_INF for important events and log a
warning if the data is discarded.
Fixes #17488

samples: cdc_acm: rework sample logging 
Use  LOG_LEVEL_INF as defautl log level and use LOG_INF
for important messages. Relax while loop and
give CPU resources to low priority threads like logging.
commit 5fb7c114 (2019-08-23): usb: cdc_acm: update state after RESUME event

Update state after RESUME event.
Use LOG_INF for important events and log a
warning if the data is discarded.

Fixes #17488
closed 2019-08-23 by carlescufi