zephyrproject-rtos/zephyr #17496: Possible redundant code snippet in boilerplate.cmake

smwikipedia, 2019-07-12:
**Is your enhancement proposal related to a problem? Please describe.**
The related CMake code is redundant.

**Describe the solution you'd like**
Remove the redundant code.

**Describe alternatives you've considered**
No.

**Additional context**
No.

**Below 2 code snippets seem to be redundant.**

**Snippet 1:**

[image]

**Snippet 2:**
[image]

I tried 3 build scenarios:
- build with an empty build folder
- build with an existing build folder
- build with an existing build folder but specifying a different BOARD from the CACHE value at command line with -D

With all 3 scenarios:
- Snippet 1 is always executed, which set and unset `board_app_cmake_lists` at the same time. Effectively did nothing.
- Snippet 2 is never executed, because `board_app_cmake_lists` is always unset by Snippet 1.

So these 2 snippets may be removed.

aescolar, 2019-07-12:
CC @SebastianBoe

SebastianBoe, 2019-07-12:
This is for the scenario when the board is specified from the app's CMakeLists.txt, could you test that scenario as well?

smwikipedia, 2019-07-12:
I tried adding BOARD to the CMakeLists.txt in the "\zephyr\samples\hello_world\CMakeLists.txt". As below:

[image]

Then run below command without specifying BOARD with `-D` option. The build folder is empty.

`cmake -S zephyr\samples\hello_world -G Ninja -B zephyr.build\helloworld_ARM`

This time, Snippet 1 is not executed, because there's no `cached_board_value`. So line 149 condition is not met.

Snippet 2 is executed. `BOARD` is set to `reel_board` **again**.

Since the `BOARD` variable is already set in app's CMakeLists.txt, why do we need Snippet 2 to re-set the `BOARD` variable?

Or is there some other scenario I missed?

SebastianBoe, 2019-07-12:
Hi, it's been so long since I wrote this that I can't explain how it works.

But I observed that removing the snippets caused specifying BOARD through the app build scripts to no longer work. I can't explain that you are observing otherwise.

```
sebo@sebo-nix:~/ncs/zephyr/samples/hello_world$ cb
Zephyr version: 1.14.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4") 
CMake Error at /home/sebo/ncs/zephyr/cmake/app/boilerplate.cmake:167 (message):
  BOARD is not being defined on the CMake command-line in the environment or
  by the app.
Call Stack (most recent call first):
  CMakeLists.txt:7 (include)

-- Configuring incomplete, errors occurred!
sebo@sebo-nix:~/ncs/zephyr/samples/hello_world$ git diff | cat
diff --git a/cmake/app/boilerplate.cmake b/cmake/app/boilerplate.cmake
index f95fa48fb1..4135b1ff13 100644
--- a/cmake/app/boilerplate.cmake
+++ b/cmake/app/boilerplate.cmake
@@ -146,13 +146,6 @@ if(board_cli_argument STREQUAL CACHED_BOARD)
   unset(board_cli_argument)
 endif()
 
-set(board_app_cmake_lists ${BOARD})
-if(cached_board_value STREQUAL BOARD)
-  # The app build scripts did not set a default, The BOARD we are
-  # reading is the cached value from the CLI
-  unset(board_app_cmake_lists)
-endif()
-
 if(CACHED_BOARD)
   # Warn the user if it looks like he is trying to change the board
   # without cleaning first
@@ -170,9 +163,6 @@ elseif(board_cli_argument)
 elseif(DEFINED ENV{BOARD})
   set(BOARD $ENV{BOARD})
 
-elseif(board_app_cmake_lists)
-  set(BOARD ${board_app_cmake_lists})
-
 else()
   message(FATAL_ERROR "BOARD is not being defined on the CMake command-line in the environment or by the app.")
 endif()
diff --git a/samples/hello_world/CMakeLists.txt b/samples/hello_world/CMakeLists.txt
index 628615f491..0eb935bb1e 100644
--- a/samples/hello_world/CMakeLists.txt
+++ b/samples/hello_world/CMakeLists.txt
@@ -2,6 +2,8 @@
 
 cmake_minimum_required(VERSION 3.13.1)
 
+set(BOARD reel_board)
+
 include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
 project(hello_world)
 

```

About scenario 2, I haven't seen that before, if it works, great, if not, then it is not supported.

smwikipedia, 2019-07-12:
Thanks for your hint. I think I can explain it now. I will post the details tomorrow.

smwikipedia, 2019-07-15:
The current logic is to first collect the `BOARD` variable specification from all 3 possible **external** sources:

1. environment variable `%BOARD%`
2. the CLI `-D BOARD` option
3. and the `set(BOARD)` command in the app's `CMakeLists.txt`.

There's another **internal** source `CACHE entry CACHED_BOARD` is checked. This is the actual storage of the `BOARD` value used for build. It doesn't exist for the initial build.

During the collection process, some redundant variables are unset. 

And finally, the remaining collection result is used to set the `BOARD` variable according to below precedence rule:

> 2 > 1 > 3

Then the final `BOARD` value is stored in the CACHE entry `CACHED_BOARD` as the "selected board".

(I am new to CMake language. It took me some time to get used to its unique paradigm.)

tejlmand, 2022-02-17:
This has been fixed here: #29990
in this commit: https://github.com/zephyrproject-rtos/zephyr/pull/29990/commits/bfa77522dd72caacd38053308cddda92101a0642

> Cleanup board and shield cmake code so that code duplication is reduced.

tejlmand, 2022-02-17:
Re-opened, as the referred PR and commit didn't remove the snippets in question, only relocated them.

tejlmand, 2022-02-17:
The reason for the apparent redundant code is not for the initial run, but for CMake reruns.

For example, a sample does:
```cmake
set(BOARD foo)

find_package(Zephyr)
```
then `foo` will be used if user does `cmake <path-to-sample>`.

But if user initially does `cmake -DBOARD=bar`, then `bar` must be used as `BOARD` and not `foo`.
Also user must be allowed to do a subsequent CMake run, as `cmake -DBOARD=bar <path-to-build-folder>` as long as user doesn't change the actual board.

On subsequent CMake re-runs, it is important that build system continues to use `bar` and not suddenly switches to `foo`. Hence one set of the `set()` / `unset()` and checking.

Similar for the use-case, where user initially runs: `cmake <path-to-sample>`
if user then manually tries to switch board with: `cmake -DBOARD=newboard <path-to-build-folder>`
we must not switch board.

Currently the snippets referred are all part of those use-cases and are not redundant as they cater each specific flow.