zephyrproject-rtos/zephyr #17497: Write provistion and configure data to flash

david-foo, 2019-07-12:
how to store provision and config information so that the information can still be used after the next power up?
My board is microbit. what can I do?

jhedberg, 2019-07-12:
Are you referring to Bluetooth Mesh? If so, then that's already supported - you just need to make sure you have enabled `CONFIG_BT_SETTINGS`. That has some dependencies such as flash and FCB support. There are several existing samples that use this already in the current tree. That said, this does increase the required RAM, and since you're using the micro:bit (with only 16k of RAM) it might be challenging to make it all fit.

jhedberg, 2019-07-12:
Slight addition: enabling `CONFIG_BT_SETTINGS` is not quite sufficient - you also need to call `settings_load()` at the right time during bootup. Please see the existing samples for where/how that's done. You could e.g. do a `git grep BT_SETTINGS tests/ samples/` or `git grep settings_load samples/bluetooth/ tests/bluetooth/` to see where sample code can be found.

david-foo, 2019-07-16:
@jhedberg  Thank you, we are going to try this. If the RAM size can not be met, we have to make a change.

carlescufi, 2019-08-15:
Question answered

closed 2019-08-15 by carlescufi