zephyrproject-rtos/zephyr #17517: Consider marking vector tables as `const` when CONFIG_DYNAMIC_INTERRUPTS=0

chrisc11, 2019-07-12:
**Is your enhancement proposal related to a problem? Please describe.**

The elf section with the vector table gets marked as (`W`)ritable even if it's placed in `FLASH`, i.e

```
arm-none-eabi-readelf -S build/zephyr/zephyr.elf | grep text
  [ 1] text              PROGBITS        00000000 0000b4 000100 00 WAX  0   0  4
```

This happens because the vector tables created in `gen_isr_tables.py` are not defined with a `const` qualifier. If non-const data is placed in a read-only `MEMORY` section, the gnu linker will auto-promote the section to have the `SHF_WRITE` bit set.

While this does not cause any runtime issues, it does:

* allow attempts to hot patch the vector table to successfully compile even though the vector table is in a non-writable region
* cause tools like `arm-none-eabi-size` to classify the region as data rather than text

**Describe the solution you'd like**

I think `sw_isr_table.h`, `isr_tables.c`, & `gen_isr_tables.py` could simply be patched to add a `const` qualifier to `_irq_vector_table` & `_sw_isr_table` when the vector table is not dynamic

**Additional context**

Here's an example output of the ` west -v build -b nrf52840_pca10056 samples/basic/threads/` before/after a patch marking the sections as const was made:

Before:
```
arm-none-eabi-readelf -S build/zephyr/zephyr.elf | grep text
  [ 1] text              PROGBITS        00000000 0000b4 000100 00 WAX  0   0  4

arm-none-eabi-size build/zephyr/zephyr.elf
   text	   data	    bss	    dec	    hex	filename
  22576	    864	   7453	  30893	   78ad	build/zephyr/zephyr.elf
```

After:
```
# arm-none-eabi-size build/zephyr/zephyr.elf
   text	   data	    bss	    dec	    hex	filename
  22960	    480	   7453	  30893	   78ad	build/zephyr/zephyr.elf

arm-none-eabi-readelf -S build/zephyr/zephyr.elf | grep text
  [ 1] text              PROGBITS        00000000 0000b4 000100 00  AX  0   0  4
```

(NB: It also looks like the structs placed in `.devconfig.init` are also not marked as `const` but are placed in `ROM`. For similar reasons might be nice to mark those structures as `const` but that's probably outside the scope of this issue)

aescolar, 2019-07-15:
CC @andrewboie @ioannisg

go2sh, 2024-09-27:
Hey, just a quick drive by. I'm having the problem that my architecture has 2048 irqs and thus it uses 16kb of ram for the isr table. It has 2MB of flash but only 64k of fast sram. It would be wonderful to have some improvements here.
I see to options:
1) Move ISR table to ROM
2) Add the option to use hash table with open addressing, where the initial table is calculate through the python script.

BR,
Christoph