zephyrproject-rtos/zephyr #17543: dtc version 1.4.5 with ubuntu 18.04 and zephyr sdk-0.10.1

mas-co, 2019-07-15:
**Describe the bug**
Trying to build the hello_world project (https://docs.zephyrproject.org/latest/getting_started/index.html).  When trying to build via west or cmake I get an error that the detected version of dtc is unsupported (1.4.5 versus minimum supported 1.4.6).

echo $ZEPHYR_SDK_INSTALL_DIR 
yields "zephyr"
echo $ZEPHYR_TOOLCHAIN_VARIANT  
yields "/home/mas/zephyr-sdk-0.10.1"  (/home/mas is my home folder).

I had found one other related bug report, but this had been resolved when they discovered a misspelling in the environment variables.  I did check this (even cut and pasted the environment variables from the above referenced webpage).

**To Reproduce**
cd ~/source/zephyrproject/zephyr
source zephyr-env.sh
west build -b reel_board samples/hello_world
** or **
cmake -B build -GNinja -DBOARD=reel_board samples/hello_world

**Expected behavior**
no dtc error

**Impact**
cannot proceed

**Screenshots or console output**
-- west build: build configuration:
       source directory: /home/mas/source/zephyrproject/zephyr/samples/hello_world
       build directory: /home/mas/source/zephyrproject/zephyr/build
       BOARD: reel_board (origin: command line)
-- west build: generating a build system
Zephyr version: 1.14.99
-- Selected BOARD reel_board
-- Found west: /home/mas/.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
CMake Error at /home/mas/source/zephyrproject/zephyr/cmake/extensions.cmake:1324 (message):
  Assertion failed: The detected dtc version is unsupported.

      The version was found to be 1.4.5                                   
      But the minimum supported version is 1.4.6                        
      See https://docs.zephyrproject.org/latest/getting_started/                     
      for how to use the SDK's dtc alongside a custom toolchain.
Call Stack (most recent call first):
  /home/mas/source/zephyrproject/zephyr/cmake/host-tools.cmake:69 (assert)
  /home/mas/source/zephyrproject/zephyr/cmake/app/boilerplate.cmake:414 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
ERROR: command exited with status 1: /usr/local/bin/cmake -B/home/mas/source/zephyrproject/zephyr/build -S/home/mas/source/zephyrproject/zephyr/samples/hello_world -GNinja -DBOARD=reel_board
run as "west -v build -b reel_board samples/hello_world" for a stack trace

**Environment (please complete the following information):**
Ubuntu 18.04
Zephyr SDK sdk-0.10.1

**Additional context**
I am a complete Zephyr-OS noob.

andrewboie, 2019-07-16:
I don't even have 'dtc' installed on my system, AFAIK it gets it out of the SDK.

What is your output of:

```
$ $ZEPHYR_SDK_INSTALL_DIR/sysroots/x86_64-pokysdk-linux/usr/bin/dtc --version
```

I get:

```
Version: DTC 1.4.7-g78e113e8-dirty
```

If you get the same output, perhaps the build system is preferring a host installation of dtc over the SDK one?

mas-co, 2019-07-16:
Yes, I get the same output:
`Version: DTC 1.4.7-g78e113e8-dirty`
Any idea how to force it to use the one it came with?

andrewboie, 2019-07-16:
@mas-co I don't know the build system too well, unfortunately.
As a short-term bandaid you could remove the host dtc.
But definitely sounds like we have a bug here.
@SebastianBoe @galak FYI

joerchan, 2019-07-16:
@mas-co I get `DTC:FILEPATH=/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc`
even though `which dtc` returns `/usr/bin/dtc`. Maybe there is something wrong in your setup, do you have the env variable `ZEPHYR_SDK_INSTALL_DIR` set correctly?

The DTC can be overriden with `-DDTC=<your/dtc>` when invoking either cmake or west.

carlescufi, 2019-07-16:
@mas-co I can't reproduce this, although I force-installed an older dtc version.
I can't explain why this is happening, but can you try please cleaning your repository clone so that we are sure there's not some old CMake cache polluting the build? Try `git clean -fdx` from the root of your zephyr repo, though bear in mind that will delete any untracked files

carlescufi, 2019-07-16:
@mas-co can you try adding this line:

```
diff --git a/cmake/toolchain/zephyr/0.10.1/host-tools.cmake b/cmake/toolchain/zephyr/0.10.1/host-tools.cmake
index 604c8ade4c..4d1577fa29 100644
--- a/cmake/toolchain/zephyr/0.10.1/host-tools.cmake
+++ b/cmake/toolchain/zephyr/0.10.1/host-tools.cmake
@@ -6,6 +6,7 @@ set(HOST_TOOLS_HOME ${ZEPHYR_SDK_INSTALL_DIR}/sysroots/${TOOLCHAIN_ARCH}-pokysdk
 # suffixes added. Ensures that the SDK's host tools will be found when
 # we call, e.g. find_program(QEMU qemu-system-x86)
 list(APPEND CMAKE_PREFIX_PATH ${HOST_TOOLS_HOME}/usr)
+message(${HOST_TOOLS_HOME}/usr)

 # TODO: Use find_* somehow for these as well?
 set_ifndef(QEMU_BIOS            ${HOST_TOOLS_HOME}/usr/share/qemu)
```
and seeing what gets printed when you build?
In my case I see:

```
/home/carles/bin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr
```

mas-co, 2019-07-16:
@joerchan if i type `ls $ZEPHYR_SDK_INSTALL_DIR`  I get the contents of the /zephyr-sdk-0.10.1 folder.  The dtc in the zephyr sdk folder is  1.4.7-g78e113e8-dirty and if I add that to the west command using the `-DDTC=/home/mas/zephyr-sdk-0.10.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc` option, I get the following output:

> Zephyr version: 1.14.99
-- Selected BOARD reel_board
-- Found west: /home/mas/.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
CMake Error at /home/mas/source/new/zephyrproject/zephyr/cmake/extensions.cmake:1324 (message):
  Assertion failed: Zephyr toolchain variant invalid: please set the
  ZEPHYR_TOOLCHAIN_VARIANT-variable
Call Stack (most recent call first):
  /home/mas/source/new/zephyrproject/zephyr/cmake/generic_toolchain.cmake:54 (assert)
  /home/mas/source/new/zephyrproject/zephyr/cmake/app/boilerplate.cmake:430 (include)
  CMakeLists.txt:5 (include)

I triple-checked ZEPHYR_TOOLCHAIN_VARIANT and it is set to "zephyr"

I cut and pasted the environment variable settings from the install host dependencies webpage into my .bashrc file, so I dont' know what's up here.

mas-co, 2019-07-16:
@carlescufi I get the same results after running the git clean -fdx in the zephyrproject/zephyr 
Where do you want me to add the --diff... line?

carlescufi, 2019-07-16:
@mas-co 

Please add the line: 
```
message(${CMAKE_PREFIX_PATH}) 
```
in the file `cmake/toolchain/zephyr/0.10.1/host-tools.cmake`, after the line ` list(APPEND CMAKE_PREFIX_PATH ${HOST_TOOLS_HOME}/usr)`

mas-co, 2019-07-16:
@carlescufi 
After including the message statement in the host-tools.cmake file, the output of `cmake -B build -GNinja -DBOARD=reel_board samples/hello_world` is below.

Zephyr version: 1.14.99
-- Selected BOARD reel_board
-- Found west: /home/mas/.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
CMake Error at /home/mas/source/new/zephyrproject/zephyr/cmake/extensions.cmake:1324 (message):
  Assertion failed: The detected dtc version is unsupported.

      The version was found to be 1.4.5
      But the minimum supported version is 1.4.6
      See https://docs.zephyrproject.org/latest/getting_started/
      for how to use the SDK's dtc alongside a custom toolchain.
Call Stack (most recent call first):
  /home/mas/source/new/zephyrproject/zephyr/cmake/host-tools.cmake:69 (assert)
  /home/mas/source/new/zephyrproject/zephyr/cmake/app/boilerplate.cmake:414 (include)
  CMakeLists.txt:5 (include)

aescolar, 2019-07-16:
Just to confirm the issue: This also started happening for me in a Ubuntu 18.04 machine around 2 weeks ago (I just took the shortest path and updated the dtc installed in the system).

joerchan, 2019-07-17:
@mas-co The message you added did not appear. It seems to me that you have some misconfiguration of your SDK.
The DTC error you get could be that the path to DTC is invalid.
Please check these (`env | grep ZEPH`):
```
ZEPHYR_TOOLCHAIN_VARIANT=zephyr
ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk
ZEPHYR_BASE=/home/joakim/repos/zephyrproject/zephyr
```
And double check that the path to everything is correct.
Check that you are in fact pointing to the correct SDK:
```
cat $ZEPHYR_SDK_INSTALL_DIR/sdk_version
0.10.1
```

Check that cmake picked up on your SDK install:
```
cat CMakeCache.txt| grep SDK      
//Zephyr SDK install directory
ZEPHYR_SDK_INSTALL_DIR:PATH=/opt/zephyr-sdk
```
Lastly check that you DTC was correctly selected:
```
cat CMakeCache.txt| grep DTC
DTC:FILEPATH=/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc
```

carlescufi, 2019-07-17:
@mas-co please delete your build folder every time you run. I don't see any debug output from the `message()` statement, I assume because you didn't delete the build folder.

mas-co, 2019-07-17:
@carlescufi 
After deleting the build folder the Hello World project built without error and zephyr.elf was in the build/zephyr folder.
I wish I knew that you needed to delete the build folder every time.  (sorry, I am transitioning to Linux from Windows and I am new to cmake---and only occasionally using makefiles)

I appreciate everyone's efforts to get me going.  I hope to be able to share the kindness in the future.

@joerchan 
All of the variables were ok.  Just finished checking when I deleted the build folder.

closed 2019-07-17 by mas-co

carlescufi, 2019-07-17:
Thanks for updating and closing the issue for future reference @mas-co

mas-co, 2019-07-17:
@carlescufi 
You are welcome.
And again, thanks to everyone for their assistance!

mrquincle, 2020-01-02:
@mas-co Deleting the build folder is a drastic measure. Probably you did run `west build` before you had installed `zephyr`:

```
cat build/CMakeCache.txt| grep SDK
//Zephyr SDK install directory
ZEPHYR_SDK_INSTALL_DIR:PATH=
```

Then if you install `zephyr` after all, the `cmake` build system doesn't pick it up. It uses the now stale `ZEPHYR_SDK_INSTALL_DIR` instead of picking it up from an environmental variable. 

If I look into the source code I see namely

```
set_ifndef(ZEPHYR_SDK_INSTALL_DIR "$ENV{ZEPHYR_SDK_INSTALL_DIR}")
set(ZEPHYR_SDK_INSTALL_DIR ${ZEPHYR_SDK_INSTALL_DIR} CACHE PATH "Zephyr SDK install directory")
```

If the environmental variable is not defined this cache variable will become empty.

```
function(set_ifndef variable value)
  if(NOT ${variable})
    set(${variable} ${value} ${ARGN} PARENT_SCOPE)
  endif()
endfunction()  
```

It can be solved by either checking also in set_ifndef if `${${variable}}` is not EQUAL to an empty string (even if it is defined). Or the variable should only be set if it's not empty. Or the function where is checked if the `ZEPHYR_SDK_INSTALL_DIR` variable is set should `return()` before it is written to the CMake cache.

medabme, 2020-04-11:
@mas-co 
Hi, I've got the same problem. What is exactly the "build folder" and where could I fond it?