zephyrproject-rtos/zephyr #17545: Licensing and reference to public domain material

pabigot, 2019-07-15:
PR #17468 exposes what may be a gap in the licensing infrastructure: the code is Apache-2.0 as required, but because I'm documenting that it's derived from another source that includes the words "public domain" the licensing check fails.  

Moving those words to a separate comment outside the standard header didn't fix it.

Rather than continue to try to game the system by inserting hard spaces or splitting the sentence  across comments, I'd like some insight from those who deal with Zephyr licensing:

Is the way I've used and referenced the public domain description of the implemented algorithm acceptable?

If it is, what's the process of getting the license block removed so people will start to review this PR rather than dismiss it because it's got a big red X next to it in the list?

If it's not, how should I provide credit to the author of the original material, who has placed it in the public domain?  See also: https://github.com/zephyrproject-rtos/zephyr/pull/17468#issuecomment-510138292

pabigot, 2019-07-15:
@galak @nashif fyi, not sure who else might need to be in on this.

dleach02, 2019-10-14:
@kestewart , Do you have any feedback on this item?

kestewart, 2019-10-22:
"Public Domain" as a concept varies from jurisdiction to jurisdiction, so its a bit awkward some ways to handle.    There are a couple of license on the SPDX list that encompass the public domain concept,  but the term public-domain is not a valid license, and so shouldn't be used in SPDX-License-Id:  statement.   

In particular,  if it fits,  can you use https://spdx.org/licenses/Unlicense.html?

@pabigot -  do you have access to the author/copyright holder, and can confirm that Unlicense matches the spirit?

closed 2020-10-04 by github-actions[bot]