zephyrproject-rtos/zephyr #17546: Bluetooth: Central Scan fails continuously if last connect attempt failed to complete

lanvis, 2019-07-15:
**Cannot Scan after failed connect attempt when operating as central**
Scan is Passive 0x64 for scan window and interval.
bt_le_scan_start() returns -5  (IOERROR).

**To Reproduce**
Configure to operate as a central.
Scan and connect to a peripheral.
Force a disconnect (place peripheral in a low rf visible location) where it peripheral can be found but the connect command fails (intermittently). 
    
   

Trace of my occurrence (is repeatable):
<dbg> app_ble.set_systick_scan:  ...Scan start at 0x0060b981
[01:43:10.376,312] <dbg> app_ble.new_ble_state_start: ...exiting 1
[01:43:10.376,342] <dbg> app_ble.start_next_ble_action: ...exiting next_ble_action
[01:43:14.029,327] <dbg> app_ble.cb_ble_device_found: ||||||||...found my transmitter.....|||||||
[01:43:14.029,327] <dbg> app_ble.ble_show_adv_data: ==>adv 23 bytes 
[01:43:14.029,327] <dbg> app_ble.ble_show_adv_data: ==> 
[01:43:14.029,327] <dbg> app_ble.ble_show_adv_data: 09,
[01:43:14.029,357] <dbg> app_ble.ble_show_adv_data: 09,
[01:43:14.029,357] <dbg> app_ble.ble_show_adv_data: 44,
[01:43:14.029,357] <dbg> app_ble.ble_show_adv_data: 65,
[01:43:14.029,357] <dbg> app_ble.ble_show_adv_data: 78,
[01:43:14.029,357] <dbg> app_ble.ble_show_adv_data: 63,
[01:43:14.029,357] <dbg> app_ble.ble_show_adv_data: 6f,
[01:43:14.029,388] <dbg> app_ble.ble_show_adv_data: 6d,
[01:43:14.029,388] <dbg> app_ble.ble_show_adv_data: 33,
[01:43:14.029,388] <dbg> app_ble.ble_show_adv_data: 32,
[01:43:14.029,388] <dbg> app_ble.ble_show_adv_data: 02,
[01:43:14.029,388] <dbg> app_ble.ble_show_adv_data: 01,
[01:43:14.029,388] <dbg> app_ble.ble_show_adv_data: 06,
[01:43:14.029,388] <dbg> app_ble.ble_show_adv_data: 03,
[01:43:14.029,418] <dbg> app_ble.ble_show_adv_data: 02,
[01:43:14.029,418] <dbg> app_ble.ble_show_adv_data: bc,
[01:43:14.029,418] <dbg> app_ble.ble_show_adv_data: fe,
[01:43:14.029,418] <dbg> app_ble.ble_show_adv_data: 05,
[01:43:14.029,418] <dbg> app_ble.ble_show_adv_data: ff,
[01:43:14.029,418] <dbg> app_ble.ble_show_adv_data: d0,
[01:43:14.029,418] <dbg> app_ble.ble_show_adv_data: 00,
[01:43:14.029,449] <dbg> app_ble.ble_show_adv_data: ae,
[01:43:14.029,449] <dbg> app_ble.ble_show_adv_data: 03,
[01:43:14.029,449] <dbg> app_ble.ble_show_adv_data: ||||||||...found my transmitter.....|||||||
[01:43:14.029,663] <inf> app_ble.ble_gap_stop_scan: Scan stopped

[01:43:14.029,693] <inf> app_ble.ble_gap_start_connect: Connect to peer address: D5:E2:FE:74:41:F7
[01:43:14.029,937] <inf> app_ble.ble_gap_start_connect: Connect started
[01:43:17.211,944] [1;31m<err> app_ble.scan_time_out: !!Connect Request did not complete!!![0m
[01:43:17.211,975] <dbg> app_ble.update_interval_control_data: !!! Next Scan in 53000 msec !!!
[01:44:08.969,757] <dbg> app_ble.ble_thread: ..ble evt rx =0x1003
[01:44:08.969,787] <dbg> app_ble.new_ble_state_start:
[01:44:08.969,787] <dbg> app_ble.new_ble_state_start: new state = STATE_TRANSMITTER_SCAN
[01:44:08.970,001] [1;31m<err> app_ble.ble_gap_start_scan: Scanning failed to start (err -5)
[0m
[01:44:08.970,001] <dbg> app_ble.new_ble_state_start: ...exiting 4
[01:44:08.970,001] <dbg> app_ble.start_next_ble_action: ...exiting next_ble_action
[01:44:08.970,001] <dbg> app_ble.update_interval_control_data: !!! Next Scan in 53000 msec !!!

More Information/Questions
Is there a way to restart the BLE stack without resetting?  There is a bt_enable() but as far as I can tell there is no bt_disable()????  How can a problem like this be recovered without a hard or software reset?

**Environment :**
 - OS: (Linux 16.04.10 )
 - Toolchain (Zephyr SDK 0.9.5)
 - 1.13.99
This is a virtual HCI environment with one nRF52840 dev board.

cvinayak, 2019-07-18:
@lanvis do you call bt_conn_disconnect before you start scanning again? (You need to cancel the connection creation)

lanvis, 2019-07-19:
I don't because I never got the connect callback so I assumed bt_conn * was invalid but now I see that I get a bt_conn* returned when I call connect (and this was OK because above we see "Connect started" in trace).

I will try and post the result and maybe we can close this.  Thanks!

lanvis, 2019-07-23:
I tried the disconnect after the connect request timeout but I am still hanging with same error code.

[00:07:46.219,787] <dbg> app_ble.new_ble_state_start: 
[00:07:46.219,787] <dbg> app_ble.new_ble_state_start: new state = STATE_TRANSMITTER_SCAN
[00:07:46.220,062] <inf> app_ble.ble_gap_start_scan: Scanning successfully started
[00:07:46.220,062] <dbg> app_ble.set_systick_scan:  ...Scan start at 0x000748e1
[00:07:46.220,062] <inf> app_ble.scan_state_start:    Scanning for 7 sec
[00:07:46.220,062] <dbg> app_ble.new_ble_state_start: ...exiting 1
[00:07:46.220,092] <dbg> app_ble.start_next_ble_action: ...exiting next_ble_action
[00:07:47.302,154] <dbg> app_ble.cb_ble_device_found: ||||||||...found my transmitter.....|||||||
[[00:07:47.302,307] <dbg> app_ble.ble_show_adv_data: ||||||||...found my transmitter.....|||||||
[00:07:47.302,520] <inf> app_ble.ble_gap_stop_scan: Scan stopped
[00:07:47.302,551] <inf> app_ble.ble_gap_start_connect: Connect to peer address: 5B:50:6A:4E:13:E7
[00:07:47.302,795] <inf> app_ble.ble_gap_start_connect: Connect started
[00:07:52.988,403] <dbg> app_ble.ble_thread: ..ble evt rx =0x01

[00:07:53.055,816] [1;31m<err> app_ble.scan_time_out: disconnect in incomplete connect state after scan: err=0[0m
[00:07:53.055,816] [1;31m<err> app_ble.scan_time_out: !!Connect Request did not complete!!![0m
[00:07:53.055,847] <dbg> app_ble.update_interval_control_data: !!! Next Scan in 53000 msec !!!
[00:07:53.055,877] <dbg> app_alg.alg_thread: ..alg evt rx =0x3804

[00:08:44.813,537] <dbg> app_ble.new_ble_state_start: 

[00:08:44.813,537] <dbg> app_ble.new_ble_state_start: new state = STATE_TRANSMITTER_SCAN
[00:08:44.813,720] [1;31m<err> app_ble.ble_gap_start_scan: Scanning failed to start (err -5)
[0m
[00:08:44.813,720] <dbg> app_ble.new_ble_state_start: ...exiting 4
[00:08:44.813,720] <dbg> app_ble.start_next_ble_action: ...exiting next_ble_action
[00:08:44.813,751] <dbg> app_ble.update_interval_control_data: !!! Next Scan in 53000 msec !!!
[00:08:44.813,781] <dbg> app_alg.alg_thread: ..alg evt rx =0x3804

lanvis, 2019-08-05:
Any ideas?  Is there a way to reinit BLE stack because this puts me in an unrecoverable state?

carlescufi, 2019-08-23:
@lanvis we have made quite a few changes to the controller, could you try with the latest master?

carlescufi, 2019-08-23:
@jhedberg @joerchan any chance this could be an issue in the Host?

joerchan, 2019-08-23:
@lanvis I cannot reproduce your issue.

If I start a connect command that does not find the device i can cancel it with bt_conn_disconnect without issue and start the scanner again. If the le_create_conn command was issued it will be stopped by a timeout in the host.

Can you specify which commit you get this on?
You say v1.13.99, which is not an actual version, and also very old. Do you have a chance to upgrade and reproduce it on something more recent?

carlescufi, 2019-10-15:
@cvinayak could this be the same Host buffer leak you are chasing?

joerchan, 2019-10-16:
@lanvis Please include the commit in your reports. Whenever it says x.x.99 it is a development version, and not a tag.

We have fixed a buffer leak in ATT. Can you check if this fixes your issue: https://github.com/zephyrproject-rtos/zephyr/pull/19580
Either cherry-pick it or update your version to something that includes it.

lanvis, 2019-10-17:
I have switch to using the Nordic BLE controller (beta) that will be released as a full support package from Nordic at some point (CONFIG_BT_LL_NRFXLIB=y).  It exhibits the same problem.  I will try this though.

carlescufi, 2019-10-17:
@lanvis I am going to close this issue since the scanning problem you described originally does not seem to be present anymore.
Regarding the other issue you see, please open a new GitHub issue.

Thanks in advance!