zephyrproject-rtos/zephyr #17548: Can't set thread name with k_thread_create prevents useful tracing information

nslowell, 2019-07-15:
The only way to set dynamic thread names is by calling k_thread_create() followed by k_thread_set_name().  However, sys_trace_thread_create() is called within k_thread_create(), and so the thread name is always NULL at that point (resulting in "Unnamed thread" in CTF).

For thread name information to be useful in trace, we need a way to have the name set before sys_trace_thread_create() is called.  Adding the parameter to k_thread_create() seems the simplest but requires a huge amount of code reworks.  Attempting to set the name in the k_thread struct 
before k_thread_create() could be acceptable, but it is currently wiped out and set back to NULL during initialization.

andrewboie, 2019-07-16:
>  sys_trace_thread_create() is called within k_thread_create()

Would it be possible to put a hook in k_thread_name_set() to update the tracing framework?

nslowell, 2019-07-16:
#17583

nslowell, 2019-07-16:
#17583

PR #17583 (2019-09-19): debug: tracing: add sys_trace_thread_name_set
Initial thread creation and tracing information
occurs with empty thread names.  For better tracing information,
we need to a way to get actual thread names if they are set
in order to better track thread names and their IDs.

Fixes #17548
closed 2019-09-19 by nashif