zephyrproject-rtos/zephyr #17550: SimpleLink WiFi host driver should revert back to using static memory model

vanti, 2019-07-15:
**Describe the bug**
During the upgrade to TI SDK 2.40, the SimpleLink WiFi host driver code has moved to using dynamic memory model. This only works well if malloc() is thread-safe, but the implementation in newlib does not appear to be. So it is best to revert back to using static memory allocation in the driver, the way it used to be before the upgrade.

**Impact**
Memory corruption issues might occur if SimpleLink host driver is used concurrently with other threads calling malloc().

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 0.10.1
 - Commit SHA: 92ddf63cb80141bc6a032f2c96801556b2f755bd

andrewboie, 2019-07-16:
> This only works well if malloc() is thread-safe, but the implementation in newlib does not appear to be

This seems like the bug right here.
I suspect the problem is in _sbrk(), in lib/libc/newlib/libc-hooks.c.
Probably need to change it to do an atomic_add() perhaps?

andrewboie, 2019-07-16:
@vanti can you try #17552

vanti, 2019-07-19:
Verified build of #17552. The latter should address the issue. I suggest we only apply this PR if #17552 does not make it in time for the 2.0 release.

galak, 2019-08-22:
Merged #17552, so I think we can close this.  @vanti, can you verify.

vanti, 2019-08-22:
Thanks @andrewboie @galak. Now that malloc() is thread-safe, we can keep using the dynamic memory model.

closed 2019-08-22 by vanti