zephyrproject-rtos/zephyr #17554: pyocd flash does not support the -b option for board ID so that the sanitycheck script can’t specified the board ID to flash when the host connected with multiple boards.

xuhao8210, 2019-07-16:
**Describe the bug**

pyocd flash does not support the -b option for board ID so that the sanitycheck script can’t specified the board ID to flash when the host connected with multiple boards.

**To Reproduce**
Steps to reproduce the behavior:
1.	One server connected to 2 boards : “reel_board” and “FRDM-K64F”
2.	Use sanitycheck script to run the “Hello World!” sample on the server , and specified the board ID.
3.	It is display that let me to choose which board to be flashed .
4.	the “Hello World!” sample can be run successfully after I inputted "0".

**Expected behavior**
For the step3, the question should not be displayed because I have specified the board ID for the parameter: “--west-flash="--board-id=” for the sanitycheck script, please refer the details in the Screenshots area
BTW, above step3 is not be happened when the host only connected with 1 board : “reel_board”, please also refer details in the Screenshots area

**Impact**

when the host connected with multiple real boards, the sanitycheck can’t support to auto run test cases .

**Screenshots** 

==============the host connected with 2 boards===========
[user@zbot1 zephyr]$ pyocd list
## => Board Name | Unique ID
-- -- ----------------------
0 => ARM DAPLink CMSIS-DAP | OSHW000041114e4500623006bcf000220571000097969900
1 => FRDM-K64F [k64f] | 0240020116B95E69EB47A3D1
[user@zbot1 zephyr]$ sanitycheck -p reel_board --device-testing --device-serial /dev/ttyACM0 --west-flash="--board-id=OSHW000041114e4500623006bcf000220571000097969900" --testcase-root ./samples/hello_world/ --detailed-report ~/work/testscript/outputtmp/test-report.xml -v
JOBS: 8
Cleaning output directory /home/user/work/zephyrproject/zephyr/sanity-out
Reading test case configuration files under /home/user/work/zephyrproject/zephyr/samples/hello_world...
Reading platform configuration files under /home/user/work/zephyrproject/zephyr/boards...
Reading platform configuration files under /home/user/work/zephyrproject/zephyr/scripts/sanity_chk/boards...
Building testcase defconfigs...
Filtering test cases...
1 tests selected, 189 tests discarded due to filters
-- west flash: using runner pyocd
-- runners.pyocd: Flashing Target Device
## => Board Name | Unique ID
-- -- ----------------------
0 => ARM DAPLink CMSIS-DAP | OSHW000041114e4500623006bcf000220571000097969900
1 => FRDM-K64F [k64f] | 0240020116B95E69EB47A3D1
q => Quit

Enter the number of the debug probe to connect:
> 0
[====================] 100%
0002112:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 16384 bytes (4 pages) at 38.24 kB/s
1/1 reel_board                samples/hello_world/sample.helloworld              PASSED (device)
1 of 1 tests passed with 0 warnings in 4 seconds

==============END: the host connected with 2 boards=======

============== the host connected with 1 board===========
[user@zbot1 zephyr]$ pyocd list
## => Board Name | Unique ID
-- -- ----------------------
0 => ARM DAPLink CMSIS-DAP | OSHW000041114e4500623006bcf000220571000097969900
[user@zbot1 zephyr]$ sanitycheck -p reel_board --device-testing --device-serial /dev/ttyACM0 --west-flash="--board-id=OSHW000041114e4500623006bcf000220571000097969900" --testcase-root ./samples/hello_world/ --detailed-report ~/work/testscript/outputtmp/test-report.xml -v
JOBS: 8
Cleaning output directory /home/user/work/zephyrproject/zephyr/sanity-out
Reading test case configuration files under /home/user/work/zephyrproject/zephyr/samples/hello_world...
Reading platform configuration files under /home/user/work/zephyrproject/zephyr/boards...
Reading platform configuration files under /home/user/work/zephyrproject/zephyr/scripts/sanity_chk/boards...
Building testcase defconfigs...
Filtering test cases...
1 tests selected, 189 tests discarded due to filters
-- west flash: using runner pyocd
-- runners.pyocd: Flashing Target Device
[====================] 100%
0000768:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 16384 bytes (4 pages) at 38.28 kB/s
1/1 reel_board                samples/hello_world/sample.helloworld              PASSED (device)
1 of 1 tests passed with 0 warnings in 3 seconds

==============END: the host connected with 1 board========

**Environment**

OS: Fedora release 30 (Thirty)
Version :   8401fda961

mbolivar, 2019-08-18:
I think the --west-flash option to sanitycheck already handles this explicitly:

```
$ sanitcheck -h
[...]
  --west-flash [WEST_FLASH]
                        Uses west instead of ninja or make to flash when
                        running with --device-testing. E.g "sanitycheck
                        --device-testing --device-serial /dev/ttyACM0 --west-
                        flash="--board-id=foobar" will translate to "west
                        flash -- --board-id=foobar" NOTE: device-testing must
                        be enabled to use this option.
```

mbolivar, 2019-08-18:
@xuhao8210 @ioannisg if you agree with my above comment, consider closing this as it is not a bug.

xuhao8210, 2019-08-19:
@mbolivar  I have used the --west-flash option for sanitycheck script to specified board ID, but it is still display that let me to choose which board to be flashed when one server connected to 2 boards.   So I think that the option does not work，and please refer the log in Screenshots section.  Thanks for you checking this problem.

mbolivar, 2019-08-19:
@xuhao8210 I'm so sorry for missing that in your original report.

The error turns out to be due to using an old -b option in our pyocd script, when we should be using -u for the newest pyocd.

I have tested the following PR and it fixes the issue for me:
https://github.com/zephyrproject-rtos/zephyr/pull/18474

Could you please test and +1 the pull request if it works for yoU?

xuhao8210, 2019-08-19:
@mbolivar  After I tried your solution to update the option from ‘-b’ to ‘-u’ for the row :” board_args = ['-b', board_id]” in the file:” scripts/west_commands/runners/pyocd.py”, it worked well. Thanks for your support.  But I checked that Shippable check was failure a few hours ago, have you fix it ? because I checked that the Pull request have been run the check again.

mbolivar, 2019-08-19:
> Thanks for your support.

Thanks for the report and sorry for misunderstanding out at first.

> But I checked that Shippable check was failure a few hours ago, have you fix it ?

I did fix it. Shippable is passing now.

xuhao8210, 2019-08-19:
@mbolivar  Got it, Thanks again.

PR #18474 (2019-08-19): scripts: west flash: fix --board-id for pyocd
Easy one-liner (one char-er?).

Fixes #17554

CC: @xuhao8210
commit f7a42753 (2019-08-19): scripts: west flash: fix --board-id for pyocd

Newer pyocd versions (specifically the 0.21.0 we have in our
requirements.txt) no longer support -b and have moved the same option
to -u. Keep up.

Fixes: #17554
closed 2019-08-19 by ioannisg