zephyrproject-rtos/zephyr #17555: assertion fail on qemu_x86_64 "log_backend_count_get() > 0"

andrewboie, 2019-07-16:
Something seems to be broken with logging enabled on the x86_64 platform.

Steps to reproduce:

Modify tests/kernel/fatal/prj.conf and add CONFIG_LOG=y

Build and run the test on qemu_x86_64:

```
SeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..Running test suite fatal
===================================================================
starting test - test_fatal
test alt thread 1: generic CPU exception
*** FATAL ERROR vector 6 code 0
***  RIP 8:0x1002f7 RSP 16:0x106d78 RFLAGS 0x202
***  RAX 0x1002f7 RCX 0x0 RDX 0x0 RSI 0x0 RDI 0x0
***  R8 0x0 R9 0x0 R10 0x0 R11 0x0
Caught system error -- reason 0
test alt thread 2: initiate kernel oops
*** FATAL ERROR vector 5 code 0
***  RIP 8:0x1003fe RSP 16:0x106d70 RFLAGS 0x2
***  RAX 0x1003ee RCX 0x0 RDX 0x0 RSI 0x0 RDI 0x0
***  R8 0x0 R9 0x0 R10 0x0 R11 0x0
Caught system error -- reason 2
test alt thread 3: initiate kernel panic
*** FATAL ERROR vector 5 code 0
***  RIP 8:0x1003b4 RSP 16:0x106d70 RFLAGS 0x2
***  RAX 0x1003a4 RCX 0x0 RDX 0x0 RSI 0x0 RDI 0x0
***  R8 0x0 R9 0x0 R10 0x0 R11 0x0
Caught system error -- reason 3
test alt thread 4: fail assertion
ASSERTION FAIL [0] @ ZEPHYR_BASE/tests/kernel/fatal/src/main.c:125
	intentionally failed assertion
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x100ece RSP 16:0x106d68 RFLAGS 0x202
***  RAX 0x104f1f RCX 0xa RDX 0x3f8 RSI 0x7d RDI 0x104eb8
***  R8 0xffffffff R9 0x0 R10 0x0 R11 0x0
Caught system error -- reason 3
PASS - test_fatal
===================================================================
Test suite fatal succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
ASSERTION FAIL [log_backend_count_get() > 0] @ ZEPHYR_BASE/subsys/logging/log_core.c:829
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x100ece RSP 16:0x107068 RFLAGS 0x202
***  RAX 0x104efe RCX 0xa RDX 0x3f8 RSI 0x33d RDI 0x1050fc
***  R8 0xffffffff R9 0x0 R10 0x0 R11 0x0
Caught system error -- reason 3
```

It's not just this test either.

andrewboie, 2019-07-16:
@andyross Found this while working on #17544

dcpleung, 2019-07-16:
@marc-hb saw this also. It was because of no logging backend being defined under `qemu_x86_64`. In contrast, `CONFIG_LOG_BACKEND_UART=y` is defined for `qemu_x86`.

marc-hb, 2019-07-16:
Yes I mentioned this at the bottom of PR #14777 and meant to file it, thanks @andrewboie.

I tried to enable UARTs in qemu_x86_64's Kconfig but that quickly turned out into much more effort than I anticipated. @andyross can you confirm it's not trivial?

Is there some alternative, "cheaper" backend that could be used to avoid the ASSERT? `/dev/null` maybe? :-) This ASSERT is probably the most frequent in the entire sanitycheck run. It's also racing with other outputs (as expected I guess) which makes test output non deterministic. Worst case:
```
Booting from ROM..Running test suite posix_fs_test
===================ASSE==================RTION =======FAIL===== [log_==backend_count_get(===) > 0] @ Z==EPHYR_BASE/subsy===s/logging/log_core=====.c:===
starting test - test_fs_mount
```

BTW I'm curious how the qemu_x86_64 prints to the console if not with UART(s)?

Here are [most of] the tests where I observed this:
```
./sanity-out/qemu_x86_64/tests/posix/fs/portability.posix/handler.log
./sanity-out/qemu_x86_64/tests/crypto/rand32/crypto.rand32.random_sw_systimer/handler.log
./sanity-out/qemu_x86_64/tests/crypto/rand32/crypto.rand32/handler.log
./sanity-out/qemu_x86_64/tests/subsys/logging/log_msg/logging.log_msg/handler.log
./sanity-out/qemu_x86_64/tests/subsys/logging/log_output/logging.log_output/handler.log
./sanity-out/qemu_x86_64/tests/kernel/common/kernel.common.misra/handler.log
./sanity-out/qemu_x86_64/tests/kernel/common/kernel.common/handler.log
```

andyross, 2019-07-16:
This is a known issue.  The 16550 UART driver didn't work on x86_64 when I enabled it the first time and I never got back to fix it.  There are two bugs for this filed somewhere that would address this: one is that the UART driver needs to be fixed, the other is that the logging system needs to work to log to the minimal printk hooks where provided (x86_64 and esp32 both have these so that printk logging works from boot before any OS code runs).

andrewboie, 2019-07-16:
I think this is a high-priority issue as at least in its current state, we're getting all kinds of CI failures due to this

marc-hb, 2019-07-16:
> we're getting all kinds of CI failures due to this

Interesting, do you have pointers to specific examples? I didn't think CI cared about ASSERTs, is it because of the race I mentioned earlier?

andrewboie, 2019-07-16:
> I didn't think CI cared about ASSERTs 

It does. ASSERTS are unconditionally enabled for sanitycheck runs unless you force them off.
I noticed an assortment of failures related to this in the context of developing #17544

marc-hb, 2019-07-16:
> It does. 

I meant: I didn't think CI or any other automation ever scanned any *.log files and checked for any ASSERT FAIL printed there, sorry for any confusion. At the sanitycheck level I see every day plenty of PASS with an ASSERT FAIL in the handler.log, see sample tests above.

> I noticed an assortment of failures related to this in the context of developing #17544

Have you noticed any CI, nightly or other automated build FAILures because of this ASSERT (or because any other ASSERT) *outside* any error handling change(s)?

andrewboie, 2019-07-16:
@marc-hb so what is supposed to happen is that sanitycheck scans the log output looking for fatal exception messages. Unless the test is tagged as "expect_fault", it's supposed to fail the test if it sees one. In addition, there's a delay after the "PROJECT_EXECUTION_SUCCESSFUL" to see if there are any crashes after the test completes.

It seems that there is some problem with detecting these crashes in x86_64, possibly due to how the exception messages were formatted. This might explain why I'm getting all kinds of failures with my patch (which standardizes exception formatting).

Long-term, we need to get CONFIG_LOG working with x86_64, this is a major functionality gap.
Short-term, perhaps this assertion could be removed.

andrewboie, 2019-07-16:
I sent a patch to remove the assert on x86_64 to alleviate CI-related pain.
Leave this open to address creating a working backend for this arch.

marc-hb, 2019-07-16:
> @marc-hb so what is supposed to happen is that sanitycheck scans the log output looking for fatal exception messages. Unless the test is tagged as "ignore_faults", it's supposed to fail the test if it sees one.

Thanks, found it. This one-line change shows most the tests affected by this issue - probably all of them if you test enough times.
```diff
--- a/scripts/sanity_chk/harness.py
+++ b/scripts/sanity_chk/harness.py
@@ -13,6 +13,7 @@ class Harness:
             "Kernel Panic",
             "Kernel OOPS",
             "BUS FAULT",
+            "ASSERTION FAIL",
             "CPU Page Fault"
             ]
```
I ran this once and got this list:
```
 qemu_x86_64/tests/crypto/rand32/crypto.rand32/handler.log
 qemu_x86_64/tests/kernel/common/kernel.common/handler.log
 qemu_x86_64/tests/kernel/common/kernel.common.misra/handler.log
 qemu_x86_64/tests/posix/fs/portability.posix.newlib/handler.log
 qemu_x86_64/tests/subsys/logging/log_msg/logging.log_msg/handler.log
 qemu_x86_64/tests/shell/shell/handler.log
 qemu_x86_64/tests/posix/fs/portability.posix/handler.log
 qemu_x86_64/tests/subsys/logging/log_list/logging.log_list/handler.log
```
The scan seems case-insensitive because it also caught this unrelated failure:
```
starting test - test_cmd_help
shell_execute_cmd(help): -8

    Assertion failed at ZEPHYR_BASE/tests/shell/src/main.c:32: test_shell_execute_cmd: (ret == result is false)
help
FAIL - test_cmd_help
```

PR #17599 (2019-07-18): Log backend for  x86_64 qemu
- Added log backend for x86_64 qemu
- Added `log_backend_std.h` with helper functions for simple backends
- Refactored simple logger backends to use functions (reduced code redundancy)

Fixes #17555.

If #17585 goes in before that it should be reverted in that PR.
closed 2019-07-18 by nashif