zephyrproject-rtos/zephyr #17559: Assertion failed: zephyr toolchain variant invalid

rakeshcx, 2019-07-16:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

carlescufi, 2019-07-16:
@rakeshcx could you please try deleting the `build/` folder completely and building again? I suspect a corrupted CMake cache.

rakeshcx, 2019-07-16:
still getting the same error after deleting the build directory & re-building the hello_world sample application

**This is the error after deleting the build directory and building first time**

-- west build: build configuration:
       source directory: /home/rakesh/zephyrproject/zephyr/samples/hello_world
       build directory: /home/rakesh/zephyrproject/zephyr/build (created)
       BOARD: nucleo_f446re (origin: command line)
-- west build: generating a build system
Zephyr version: 1.14.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4") 
-- Selected BOARD nucleo_f446re
CMake Error at /home/rakesh/zephyrproject/zephyr/cmake/extensions.cmake:1324 (message):
  Assertion failed: Zephyr toolchain variant invalid: please set the
  ZEPHYR_TOOLCHAIN_VARIANT-variable
Call Stack (most recent call first):
  /home/rakesh/zephyrproject/zephyr/cmake/generic_toolchain.cmake:54 (assert)
  /home/rakesh/zephyrproject/zephyr/cmake/app/boilerplate.cmake:428 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
ERROR: command exited with status 1: /usr/local/bin/cmake -B/home/rakesh/zephyrproject/zephyr/build -S/home/rakesh/zephyrproject/zephyr/samples/hello_world -GNinja -DBOARD=nucleo_f446re
run as "west -v build -b nucleo_f446re samples/hello_world" for a stack trace

**error after building for second time**

-- west build: build configuration:
       source directory: /home/rakesh/zephyrproject/zephyr/samples/hello_world
       build directory: /home/rakesh/zephyrproject/zephyr/build
       BOARD: nucleo_f446re (origin: CMakeCache.txt)
-- west build: building application
Error: could not find CMAKE_PROJECT_NAME in Cache
ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/rakesh/zephyrproject/zephyr/build
run as "west -v build -b nucleo_f446re samples/hello_world" for a stack trace

carlescufi, 2019-07-16:
I noticed you run the build as `sudo`. Can you please build as a regular user to see if that helps?

rakeshcx, 2019-07-16:
Build without sudo 

 **~/.local/bin/west build -b nucleo_f446re samples/hello_world**

Traceback (most recent call last):
  File "/home/rakesh/.local/bin/west", line 11, in <module>
    sys.exit(main())
  File "/home/rakesh/.local/lib/python3.6/site-packages/west/_bootstrap/main.py", line 499, in main
    wrap(wrap_argv)
  File "/home/rakesh/.local/lib/python3.6/site-packages/west/_bootstrap/main.py", line 485, in wrap
    west.main.main(argv)
  File "/home/rakesh/zephyrproject/.west/west/src/west/main.py", line 580, in main
    args.handler(args, unknown)
  File "/home/rakesh/zephyrproject/.west/west/src/west/main.py", line 332, in ext_command_handler
    command.run(*west_parser.parse_known_args(argv))
  File "/home/rakesh/zephyrproject/.west/west/src/west/commands/command.py", line 90, in run
    self.do_run(args, unknown)
  File "/home/rakesh/zephyrproject/zephyr/scripts/west_commands/build.py", line 135, in do_run
    self._setup_build_dir()
  File "/home/rakesh/zephyrproject/zephyr/scripts/west_commands/build.py", line 241, in _setup_build_dir
    os.makedirs(build_dir, exist_ok=False)
  File "/usr/lib/python3.6/os.py", line 220, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/home/rakesh/zephyrproject/zephyr/build'

carlescufi, 2019-07-16:
@rakeshcx can you please first clean up the build folder with sudo, then build without sudo?

rakeshcx, 2019-07-17:
problem solved. Previously I have cloned the zephyr repository with sudo(**sudo west init zephyrproject**
. This makes an issue. So I removed that previously cloned directory and again cloned with this command(**west init zephyrproject**). Now it's working fine.

closed 2019-07-17 by rakeshcx

carlescufi, 2019-07-17:
Thanks for letting us know