zephyrproject-rtos/zephyr #17564: Missing `stdlib.h` include when C++ standard library is used.

simonmarti, 2019-07-16:
**Describe the bug**

`stdlib.h` is not in the GCC search path when the C++ standard library is used.

**To Reproduce**

prj.conf:

```
CONFIG_NEWLIB_LIBC=y
CONFIG_CPLUSPLUS=y
```

main.cpp:

```cpp
#include <zephyr.h>
#include <misc/printk.h>
#include <array>

void main(void)
{
	std::array<int, 3> a{ 1, 2, 3 };

	for (const auto x: a)
	{
		printk("x: %d", x);
	}
}
```

Steps to reproduce the behavior:

1. mkdir build && cd build
2. cmake -GNinja -DBOARD=arduino_zero ..
3. ninja
4. Build fails because of missing include, see console output below

**Expected behavior**

The example code above should compile.

**Impact**

I can not use the C++ standard library for my current project.

**Screenshots or console output**

```cmd
> ninja
[15/101] Building CXX object CMakeFiles/app.dir/src/main.cpp.obj
FAILED: CMakeFiles/app.dir/src/main.cpp.obj 
C:\Users\username\tools\gcc-arm-none-eabi-7-2018-q2-update-win32\bin\arm-none-eabi-g++.exe  -DBUILD_VERSION=zephyr-v1.14.0 -DKERNEL -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -IC:/Users/username/zephyrproject/zephyr/kernel/include -IC:/Users/username/zephyrproject/zephyr/arch/arm/include -IC:/Users/username/zephyrproject/zephyr/include -IC:/Users/username/zephyrproject/zephyr/include/drivers -Izephyr/include/generated -IC:/Users/username/zephyrproject/zephyr/soc/arm/atmel_sam0/samd21 -IC:/Users/username/zephyrproject/zephyr/ext/hal/atmel/asf/sam0/include/samd21 -IC:/Users/username/zephyrproject/zephyr/ext/hal/cmsis/Include -isystem C:/Users/username/tools/gcc-arm-none-eabi-7-2018-q2-update-win32/arm-none-eabi/include -std=c++11 -Os -fcheck-new -Wno-register -fno-exceptions -fno-rtti -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros C:/Users/username/tmp/hello_world/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-common -mthumb -mcpu=cortex-m0plus -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -mabi=aapcs -march=armv6s-m -MD -MT CMakeFiles/app.dir/src/main.cpp.obj -MF CMakeFiles\app.dir\src\main.cpp.obj.d -o CMakeFiles/app.dir/src/main.cpp.obj -c ../src/main.cpp
In file included from c:\users\username\tools\gcc-arm-none-eabi-7-2018-q2-update-win32\arm-none-eabi\include\c++\7.3.1\ext\string_conversions.h:41:0,
                 from c:\users\username\tools\gcc-arm-none-eabi-7-2018-q2-update-win32\arm-none-eabi\include\c++\7.3.1\bits\basic_string.h:6352,
                 from c:\users\username\tools\gcc-arm-none-eabi-7-2018-q2-update-win32\arm-none-eabi\include\c++\7.3.1\string:52,
                 from c:\users\username\tools\gcc-arm-none-eabi-7-2018-q2-update-win32\arm-none-eabi\include\c++\7.3.1\stdexcept:39,
                 from c:\users\username\tools\gcc-arm-none-eabi-7-2018-q2-update-win32\arm-none-eabi\include\c++\7.3.1\array:39,
                 from ../src/main.cpp:8:
c:\users\username\tools\gcc-arm-none-eabi-7-2018-q2-update-win32\arm-none-eabi\include\c++\7.3.1\cstdlib:75:15: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
               ^~~~~~~~~~
compilation terminated.
[28/101] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/printk.c.obj
ninja: build stopped: subcommand failed..
```

**Environment (please complete the following information):**
* OS: Windows 10 1903
* Toolchain: GNU ARM Embedded 7-2018-q2-update
* Zephyr: v1.14.0

**Additional context**

This might be related to issue #15603. I tried the suggested workaround, but it does not help.

I tried several Zephyr versions:
* v1.14.1-rc1
* v1.14.99 (latest master, commit 1c6952c4f6de045fc5765373ff89683afd6396a4 )
* v1.14.0-rc1

v1.14.0-rc1 works but in this version the shell is not C++ compatible yet, so there are other problems to work around.

I also tried the latest GNU ARM Embedded toolchain (8-2019-q3-update), but also no success here.

If you can point my in the right direction, I will happily provide a fix myself.

carlescufi, 2019-07-16:
@simonmarti could you please try with the Zephyr SDK instead of GNU Arm Embedded and let us know if you can reproduce the same issue?

pabigot, 2019-07-21:
I'm unable to reproduce this with `zephyr-1.14.0` and `ZEPHYR_TOOLCHAIN_VARIANT=zephyr` resolves to SDK 0.10.0.  `<array>` is found and SDK 0.10.1 is unsupported in zephyr-1.14.0.  It also works with current head of `v1.14-branch`.

I can confirm the problem as reported is present when `ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb` and current `v1.14-branch`.

@carlescufi it appears you're using a C source file`main.c` instead of a C++ file `main.cc`, which would explain the failure to find `<array>`.

carlescufi, 2019-07-22:
> I'm unable to reproduce this with `zephyr-1.14.0` and `ZEPHYR_TOOLCHAIN_VARIANT=zephyr` resolves to SDK 0.10.0. `<array>` is found and correct output is written. SDK 0.10.1 is unsupported in zephyr-1.14.0. It also works with current head of `v1.14-branch`.
> 
> I can confirm the problem as reported is present when `ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb` and current `v1.14-branch`.
> 
> @carlescufi it appears you're using a C source file`main.c` instead of a C++ file `main.cc`, which would explain the failure to find `<array>`.

Thanks @pabigot. I can confirm the same in master. There is clearly an issue with GNU Arm Embedded and newlib/C++. @galak or @SebastianBoe is that something you could take a look at?

cszawisza, 2019-08-12:
I think that I've got the same error couple of months ago... this solved the issue

```
diff --git a/lib/libc/newlib/CMakeLists.txt b/lib/libc/newlib/CMakeLists.txt
index d368d1b89a8e1097eb7fda43044772c52738aba5..71061aa905131dbf85ead6dd0b56fb863e570b84 100644
--- a/lib/libc/newlib/CMakeLists.txt
+++ b/lib/libc/newlib/CMakeLists.txt
@@ -14,7 +14,7 @@ zephyr_library_sources(libc-hooks.c)
 # our libc headers to be picked instead of the toolchain's ffreestanding
 # headers.
 if(LIBC_INCLUDE_DIR)
-  zephyr_system_include_directories(${LIBC_INCLUDE_DIR})
+  zephyr_include_directories(${LIBC_INCLUDE_DIR})
 endif()
 
 if(LIBC_LIBRARY_DIR)

```

Can somebody confirm?

pabigot, 2019-08-12:
I can confirm that the problem goes away by making newlib headers not system headers, but given 7d2df83b5184b4566d43dd8a8d9abeb7c1d85694 was just merged I'm not convinced it's the right solution.  Presumably Zephyr and ARM do something different in builds that affects how system include files are handed.

SebastianBoe, 2019-08-12:
Hi, I have proposed a solution to the related issue #15603 here: https://github.com/zephyrproject-rtos/zephyr/pull/18189

The PR is intended to also fix this issue (17564), but I've not been able to reproduce all the reports here, so I'm not certain that it does so.

simonmarti, 2019-08-13:
@cszawisza Thanks, the change fixes the problem for my environment.

@SebastianBoe I tested the change proposed in #18189 with my environment, but it does not solve the issue. The error message is still the same.

The environment I used for the test:
* OS: Windows 10 1903
* Toolchain: GNU ARM Embedded 7-2018-q2-update / 8-2019-q3-update
* Zephyr: v2.0.0.0-rc1

pabigot, 2019-08-13:
Based on this and other investigations I think a solution based on @cszawisza's patch is the right way forward, but it requires additional changes to reverse #15627 and re-solve that change's issue another way.  @SebastianBoe I can take that on unless you want to handle it.

SebastianBoe, 2019-08-13:
Hi,

please do take this off my hands, I've haven't made much progress.

PR #18242 (2019-08-20): rework newlib directory prioritization to fix include processing for non-Zephyr toolchains
This work closes #17564 and related C++ failures.

Zephyr builds with `-ffreestanding` meaning that `__STDC_HOSTED__` is false.  Consequently the toolchain standard `<stdint.h>` is a minimal header that routes to `<stdint-gcc.h>` which defines integer types without defining the marker macro `__int64_t_defined` that signals newlib `<inttypes.h>` to provide support for `PRIu64` (#14310).  In hosted builds `<stdint.h>` instead sets some flags then delegates to the host-provided (newlib) implementation of `< [... cut]
commit 96c1b051 (2019-08-20): lib/newlib: revert treatment of libc files as system includes

The solution from #14312 of using -isystem to prioritize the position of
the libc directory bypasses the effect of -ffreestanding with respect to
libc symbols expected to be present in a non-hosted environment.

Further, it breaks C++ with the ARM Embedded toolchain as the system
fails to find the right file with #include_next.

Use a more fine-grained solution that explicitly includes the underlying
newlib header required for <inttypes.h> support before moving on to
include the next available one, whether system or non-system.

Cl [... cut]
commit 79e1aea7 (2019-09-25): lib/newlib: revert treatment of libc files as system includes

The solution from #14312 of using -isystem to prioritize the position of
the libc directory bypasses the effect of -ffreestanding with respect to
libc symbols expected to be present in a non-hosted environment.

Further, it breaks C++ with the ARM Embedded toolchain as the system
fails to find the right file with #include_next.

Use a more fine-grained solution that explicitly includes the underlying
newlib header required for <inttypes.h> support before moving on to
include the next available one, whether system or non-system.

Cl [... cut]
closed 2019-08-20 by galak