zephyrproject-rtos/zephyr #17581: linker script packing failure with subsys/fb fonts and native_posix_64

pabigot, 2019-07-16:
When `samples/display/cfb` is built on native_posix_64 (which requires a cloned configuration file present in #17580) it misfunctions, outputting:
```
initialized DISPLAY
[00:00:00.000,000] <dbg> cfb.cfb_framebuffer_init: number of fonts 89
font width 20, font height 32
font width 64, font height 118
font width 0, font height 0
font width 32, font height 126
font width 0, font height 0
font width 0, font height 0
font width 0, font height 0
font width 0, font height 0
```
There are only three fonts available, but the 24-byte structures that hold them are placed by `common-rom.ld` at 32-byte intervals.  The cfb driver expects them to be contiguous, and walking the synthesized array results in misbehavior.

I am unable to see an explanation for this behavior, nor a simple fix.  #17580 has an independently desirable patch that hides this, but the unpatched code could be used by somebody to figure this out.

npitre, 2019-07-16:
This would be fixed with the use of Z_STRUCT_SECTION_ITERABLE() and 
Z_STRUCT_SECTION_FOREACH().

npitre, 2019-07-16:
Something like this patch (let's see if it makes it through).
diff --git a/include/display/cfb.h b/include/display/cfb.h
index 065032f4f6..e8c4772a28 100644
--- a/include/display/cfb.h
+++ b/include/display/cfb.h
@@ -67,8 +67,7 @@ struct cfb_font {
  * @param _lc     Character mapped to last font element.
  */
 #define FONT_ENTRY_DEFINE(_name, _width, _height, _caps, _data, _fc, _lc)      \
-	static const struct cfb_font _name				       \
-	__attribute__ ((section(".font_entry."))) __attribute__((used)) =      \
+	static const Z_STRUCT_SECTION_ITERABLE(cfb_font, _name) =	       \
 	{								       \
 		.data = _data,						       \
 		.caps = _caps,						       \
diff --git a/include/linker/common-rom.ld b/include/linker/common-rom.ld
index e97a61f113..71371bba22 100644
--- a/include/linker/common-rom.ld
+++ b/include/linker/common-rom.ld
@@ -132,6 +132,6 @@
 	SECTION_DATA_PROLOGUE(font_entry_sections,,)
 	{
 		__font_entry_start = .;
-		KEEP(*(SORT_BY_NAME(".font_entry.*")))
+		KEEP(*(SORT_BY_NAME("._cfb_font.*")))
 		__font_entry_end = .;
 	} GROUP_LINK_IN(ROMABLE_REGION)

aescolar, 2019-07-16:
CC @jfischer-phytec-iot

pabigot, 2019-07-16:
@npitre Excised from email response for readability.  I'll check this and if it works add it to #17581.

```
diff --git a/include/display/cfb.h b/include/display/cfb.h
index 065032f4f6..e8c4772a28 100644
--- a/include/display/cfb.h
+++ b/include/display/cfb.h
@@ -67,8 +67,7 @@ struct cfb_font {
  * @param _lc     Character mapped to last font element.
  */
 #define FONT_ENTRY_DEFINE(_name, _width, _height, _caps, _data, _fc, _lc)      \
-	static const struct cfb_font _name				       \
-	__attribute__ ((section(".font_entry."))) __attribute__((used)) =      \
+	static const Z_STRUCT_SECTION_ITERABLE(cfb_font, _name) =	       \
 	{								       \
 		.data = _data,						       \
 		.caps = _caps,						       \
diff --git a/include/linker/common-rom.ld b/include/linker/common-rom.ld
index e97a61f113..71371bba22 100644
--- a/include/linker/common-rom.ld
+++ b/include/linker/common-rom.ld
@@ -132,6 +132,6 @@
 	SECTION_DATA_PROLOGUE(font_entry_sections,,)
 	{
 		__font_entry_start = .;
-		KEEP(*(SORT_BY_NAME(".font_entry.*")))
+		KEEP(*(SORT_BY_NAME("._cfb_font.*")))
 		__font_entry_end = .;
 	} GROUP_LINK_IN(ROMABLE_REGION)
```

aescolar, 2019-07-16:
@pagibot Silly note: If you add a "diff" in the block start (```diff) you get syntax highlighting.

pabigot, 2019-07-16:
Never much cared for syntax highlighting (did much of my work back in the day on a Wyse-75 in 132-column mode), but that's worth knowing.  Thanks.

pabigot, 2019-07-16:
@npitre Thanks, that solves it.  Patch replaces the previous approach in #17580.

npitre, 2019-07-16:
On Tue, 16 Jul 2019, Peter A. Bigot wrote:

> Never much cared for syntax highlighting (did much of my work back in the day on a Wyse-75 in 132-column mode), but that's worth knowing.  Thanks.

I have personal reasons to wish things were still that way. Sorry for 
messing up things.

I think github should at least honor markdown in email responses.

PR #17580 (2019-07-17): sample updates supporting native_posix_64
The first commit clones the board overrides so Kconfig options are set for this variant.

The second fixes  #17581.  (The cfb-specific patch will be pushed separately to prevent holding this PR up).
commit a66a036a (2019-07-17): subsys/cfb: correct font definition macro for linker script

This macro provides the required alignment directives to ensure that the
font definitions are placed properly for iteration as members of an
array object.

Closes #17581
closed 2019-07-17 by andrewboie