zephyrproject-rtos/zephyr #17584: k_mutex is not SMP-safe

andrewboie, 2019-07-16:
The mutex code was written to rely on a k_sched_lock() to prevent
preemption by other threads while inspecting the mutex state for the
uncontended fast paths. The k_mutex APIs cannot be legally used in
ISRs, so this is a safe and clever way to reduce the size of the
critical section.

Unfortunately it just doesn't work in SMP. Preventing the kernel from
preempting your thread says nothing about threads running on other
CPUs.

andrewboie, 2019-07-16:
There is an unfinished patch related to this #15164

andrewboie, 2019-07-16:
This is at least medium if not high priority, although k_mutex is not used much within the kernel itself, we cannot make such assertions for applications

andrewboie, 2019-07-16:
@jocelyn-li can you have someone look into this?

andyross, 2019-08-13:
Drop priority.  This is a real race, but it was found by inspection and has never been seen in the wild.

PR #18533 (2019-08-22): kernel/mutex: Fix locking to be SMP-safe
The mutex locking was written to use k_sched_lock(), which doesn't
work as a synchronization primitive if there is another CPU running
(it prevents the current CPU from preempting the thread, it says
nothing about what the others are doing).

Use the pre-existing spinlock for all synchronization.  One wrinkle is
that the priority code was needing to call z_thread_priority_set(),
which is a rescheduling call that cannot be called with a lock held.
So that got split out with a low level utility that can update the
schedule state but allow the caller to de [... cut]
commit 6f13980f (2019-08-22): kernel/mutex: Fix locking to be SMP-safe

The mutex locking was written to use k_sched_lock(), which doesn't
work as a synchronization primitive if there is another CPU running
(it prevents the current CPU from preempting the thread, it says
nothing about what the others are doing).

Use the pre-existing spinlock for all synchronization.  One wrinkle is
that the priority code was needing to call z_thread_priority_set(),
which is a rescheduling call that cannot be called with a lock held.
So that got split out with a low level utility that can update the
schedule state but allow the caller to d [... cut]
closed 2019-08-22 by nashif