zephyrproject-rtos/zephyr #17608: NMP timeout when uploading image with mcumgr over BLE under Linux

PhilippFinke, 2019-07-17:
**Describe the bug**
Every second time I try to upload an image with mcumgr over BLE under Linux (both Fedora 30 and Yocto based on our custom board) I receive a NMP timeout and the upload aborts. After starting another upload after this timeout the upload succeeds.

The problem occurs with the smp_svr sample on a nrf52840_pca10056 board as well on our custom HW that is also based on a nrf52840. As BLE dongles I used a hama Bluetooth 4.0 USB dongle and the hci_usb sample running on a nrf52840_pca10056.

Additional information: We have a LED display and a status LED on our custom HW that strangely flickers when the NMP timeout occurs. This will last a few seconds and then everything goes back to normal operation.

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash the smp_svr sample as described [here](https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html#smp-svr-sample).
2. Upload the new image as described in the link above. This shouldn't give a NMP timeout.
3. Right after 2. upload the same image again. This should give a NMP timeout.

**Expected behavior**
Uploading an image should work every time without a NMP timeout.

**Screenshots or console output**
Example output. Here I'm uploading an image for my custom HW but the problem occurs with any other too.

`root@gw-ppf:~# mcumgr --conntype ble --connstring ctlr_name=hci0,peer_id="de:e6:ba:92:2f:60" image upload pa_fw-V0.6.3.bin 
 166.34 KiB / 166.34 KiB [=======================================================================================================================================================================================================================] 100.00% 7.91 KiB/s 21s
Done

root@gw-ppf:~# mcumgr --conntype ble --connstring ctlr_name=hci0,peer_id="de:e6:ba:92:2f:60" image upload pa_fw-V0.6.3.bin 
 0 B / 166.34 KiB [---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%
Error: NMP timeout

root@gw-ppf:~# mcumgr --conntype ble --connstring ctlr_name=hci0,peer_id="de:e6:ba:92:2f:60" image upload pa_fw-V0.6.3.bin 
 166.34 KiB / 166.34 KiB [=======================================================================================================================================================================================================================] 100.00% 7.94 KiB/s 20s
Done
`

**Environment**
 - OS: Linux Fedora 30 and Yocto based on custom board.
 - Toolchain: gnuarmemb
 - Zephyr v1.14.0

nvlsianpu, 2019-08-13:
@PhilippFinke 

I can quote myself:
From: https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html?highlight=smp_svr
> At the beginning of the upload process, the target might start erasing the image slot, taking several dozen seconds for some targets. This might cause an NMP timeout in the management protocol tool. Use the -t <timeout-in-seconds option to increase the response timeout for the mcumgr command line tool if this occurs.

Please call mcumgr with `-t 20`, which should solve issue you observed.

Does this help you?

carlescufi, 2019-08-15:
@PhilippFinke is this still an issue, even with the comment above? If not please close this. Thanks!

PhilippFinke, 2019-08-15:
Just tried it. Works like a charme. I'll close the issue.

closed 2019-08-15 by PhilippFinke