zephyrproject-rtos/zephyr #17624: SRAM size configurations aren't always consistent

andyross, 2019-07-17:
So... I'm trying to productize a new heap test, and would like to size appropriately it to available memory.  We have a legacy CONFIG_SRAM_SIZE value, and a newer DT_SRAM_SIZE one.  There doesn't seem to have been any attempt to keep these synchronized.  Some platforms use one, some the other, some neither.  At least nothing seems to be exposing inconsistent numbers.

Check this with a hack to hello_world:

```patch
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -9,5 +9,19 @@
 
 void main(void)
 {
-       printk("Hello World! %s\n", CONFIG_BOARD);
+       printk("Hello World! %s:", CONFIG_BOARD);
+
+#ifdef CONFIG_SRAM_SIZE
+       printk(" CONFIG_SRAM_SIZE %d", CONFIG_SRAM_SIZE);
+#else
+       printk(" CONFIG_SRAM_SIZE ??");
+#endif
+
+#ifdef DT_SRAM_SIZE
+       printk(" DT_SRAM_SIZE %d", DT_SRAM_SIZE);
+#else
+       printk(" DT_SRAM_SIZE ??");
+#endif
+
+       printk("\n");
 }
```

And run it:
```
$ scripts/sanitycheck -T samples/hello_world
$ find sanity-out -name handler.log | xargs cat | grep SRAM_SIZE
Hello World! native_posix_64: CONFIG_SRAM_SIZE ?? DT_SRAM_SIZE ??
Hello World! native_posix: CONFIG_SRAM_SIZE ?? DT_SRAM_SIZE ??
Hello World! qemu_x86: CONFIG_SRAM_SIZE 4096 DT_SRAM_SIZE 4096
Hello World! qemu_x86: CONFIG_SRAM_SIZE 4096 DT_SRAM_SIZE 4096
Hello World! qemu_nios2: CONFIG_SRAM_SIZE 128 DT_SRAM_SIZE 128
Hello World! qemu_riscv32: CONFIG_SRAM_SIZE ?? DT_SRAM_SIZE 16
Hello World! qemu_x86_64: CONFIG_SRAM_SIZE 8192 DT_SRAM_SIZE ??
Hello World! qemu_x86: CONFIG_SRAM_SIZE 4096 DT_SRAM_SIZE 4096
Hello World! mps2_an385: CONFIG_SRAM_SIZE 4096 DT_SRAM_SIZE 4096
Hello World! qemu_xtensa: CONFIG_SRAM_SIZE ?? DT_SRAM_SIZE 65536
```

We need to decide on one mechanism, implement it, and have the legacy one either deprecated or driven from the value of the other.  As is, this is sort of a mess.

aescolar, 2019-07-17:
For native_posix what would you like reported? :)

andyross, 2019-07-17:
Something reasonable; personally I'd go with something relatively constricted to match existing hardware, maybe 4M or something.  But the point is that if the values are going to work they need to be universally exposed so that anyone can guess at an answer to the question "will my thing fit?".  The work involved to do this right now is sort of a mess.

galak, 2019-08-01:
So I think this are kinda in sync, its more that some platforms don't use CONFIG_SRAM_SIZE at all.

Right now we define it for ARC || ARM || NIOS2 || X86 || X86_64 and get it from DTS.  So its the fact that NATIVE_POSIX, XTENSA, and RISCV do they're own thing for memory is the issue.

https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/Kconfig#L95

MaureenHelm, 2020-07-10:
Is this still an issue?

galak, 2020-07-22:
@andyross any updates if this is still an issue.  I think this is more an enhancement to have a unified scheme for conveying what memory is (and what flash / code storage is).

galak, 2020-09-08:
@andyross ping

galak, 2020-09-09:
Closing this for now, @andyross feel free to re-open if you feel like this is still an issue.

closed 2020-09-09 by galak