zephyrproject-rtos/zephyr #17625: driver: gpio: PCAL9535A: can't write to register (read is possible)

chris-makaio, 2019-07-17:
**Describe the bug**
I'm using PCAL9535A driver on a custom nRF52840 board. Since all ADDR-Pins are connected to ground, I'm using address 0x20.
In my main application I'm doing
```
	gpio_test = device_get_binding(CONFIG_GPIO_PCAL9535A_0_DEV_NAME);
	if(gpio_test)
		{
			gpio_pin_configure(gpio_test, 7, GPIO_DIR_OUT);
		}
```

Log output is

> gpio_pcal9535a.write_port_regs: PCAL9535A[0x20]: Write: REG[0x2] = 0xff, REG[0x3] = 0xff
> i2c_nrfx_twim: Error 195952642 occurred for message 1
> gpio_pcal9535a: PCAL9535A[0x20]: error writing from register 0x2 (-5)

But if I add

```
while(true) {
		k_sleep(1000);
		gpio_pin_read(gpio_test, 7, &p2_val);
		LOG_DBG("Val: %u", p2_val);
	}
```
It works perfectly. So reading DOES work while writing doesn't.

**To Reproduce**
Well, use the PCAL9535A driver with a nRF52840 based board

**Expected behavior**
PCAL9535A should be configured to use pins for output

ioannisg, 2019-07-30:
I can't figure out who maintains this intel GPio driver

chris-makaio, 2019-07-30:
Uh...

But is this priority label of "low" in context of this driver or the whole project?
For this driver it's not low prio, it's a show stopper.
For the kernel, of course, it's not a problem

chris-makaio, 2019-08-06:
Alright, funny thing.. the driver is trying to write registers for both ports in one step:
`i2c_burst_write(i2c_master, i2c_addr, reg, buf->byte, 2);`

Changed (in write_port_regs) to:
`i2c_reg_write_byte(i2c_master, i2c_addr, reg, buf->byte[0]);`

Works

chris-makaio, 2019-08-06:
I could go through all of the code but does it make sense because maintainer is unknown? Or can I "adopt" ownership of this driver?

MaureenHelm, 2019-09-10:
> Alright, funny thing.. the driver is trying to write registers for both ports in one step:
> `i2c_burst_write(i2c_master, i2c_addr, reg, buf->byte, 2);`
> 
> Changed (in write_port_regs) to:
> `i2c_reg_write_byte(i2c_master, i2c_addr, reg, buf->byte[0]);`
> 
> Works

`i2c_burst_write` is problematic because some drivers implement the transaction as write subaddress, restart, write data, stop.

https://github.com/zephyrproject-rtos/zephyr/blob/1a6f4a6368960ddcbeb776f45f084163c7001f36/include/drivers/i2c.h#L491-L494

PR #18805 (2019-09-10): drivers/gpio: pcal9535a: fix i2c write
writing to both port registers of pcal9535a
was broken

fixes #17625
commit ed9c1ce5 (2019-09-10): drivers/gpio: pcal9535a: fix i2c write

writing to both port registers of pcal9535a
was broken

fixes #17625
closed 2019-09-10 by MaureenHelm