zephyrproject-rtos/zephyr #17626: Change sanitycheck to use 'gcovr' instead of 'lcov'

andrewboie, 2019-07-17:
LCOV is written in Perl and introduces some hardships for people trying to produce coverage reports on Windows. It is also a third-party package that has to be installed, with 1.14 being the minimum version.

gcovr, on the other hand, is written in Python, and is installable with pip; indeed it is already in our scripts/requirements.txt.

Change sanitycheck to use gcovr instead of lcov to produce our code coverage reports.

nashif, 2019-11-21:
We are going to stay with lcov, it does branch coverage, gcovr does not.

closed 2019-11-21 by nashif

chrta, 2019-11-22:
@nashif gcovr also supports some kind of branch coverage or is this not what is needed?
https://gcovr.com/en/stable/guide.html#tabular-output-of-branch-coverage

nashif, 2019-11-24:
> @nashif gcovr also supports some kind of branch coverage or is this not what is needed?
> [gcovr.com/en/stable/guide.html#tabular-output-of-branch-coverage](https://gcovr.com/en/stable/guide.html#tabular-output-of-branch-coverage)

sorry, I meant to say function coverage. gcovr does not support this as of now: https://github.com/gcovr/gcovr/issues/325