zephyrproject-rtos/zephyr #17630: efr32mg_sltb004a tick clock error

pabigot, 2019-07-17:
The application below, when built for `efr32mg_sltb004a` emits the timestamp information at 2 s intervals.  It appears the clock is running at half the expected speed.

```
#include <zephyr.h>
#include <misc/printk.h>
#include <stdio.h>

static void timer_handler(struct k_timer *tmr)
{
	u32_t t = k_uptime_get_32();
	unsigned int ms = t % 1000;
	t /= 1000;
	unsigned int s = t % 60;
	t /= 60;
	unsigned int min = t % 60;
	unsigned int hr = t / 60;

	printf("%u:%02u:%02u.%03u ; stuff\n",
	       hr, min, s, ms);
}
K_TIMER_DEFINE(tmr, timer_handler, NULL);

void main(void)
{
	printk("Hello World! %s\n", CONFIG_BOARD);
	printf("%u ticks/s = %u us/tick\n",
	       CONFIG_SYS_CLOCK_TICKS_PER_SEC,
	       1000000U / CONFIG_SYS_CLOCK_TICKS_PER_SEC);
	k_timer_start(&tmr, 0, K_SECONDS(1));
	k_sleep(K_FOREVER);
}
```

chrta, 2019-07-27:
Yes, you are right. This is a bug.

From the reference manual pg. 305

> The supported HFRCO frequency range is from 1 MHz to 38 MHz. The default HFRCO frequency is 19 MHz

The startup code in soc/arm/silabs_exx32/common/soc.c does not modify this, if CONFIG_CMU_HFCLK_HFRCO is selected.

pabigot, 2019-07-27:
There's no entry in `CODEOWNERS` for `soc/arm/silabs_exx32` so I've assigned this to @mnkp based on frequency of recent modifications in that area.

chrta, 2019-07-27:
Since this board has an 38.4MHz xtal populated, i think to fix this, the file boards/arm/efr32mg_sltb004a/efr32mg_sltb004a_defconfig should be changed:
```diff
--- a/boards/arm/efr32mg_sltb004a/efr32mg_sltb004a_defconfig
+++ b/boards/arm/efr32mg_sltb004a/efr32mg_sltb004a_defconfig
@@ -12,6 +12,6 @@ CONFIG_SERIAL=y
 CONFIG_CORTEX_M_SYSTICK=y
 CONFIG_GPIO=y
 CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=38400000
-CONFIG_CMU_HFCLK_HFRCO=y
+CONFIG_CMU_HFCLK_HFXO=y
 CONFIG_SOC_GECKO_EMU_DCDC=y
 CONFIG_SOC_GECKO_EMU_DCDC_MODE_ON=y
```

But i am not 100% sure if it would be also better to have some kind of sanity-check in the soc.c clock initialization code if CONFIG_CMU_HFCLK_HFRCO is selected.

closed 2019-08-13 by galak

mnkp, 2019-08-13:
> But i am not 100% sure if it would be also better to have some kind of sanity-check in the soc.c clock initialization code if CONFIG_CMU_HFCLK_HFRCO is selected.

I believe we need to rework the whole start up code related to clock configuration. We've recently introduced DTS `clock-frequency` property of CPU node. We should also move the definition of HFXO, HFRCO clocks to DTS (i.e. remove CONFIG_CMU_HFXO_FREQ Kconfig property) and then make sure that the start up code configures CPU clock as defined in DTS. This should happen regardless of which clock source HFRCO, HFXO is used. For now the fix done in #18199 should suffice.

PR #18199 (2019-08-13): boards/efr32mg_sltb004a: use crystal instead of RC oscillator
Selection of the HFRCO causes the SOC to stay at its power-up frequency
of 19 MHz.  Switch to the HFXO to use the configured frequency.

Closes #17630
commit cd6eac61 (2019-08-13): boards/efr32mg_sltb004a: use crystal instead of RC oscillator

Selection of the HFRCO causes the SOC to stay at its power-up frequency
of 19 MHz.  Switch to the HFXO to use the configured frequency.

Closes #17630
commit e98ecd0a (2019-08-13): boards/efr32mg_sltb004a: use crystal instead of RC oscillator

Selection of the HFRCO causes the SOC to stay at its power-up frequency
of 19 MHz.  Switch to the HFXO to use the configured frequency.

Closes #17630
commit 189aac17 (2019-08-13): boards/efr32mg_sltb004a: use crystal instead of RC oscillator

Selection of the HFRCO causes the SOC to stay at its power-up frequency
of 19 MHz.  Switch to the HFXO to use the configured frequency.

Closes #17630