zephyrproject-rtos/zephyr #17645: VSCode debugging Zephyr application

n3rd4i, 2019-07-18:
**Is your feature request related to a problem? Please describe.**
I would be helpful to have a working [VSCode](http://aka.ms/vscode) debugging (using [C/C++ extension](https://github.com/microsoft/vscode-cpptools/tree/master/Documentation) with [Cortex-Debug](https://github.com/Marus/cortex-debug)) facility for [Zephyr](https://docs.zephyrproject.org/latest/introduction/index.html) application using the Open Source editor.

**Describe the solution you'd like**
A working configuration which includes the necessary dependency for this to work on Linux with [Zephyr-SDK](https://docs.zephyrproject.org/latest/getting_started/installation_linux.html#zephyr-sdk) which in theory should be similar to [GNU ARM Embedded](https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html#gnu-arm-embedded)?

**Describe alternatives you've considered**
N/A

**Additional context**
I could only make this work on [Windows](https://docs.zephyrproject.org/latest/getting_started/installation_win.html) with [GNU ARM Embedded](https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html#gnu-arm-embedded) from this [example](https://blog.makerdiary.com/debugging-nrf52840-mdk-with-vscode/)

alexanderwachter, 2019-07-19:
@n3rd4i some of my configurations (launch.json)
``` JSON
{
			"type": "cortex-debug",
			"request": "launch",
			"servertype": "openocd",
			"cwd": "${workspaceRoot}",
			"executable": "build/zephyr/zephyr.elf",
			"name": "zephyr STM32L432KC",
			"device": "STM32L432KC",
			"configFiles": [
 				"zephyr/boards/arm/nucleo_l432kc/support/openocd.cfg"
					]
		},
		{
			"type": "cortex-debug",
			"request": "launch",
			"servertype": "openocd",
			"cwd": "${workspaceRoot}",
			"executable": "build/zephyr/zephyr.elf",
			"name": "zephyr STM32F746ZG",
			"device": "STM32F746ZG",
			"configFiles": [
 				"zephyr/boards/arm/nucleo_f746zg/support/openocd.cfg"
					]
		},
		{
			"type": "cortex-debug",
			"request": "launch",
			"servertype": "openocd",
			"cwd": "${workspaceRoot}",
			"executable": "build/zephyr/zephyr.elf",
			"name": "zephyr STM32F072B",
			"device": "STM32F072B",
			"configFiles": [
 				"zephyr/boards/arm/stm32f072b_disco/support/openocd.cfg"
					]
		}
```

I'm using Code-OSS Version: 1.35.1 on MANJARO. I'm not using the Zephyr SDK but native installed openocd and gcc build-essential for ARM

markusr, 2019-07-26:
Here is my `c_cpp_properties.json`
```
{
    "env" : {
        "myZephyrSourcePath": "/home/markus/zephyr/west/zephyr",
        "myZephyrSDKPath": "/home/markus/zephyr/sdk/sdk_0.10.0/"
    },    
    "configurations": [
        {
            "name": "Linux",
            "includePath": [
                "${myZephyrSourcePath}/include/",
                "${myZephyrSourcePath}/lib/libc/minimal/include/",
                "${myZephyrSourcePath}/arch/arm/include/",
                "${workspaceFolder}/build/zephyr/include/generated/"                
            ],
            "compilerPath": "${myZephyrSDKPath}/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "intelliSenseMode": "gcc-x64"
        }
    ],
    "version": 4
}
```

n3rd4i, 2019-07-28:
I forgot to mention that I'm using Windows.

bus710, 2019-11-15:
So here is my lounch.json for these:
- Ubuntu 19.10
- VSCODE + Cortex-Debug
- nRF52832-DK board and its internal JLink probe
- OpenOCD host
- gdb-multiarch is installed

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Zephyr nRF52832 openocd",
            "cwd": "${workspaceRoot}",
            "executable": "build/zephyr/zephyr.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "nrf52832_xxaa",
            "targetId": "nrf52",
            "boardId": "",
            "armToolchainPath": "${HOME}/zephyr-sdk-0.10.3/arm-zephyr-eabi/bin",
            "interface": "swd",
            "gdbpath": "/usr/bin/gdb-multiarch",
             "configFiles": [
                 "/usr/share/openocd/scripts/board/nordic_nrf52_dk.cfg"
            ]
        },
    ]
}
```

With the configuration above, I could run the debug session.
Although Zephyr document mentioned that OpenOCD awares threads, I didn't try it by myself yet to see the stacks of several threads in VSCODE.

n3rd4i, 2019-11-19:
There is news on this front from [PlatformIO](https://github.com/platformio/platformio-core/issues/1613#issuecomment-554683964).

ivankravets, 2019-11-19:
@n3rd4i have you already tried PlatformIO + Zephyr? Does it work for you?

n3rd4i, 2019-11-20:
I tried it yesterday, I have some issue configuring it. I wanted to set-up for nrf52832 but I get:
```
Processing development (platform: https://github.com/platformio/platform-nordicnrf52.git; framework: zephyr; board: nrf52_dk)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

PackageManager: Installing framework-zephyr-hal-nordicnrf52 @ ~0.0.191028
Error: Detected unknown package 'framework-zephyr-hal-nordicnrf52'
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

This is my platformio.ini
```
[env:development]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = zephyr
board = nrf52_dk
```

ivankravets, 2019-11-20:
Could you re-run `pio update` command and try again?

n3rd4i, 2019-11-20:
I've run:
`pio update`

Then I run build again:
```
Processing development (platform: https://github.com/platformio/platform-nordicnrf52.git; framework: zephyr; board: nrf52_dk)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-gccarmnoneeabi @ ~1.80201.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
toolchain-gccarmnoneeabi @ 1.80201.190214 has been successfully installed!
PackageManager: Installing framework-zephyr-hal-nordicnrf52 @ ~0.0.191028
Error: Detected unknown package 'framework-zephyr-hal-nordicnrf52'
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

n3rd4i, 2019-11-20:
I did some more experiments changing the `platformio.ini`
Experiment #1
```
[env:development]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = zephyr
board = nrf52_dk
platform_packages =
  ; use upstream Git version
  framework-zephyr-hal-nordicnrf52 @ https://github.com/platformio/platform-nordicnrf52.git
```
Bring this:
```
Processing development (platform: https://github.com/platformio/platform-nordicnrf52.git; framework: zephyr; board: nrf52_dk)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52_dk.html
PLATFORM: Nordic nRF52 3.8.0 #ef37e8e > Nordic nRF52-DK
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (blackmagic) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES: tool-dtc 1.4.7, framework-zephyr 1.20000.191120 (2.0.0), framework-zephyr-hal-nordicnrf52 ef37e8e, tool-sreccat 1.164.0 (1.64), toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), tool-ninja 1.9.0, tool-cmake 3.15.5
Reading CMake configuration...
KeyError: 'framework-zephyr-hal-nordic':
  File "C:\users\Anonymous\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\Anonymous\.platformio\platforms\nordicnrf52@src-1ba74587582d452dca0bdd0b9f5bf6ad\builder\main.py", line 166:
    target_elf = env.BuildProgram()
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\Anonymous\.platformio\packages\framework-zephyr\scripts\platformio\platformio-build-pre.py", line 23:
    env.ProcessProgramDeps()
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\Anonymous\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 107:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\Anonymous\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 328:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 668:
    return method(*args, **kw)
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\Anonymous\.platformio\platforms\nordicnrf52@src-1ba74587582d452dca0bdd0b9f5bf6ad\builder\frameworks\zephyr.py", line 30:
    "platformio", "platformio-build.py"), exports="env")
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 668:
    return method(*args, **kw)
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\Anonymous\.platformio\packages\framework-zephyr\scripts\platformio\platformio-build.py", line 746:
    codemodel = get_cmake_code_model()
  File "C:\users\Anonymous\.platformio\packages\framework-zephyr\scripts\platformio\platformio-build.py", line 206:
    run_cmake()
  File "C:\users\Anonymous\.platformio\packages\framework-zephyr\scripts\platformio\platformio-build.py", line 179:
    "framework-zephyr-hal-" + PLATFORMS_WITH_EXTERNAL_HAL[platform_name]),
  File "c:\users\Anonymous\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 345:
    version = self.packages[name].get("version", "")
=============================================================================================== [FAILED] Took 1.27 seconds ===============================================================================================
```

Experiment #2:
[... 37 lines cut ...]
Configuration saved to 'T:/platform-nordicnrf52/examples/zephyr-ble-beacon/.pio/build/development/zephyr/.config'

-- Cache files will be written to: C:\Users\Anonymous\AppData\Local/.cache/zephyr
-- The C compiler identification is GNU 8.2.1
-- The CXX compiler identification is GNU 8.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Anonymous/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc.exe
-- Performing Test toolchain_is_ok
[... cut ...]
  Assertion failed: The toolchain is unable to build a dummy C file.  See
  CMakeError.log.
Call Stack (most recent call first):
  C:/Users/Anonymous/.platformio/packages/framework-zephyr/CMakeLists.txt:44 (assert)

============================================================================================== [FAILED] Took 15.34 seconds ==============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

valeros, 2019-11-20:
Hi @n3rd4i ! Please run `pio update` again to pull the latest changes. As for the second error that's really strange, if it doesn't work even with the latest version of `nordicnrf52` platform please open an issue [here](https://github.com/platformio/platform-nordicnrf52/issues) as this thread doesn't seem like a right place to continue our discussion. Thanks!

n3rd4i, 2019-11-21:
I was able to successfully able to debug with PlatformIO and Zephyr after doing the updates as suggested. I believe with this, we could close this ticket since its kind of resolved with this combination.

closed 2019-11-21 by n3rd4i

manoj153, 2021-01-11:
> ```json
> /usr/share/openocd/scripts/board/nordic_nrf52_dk.cfg
> ```

Hi, I am following a similar launch.json script but I am getting some error, I hope someone can point out the issue. Below are my error and launch.json
**Error :**
`Launching server: "openocd" "-c" "gdb_port 50000" "-s" "/home/manoj/Documents/Pingspace/h7_basic_bring_up" "-f" "/home/manoj/zephyrproject/zephyr/boards/arm/nucleo_h723zg/support/openocd.cfg"
`

**launch.json :**
```
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "executable": "build/zephyr/zephyr.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "STM32H723ZGT",
            "armToolchainPath": "/home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/bin/",
            "interface": "swd",
             "gdbpath": "/usr/bin/gdb-multiarch",
            // "gdbTarget": "/home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb",
            "configFiles": ["/home/manoj/zephyrproject/zephyr/boards/arm/nucleo_h723zg/support/openocd.cfg"]
        }
    ]
}
```