zephyrproject-rtos/zephyr #17650: devicetree: missing preferred instance presence macro

pabigot, 2019-07-18:
A recent change prefers ` DT_INST_0_JEDEC_SPI_NOR_LABEL` over `DT_JEDEC_SPI_NOR_0_LABEL`, however the flag symbol `DT_JEDEC_SPI_NOR_0` remains undeprecated and there is no `DT_INST_0_JEDEC_SPI_NOR` that can be used to detect presence of the device.

This should be made consistent.

pabigot, 2019-07-27:
@ioannisg I disagree with the priority low: we need a standard way to detect presence of specific instances of a device using `IS_ENABLED()` and `#ifdef`.  At present this is inconsistent and incomplete.  This should be medium and fixed before 2.0.

@ulfalizer please address this in or after merge of #17660.  Currently the new tooling generates `DT_INST_0_JEDEC_SPI_NOR_LABEL` and does not generate `DT_JEDEC_SPI_NOR_0_LABEL`, but it still generates `DT_JEDEC_SPI_NOR_0` and not `DT_INST_0_JEDEC_SPI_NOR`.

ulfalizer, 2019-07-28:
Will look later, unless @galak gets to it first.

Honestly a lot of those macros still feel like random combinations of words to me. Probably a bunch of stuff that could be cleaned up.

PR #17890 (2019-07-30): dts: Make instance defines consistent
We generated a define for each instance to convey its existance of the
form:

	#define DT_<COMPAT>_<INSTANCE> 1

However we renamed all other instance defines to be of the form
`DT_INST_<INSTANCE>_<FOO>`.  To make things consistent we now generate a
define of the form:

	#define DT_INST_<INSTANCE>_<COMPAT> 1

We also now deprecate the `DT_<COMPAT>_<INSTANCE>` form and fixup all uses
to use the new form.

Fixes: #17650
commit 4e7863dc (2019-07-30): dts: Make instance defines consistent

We generated a define for each instance to convey its existance of the
form:
	#define DT_<COMPAT>_<INSTANCE> 1

However we renamed all other instance defines to be of the form
DT_INST_<INSTANCE>_<FOO>.  To make things consistent we now generate a
define of the form:

	#define DT_INST_<INSTANCE>_<COMPAT> 1

We also now deprecate the DT_<COMPAT>_<INSTANCE> form and fixup all uses
to use the new form.

Fixes: #17650
closed 2019-07-30 by MaureenHelm