zephyrproject-rtos/zephyr #17656: k_is_in_isr() behaves inconsistently across arches

andrewboie, 2019-07-18:
In the context of fatal error handling, we would like to be able to determine if the kernel was servicing an interrupt or exception (aka interrupt context) when the fatal error occurred.

However, on some arches, if k_is_in_isr() is called from the fatal error handling code, it always returns true. I am seeing consistently on x86_64 and ARC. I see this happening sporadically on xtensa.

k_is_in_isr() is also used in kernel object APIs to detect whether we are *currently* in IRQ or exception context. In this case, behavior is also inconsistent.

What we need is:

- k_is_in_isr() to always return true if we are currently in ISR context, regardless of how we got there (some arches only report it if we are servicing an IRQ and not an exception). The only exception should be when handling a system call, this should still be considered thread context.
- Need a new API z_was_in_isr(), only to be called in exception context, to return whether we *were* in IRQ context when the exception happened, and use that in the fatal error handling logic instead.

Rationale:
- Kernel object APIs need to work properly when in IRQ context, including when handling exceptions
- Fatal error handling needs to know whether we crashed when handling an interrupt, so this can be clearly communicated to the user, and I think when this happens we should treat it as a kernel panic (i.e. never try to recover from it, instead reboot or shut down)

ioannisg, 2019-07-27:
@andrewboie shall we try to address this for 2.0?

ioannisg, 2019-10-04:
@andrewboie I am trying to solve this for ARM and I 'd like tto double-check with you.

>k_is_in_isr() to always return true if we are currently in ISR context, regardless of how we got there (some arches only report it if we are servicing an IRQ and not an exception). The only exception should be when handling a system call, this should still be considered thread context.

System Calls in ARM are executed in thread context. **Except** for the System-Call-Entry code, that is executed in SVC handler (effectively, that is an exception that elevates privileges and returns to the the system call function). So: what do you want the k_is_in_isr() to return during this SCV handler? I would recommend TRUE

>Need a new API z_was_in_isr(), only to be called in exception context, to return whether we were in IRQ context when the exception happened, and use that in the fatal error handling logic instead. This should also not return true if we were handling a system call.

Again, system calls are thread level, so z_was_in_isr will return false _except_ if we are in the SVC handler which is preparing for the system call (system-call-entry code). Is this fine?

ioannisg, 2019-10-04:
@andrewboie 

>k_is_in_isr() to always return true if we are currently in ISR context, regardless of how we got there (some arches only report it if we are servicing an IRQ and not an exception). The only exception should be when handling a system call, this should still be considered thread context.

Can a system call be invoked inside an ISR handler? If so, and an exception occurs during the call handling, is this an ISR context?

andrewboie, 2019-10-04:
> Can a system call be invoked inside an ISR handler?

System calls are only made if a privilege elevation needs to take place, so no.

I think k_is_in_isr() should return true or false depending on whether the current stack pointer falls within the bounds of K_THREAD_STACK_BUFFER(_interrupt_stack).

The check in z_fatal_error() should check if the faulting stack pointer in the ESF is within the interrupt stack buffer.

ioannisg, 2019-10-07:
>I think k_is_in_isr() should return true or false depending on whether the current stack pointer falls within the bounds of K_THREAD_STACK_BUFFER(_interrupt_stack).

>The check in z_fatal_error() should check if the faulting stack pointer in the ESF is within the interrupt stack buffer.

I think these might be both problematic for ARM @andrewboie , particularly, in case of stack overflows. I believe we need a more reliable way than checking the stack pointer values.

andrewboie, 2020-08-14:
Cross-linking as related: https://github.com/zephyrproject-rtos/zephyr/issues/21923

ioannisg, 2020-12-14:
Removing the stale, once more, I guess we'd like to address this soon (hopefully already for 2.5 and not LTS)

carlescufi, 2021-02-09:
@ioannisg, I assigned this to @dcpleung as one of the common arch collaborators.

cfriedt, 2021-07-14:
@dcpleung - any luck with this one? I wonder if `k_is_in_isr()` should return `true` when handling a syscall. If so, it might be possible to construct a test case that defines a custom `__syscall`.

dcpleung, 2021-08-25:
@nashif do we still want this in v2.7?