zephyrproject-rtos/zephyr #17657: subsis/disk/disk_access_spi_sdhc: response data eaten by idle byte consumption

pabigot, 2019-07-18:
I'm working with samples/subsys/fs which uses the SPI SDHC driver.  I've got a uSD breakout board on the Particle Xenon (nRF52840 using SPI not SPIM).

The driver has several locations, starting in `sdhc_spi_cmd_r1_raw`, where the following is present:
```
	/* Ensure there's a idle byte between commands */
	sdhc_spi_rx_u8(data);
```

In my system what this does is eat the first byte of the response, causing a `Storage init ERROR!`.  If I remove that read the sample runs and displays the files present on the card.  I can also take out the other places where this is done, and it continues to work.

I've got to assume that this works for somebody as-is and so might be necessary, so I don't want to just take it out.  I'd be happy to coordinate with somebody who understands that driver to help figure out why it's there and the conditions under which it's necessary.

mrfuchs, 2019-07-18:
Does #15790 help as well?

pabigot, 2019-07-18:
#15790, when rebased, fixes it.  Thanks.

pabigot, 2019-07-18:
Duplicate of #15444

closed 2019-07-18 by pabigot