zephyrproject-rtos/zephyr #17664: Missing 'clocks' on most nodes with 'compatible = "nxp,kinetis-usbd"'

ulfalizer, 2019-07-19:
`dts/bindings/usb/nxp,kinetis-usbd.yaml` says that any node with that binding must have a `clocks` property, but only a few nodes with `compatible = "nxp,kinetis-usbd"` have a `clocks` property.

List for reference:

```
$ git grep nxp,kinetis-usbd
boards/arm/frdm_k64f/frdm_k64f.dts:     compatible = "nxp,kinetis-usbd";
boards/arm/frdm_kl25z/frdm_kl25z.dts:   compatible = "nxp,kinetis-usbd";
boards/arm/usb_kw24d512/usb_kw24d512.dts:       compatible = "nxp,kinetis-usbd";
dts/arm/nxp/nxp_k6x.dtsi:                       compatible = "nxp,kinetis-usbd";
dts/arm/nxp/nxp_kl25z.dtsi:                     compatible = "nxp,kinetis-usbd";
dts/arm/nxp/nxp_kw2xd.dtsi:                     compatible = "nxp,kinetis-usbd";
dts/arm/nxp/nxp_rt.dtsi:                        compatible = "nxp,kinetis-usbd";
dts/arm/nxp/nxp_rt.dtsi:                        compatible = "nxp,kinetis-usbd";
dts/bindings/usb/nxp,kinetis-usbd.yaml:      constraint: "nxp,kinetis-usbd"
```

This leads to failures in https://github.com/zephyrproject-rtos/zephyr/pull/17660, which turns missing `category: required` properties into an error.

Not familiar enough with how this is used to know what the proper solution is. A quickfix would be to make `clocks` optional instead in the binding (`category: optional`).

ulfalizer, 2019-07-19:
CC @pizi-nordic @jfischer-phytec-iot @galak

pizi-nordic, 2019-07-19:
It looks that I forgot cleaning this refactoring https://github.com/zephyrproject-rtos/zephyr/pull/16957 N-th time. I will create PR which will make the clocks optional.

closed 2019-07-19 by galak