zephyrproject-rtos/zephyr #17688: Unable to Read data from SCC811

olsky, 2019-07-21:
**Describe the bug**
Configured SCC811 env sensor, init is passing.

But read always same data:

dev CCS811 at 0x200279e4
> 36865;0;10393;65287 
co2: 36865 (be16:400)
voc: 0 (be16:0)
rez: 65287 (be16:2047)
err: 40 (stat:153)
SCC811 > channel: co2
is_value_in > old: 400.00 new: 400.00
SCC811 > channel: voc
is_value_in > old: 0.00 new: 0.00
SCC811 > Co2: 400.000000ppm; VOC: 0.000000ppb
SCC811 > channel: voltage and current
SCC811 > Voltage: 1.650099V; Current: 0.000001A

**Environment (please complete the following information):**
 - OS: build under Windows
 - Toolchain: Zephyr, Nordic nrf
 - Version used: nrf 1.0.0

**Additional context**
at init, seems to be ok.
CCS811 Application firmware started!
Hardware ID: 129 (exp.: 129)
Hardware Ver: 18

Am I missing some points?

olsky, 2019-08-16:
Dear @pabigot 
could you please suggest on how to get around this issue?

thanks!

Mani-Sadhasivam, 2019-08-16:
@olsky Just curious, did the sample application in zephyr work for you? `samples/sensor/ccs811`

olsky, 2019-08-16:
@Mani-Sadhasivam yes, tried it as well, here is the log:

SPM: prepare to jump to Non-Secure image.
innblue > pwr_ctrl_init bind device "GPIO_0" pin 21
innblue > pwr_ctrl_init bind device "GPIO_0" pin 12
innblue > ccs811_init 
innblue > ccs811_init > config wakeup 
innblue > ccs811_init wakeup pin 
innblue > ccs811_init config reset
ccs881 > switch_to_app_mode > Switching to Application mode...
ccs881 > switch_to_app_mode > Check for the application firmware...
ccs881 > switch_to_app_mode > Set the device to application mode...
ccs881 > switch_to_app_mode > read reg status...
ccs881 > switch_to_app_mode > Check for application mode...
CCS811 Application firmware started!Hardware ID: 129 (exp.: 129)
Hardware Ver: 18
***** Booting Zephyr OS build v1.14.99-ncs2-495-ga8933a2ead01 *****
device is 0x20020fbc, name is CCS811
Co2: 0.000000ppm; VOC: 0.000000ppb
Voltage: 0.000000V; Current: 0.000000A

> 36865;0;10393;65287 
co2: 36865 (be16:400)
voc: 0 (be16:0)
rez: 65287 (be16:2047)
err: 40 (stat:153)
Co2: 400.000000ppm; VOC: 0.000000ppb
Voltage: 1.650099V; Current: 0.000001A

> 36865;0;10393;65287 
co2: 36865 (be16:400)
voc: 0 (be16:0)
rez: 65287 (be16:2047)
err: 40 (stat:153)
Co2: 400.000000ppm; VOC: 0.000000ppb
Voltage: 1.650099V; Current: 0.000001A

pabigot, 2019-08-16:
On power up you'll read 0 eCO2 and 0 eTVOC until the `DATA_READY` bit of the `STATUS` register becomes set.  (On CCS811 firmware versions less than 2.0 this bit is never set, but the behavior is the same.)  This normally happens after two or three readings.

After that there's a 20 minute conditioning period during which the readings will be constant at the minimum reportable value 400 cCO2 and 0 eTVOC.

After that you should start seeing some variation.

Let it run.

olsky, 2019-08-16:
thanks @pabigot , I will leave it on...

olsky, 2019-08-18:
just a short update: after 8hours and 3 boards running, still the same 400/0 values... running on nRF9160

pabigot, 2019-08-18:
The Zephyr driver for CCS811 is very simplistic, but it does read the error and baseline (resistance) registers.  Looking again at the output your status register value is 0x99 and the `ERROR_ID` is perhaps 0x18, which decodes to a heater fault and the sensor being at the maximum resistance.  Also it appears that it's measuring the minimum representable current, and maximum representable voltage.

I suspect a hardware error, or misconfiguration of the pins.  Without a device to play with and its schematics there isn't much more I can say.

olsky, 2019-10-07:
I got the adafruit board, I will connect and analyze both in parallel with the logic analyzer.
Later will post the results.

// for now the adafruit works with arduino arm board (even is there is constant error on return)

closed 2020-09-14 by pabigot