zephyrproject-rtos/zephyr #17690: samples/subsys/shell/fs does not work?

pabigot, 2019-07-21:
As far as I can tell, this sample doesn't do anything.  If it can be made to work, the documentation needs to be greatly improved.

Some notes before I gave up:

* For native posix you probably need the libfuse development package installed.
* `CONFIG_FS_NFFS_FLASH_DEV_NAME` should be pushed down to the board-specific conf file, since the one in `prj.conf` is wrong for Nordic devices.

Both native posix and a Nordic board version show the same behavior:
```
uart:~$ fs mount nffs /                                                         
Error mounting nffs fs.Error Code [-22]                                         
[00:00:05.170,013] <err> fs: invalid mount point!!                              
uart:~$ fs mount nffs /m                                                        
Successfully mounted fs:/m                                                      
uart:~$ fs pwd                                                                  
/                                                                               
uart:~$ fs ls                                                                   
Unable to open / (err -22)                                                      
[00:00:13.041,809] <err> fs: invalid file name!!                                
uart:~$ fs cd /m                                                                
/m doesn't exist                                                                
[00:00:19.537,902] <err> fs: failed get file or dir stat (-22)                  
uart:~$ 
```
That https://github.com/zephyrproject-rtos/nffs bears no commits in common with https://github.com/apache/mynewt-nffs, and neither retained the history of nor track the [real? source](https://github.com/apache/mynewt-core/tree/master/fs/nffs) makes this a little more frustrating.

aescolar, 2019-07-21:
@vanwinkeljan can you help here?

About 
> For native posix you probably need the libfuse development package installed.

I hope it is only needed if you set [CONFIG_FUSE_FS_ACCESS](https://docs.zephyrproject.org/latest/boards/posix/native_posix/doc/index.html#host-based-flash-access)

pabigot, 2019-07-21:
> I hope it is only needed if you set CONFIG_FUSE_FS_ACCESS

Presumably.  But that is how the sample is configured out of the box, so the requirement should be documented.

vanwinkeljan, 2019-07-21:
@pabigot something I noticed while implementing the fuse implementation is that if you want to perform any action on the mount point you need a trailing slash.

Could you try `fs ls /m/` or `fs cd /m/`, note that this actually a zephyr FS API limitation and has nothing to do with the FS shell. In the fuse part I worked myself around this limitation

pabigot, 2019-07-21:
Yes, adding a trailing slash to the mount point helps.

I wonder if that's related to the fs infrastructure including the mount point when it provides the path to the mount-specific implementation, so every file system implementation has to strip it off themselves in every API call.

vanwinkeljan, 2019-07-21:
@aescolar 
> I hope it is only needed if you set CONFIG_FUSE_FS_ACCESS

Just double checked libfuse is only needed if you select  `CONFIG_FUSE_FS_ACCESS`.
As @pabigot  mentioned it just not stated in the sample README file.

vanwinkeljan, 2019-07-21:
@pabigot if I'm not mistaken this is the root cause (`fs_get_mnt_point `)

https://github.com/zephyrproject-rtos/zephyr/blob/8f1780f6dee29dc2b2db01744e328a1a4a20d0b8/subsys/fs/fs.c#L50-L56

pabigot, 2019-07-21:
Actually, it's not.  That does the right thing and identifies the correct mount point and passes both that structure and the full path "/m" to the nffs module.

The flaw is in the nffs driver which, because it has to strip the no-longer-necessary mount point off itself, does so with an incorrect assumption that the mount point is a proper prefix.  So, when it's given "/m" as a path it strips off "/m" and tries to look for a directory at "", which is not an absolute path within the filesystem and can't be found.  It works for "/m/", and for "/m/whatever", because the strip operation produces "/" and "/whatever" respectively.

This is annoying, so I'll see if I can provide a helper function that does it right.

pabigot, 2019-07-21:
> This is annoying, so I'll see if I can provide a helper function that does it right.

See #17695. I've checked this on nffs; I can't get shell/fs to work with fat_fs and the fat_fs module is doing something weird with mount points (they're somehow one character long) so that'll have to be a different issue.

vanwinkeljan, 2019-07-28:
^^^ Sorry for the wrong reference, copied the wrong issue number

pabigot, 2019-07-28:
Enough of the issues have been addressed that I'm calling this done.

closed 2019-07-28 by pabigot