zephyrproject-rtos/zephyr #17692: Proper way for joining a multicast group (NRF52840/OpenThread)

GeeseHoward, 2019-07-21:
*SCENARIO*
I'm developing the application on NRF52840 with OpenThread. MLD is also enabled in my application.

I am implementing a state machine which some of the states listen for data on different multicast addresses (the application uses around 4 addresses in total). It looks more or less like this:

```
	n_state cur_state= STATE_1;
	n_state next_state;
	while(true){
		if(cur_state==STATE_1){
			next_state = state_1_handler();
		}
		else if(cur_state==STATE_2){
			next_state = state_2_handler();
		}
		else if(cur_state==STATE_3){
			next_state = state_3_handler();
		}
		else{
			next_state = fail_handler();
		}
		cur_state = next_state;
	}
```
Currently, I'm joining/leaving mcast addresses as follows:

```
bool join_multicast_group(struct sockaddr* mcast_addr)
{
	struct sockaddr_in6* mcast_addr6 = (struct sockaddr_in6*)mcast_addr;
	struct net_if_mcast_addr *mcast;
	struct net_if *iface;

	iface = net_if_get_default();
	if (!iface) {
		printf("Could not get the default interface\n");
		return false;
	}

	mcast = net_if_ipv6_maddr_add(iface, &mcast_addr6->sin6_addr);
	if (!mcast) {
		printf("Could not add multicast address to interface\n");
		return false;
	}
	return true;
}

bool leave_multicast_group(struct sockaddr* mcast_addr)
{
	struct sockaddr_in6* mcast_addr6 = (struct sockaddr_in6*)mcast_addr;
	struct net_if *iface;
	bool ret;

	iface = net_if_get_default();
	if (!iface) {
		printf("Could not get the default interface\n");
		return false;
	}

	ret = net_if_ipv6_maddr_rm(iface, &mcast_addr6->sin6_addr);
	if (!ret) {
		printf("Could not leave multicast group\n");
		return false;
	}

	return ret;
}

```
When I join group like that and type 'ot ipmaddr' in the shell, it shows me the newly added IP address. However, after executing my 'leave_multicast_group' function that IP is still present when I enter ot ipmaddr in the shell.

Besides that, my application stopped behaving as expected. In my application. In the normal flow, the application transitions from STATE_1 to STATE_2 to STATE_3. STATE_3 wouldn't receive any multicast data if I executed the application normally and shell threw the error Failed to reserve net pkt". However, if I forced it such that STATE_3 was executed before 1 and 2, it received data as expected.

I'm guessing this is because I'm not joining/leaving the multicast groups properly. I started looking around and came across this suite of tests:

https://github.com/zephyrproject-rtos/zephyr/blob/21358baa72cea9c23be57444eb91444774842f97/tests/net/mld/src/main.c

Groups are being joined using the method 'net_ipv6_mld_join'. However, when I tried that I got the 'Not enough Core error' (-12).

Kindly let me know the proper way of joining and leaving groups. Thanks

ioannisg, 2019-07-22:
@rlubos could you address this question?

rlubos, 2019-07-22:
Hi @GeeseHoward,

The multicast join/leave functions you provided seem fine. There are a few issues though that may cause you trouble.

> When I join group like that and type 'ot ipmaddr' in the shell, it shows me the newly added IP address. However, after executing my 'leave_multicast_group' function that IP is still present when I enter ot ipmaddr in the shell.

That's because of limitation of the current OT port, which does not handle `NET_EVENT_IPV6_CMD_MADDR_DEL` event in its network event handler [here](https://github.com/zephyrproject-rtos/zephyr/blob/efae1a5e10512e28f174ad908ac40b4b5eedc4b5/subsys/net/l2/openthread/openthread.c#L67). W/O this event handler implemented, OpenThread port is only capable of registering new addresses. 

> However, if I forced it such that STATE_3 was executed before 1 and 2 (meaning no multicast groups are joined or left before STATE_3 executes), it received data as expected.

Note, that the maximum number of multicast addresses that can be registered is configurable in both, Zephyr and OpenThread. In Zephyr we have `CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT`, while in OpenThread it's `OPENTHREAD_CONFIG_MAX_EXT_MULTICAST_IP_ADDRS`. The difference is, that in OT, the config specifies the maximum number of externally added multicast addresses (2 by default), while Zephyrs config covers all multicast addresses registered. OpenThread registers quite a few multicast addresses on its own (in my case it was 7 out of 8 slots available), so you likely hit the limit here.

> Groups are being joined using the method 'net_ipv6_mld_join'. However, when I tried that I got the 'Not enough Core error' (-12). What does that mean?

MLD is disabled with OpenThread by default as OpenThrad does not use MLD internally, therefore you should not use MLD functions but register multicast addresses directly (`net_if_ipv6_maddr_add`) just as you did.

GeeseHoward, 2019-07-22:
Thank you for the quick response, @rlubos .

> The difference is, that in OT, the config specifies the maximum number of externally added multicast addresses (2 by default)

This might be a silly question, but does the OPENTHREAD_CONFIG_MAX_EXT_MULTICAST_IP_ADDRS specify the maximum number of addresses that a particular application can add, or the maximum number allowed in the network? I have two applications (say A and B). A adds 2 multicast addresses, and B adds 1 multicast address. Everything works fine when A adds its first address and B adds its address, but when A adds the second address it doesn't receive the data. So does the configuration represent the maximum number of addresses that all the nodes in the network can add?

rlubos, 2019-07-22:
`OPENTHREAD_CONFIG_MAX_EXT_MULTICAST_IP_ADDRS` is per-application config, AFAIK Thread does not specify a network-wide limit for the number of multicast addresses in use in the network. It's simply related to the number of buffers available.

Could it be that you hit the Zephyr limit in your case? As I wrote earlier, in my case (Leader device) the OpenThread stack itself occupied 7 out of 8 multicast address slots available in Zephyr. Some shell commands could be useful to see what's happening in your applicaction. You could use:
```
uart:~$ net iface

Interface 0x200147e0 (OpenThread) [1]
=====================================
Link addr : 05:AD:A1:3F:0C:DD:80:9C
MTU       : 1280
IPv6 unicast addresses (max 6):
        fe80::5c16:157d:391:c79e autoconf preferred infinite
        fdde:ad00:beef:0:6ca4:4628:111f:3549 autoconf preferred infinite meshlocal
        fdde:ad00:beef::1 manual preferred infinite meshlocal
IPv6 multicast addresses (max 8):
        ff02::1
        ff03::1
        ff03::fc
        ff32:40:fdde:ad00:beef::1
        ff33:40:fdde:ad00:beef::1
        ff02::2
        ff03::2
IPv6 prefixes (max 2):
        <none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 36642
IPv6 retransmit timer    : 0
```
command to see what's happening on the Zephyr side, and:
```
uart:~$ ot ipmaddr
ff33:40:fdde:ad00:beef:0:0:1
ff32:40:fdde:ad00:beef:0:0:1
ff02:0:0:0:0:0:0:2
ff03:0:0:0:0:0:0:2
ff02:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:fc
Done
```
to see what's registered in the OpenThread. Multicast address lists should be 1:1 copy, there could be an issue though when you remove a multicast address (as mentioned earlier).

GeeseHoward, 2019-07-22:
> Could it be that you hit the Zephyr limit in your case?
This does not seem to be the case. I have set the Zephyr limit to 12. 
 
I also just tried setting OPENTHREAD_CONFIG_MAX_EXT_MULTICAST_IP_ADDRS to 4, but no luck. Still getting the same 'Failed to reserve net pkt' error every time application B sends a mutlicast packet. Shouldn't I not get that error at all if I'd hit the limit, and shouldn't _net_if_ipv6_maddr_add_ return an error in that case?

rlubos, 2019-07-22:
`net_if_ipv6_maddr_add` will fail if you fail to register a multicast address in Zephyr but not in the OpenThread, as we call OpenThread API from the event handler, which has no return value. 

Why is it failing exactly with 'Failed to reserve net pkt' it's hard to tell really w/o a debugger. One could advise increasing `CONFIG_NET_PKT_RX_COUNT`/`CONFIG_NET_BUF_RX_COUNT` configs, but if you say the issue is specific to multicast packets I don't think that's the case.

GeeseHoward, 2019-07-22:
I managed to identify the problem. STATE_1 initialized a multicast socket which I neglected to close. Doing so made everything run as expected.
Thank you @rlubos and @ioannisg for taking the time to help me out.

jukkar, 2019-07-23:
Getting network interface like in your sample works if you have only one network interface, but if you later have a system where you would have multiple network interfaces, then this could be written like this instead:
```
	iface = nnet_if_get_first_by_type(&NET_L2_GET_NAME(OPENTHREAD));
	if (!iface) {
		printf("Could not get the default interface\n");
		return false;
	}
```

ioannisg, 2019-07-30:
@GeeseHoward please, re-open, if needed, to continue discussion

closed 2019-07-30 by ioannisg

AllemanMa, 2021-09-29:
@rlubos, has there been any progress or updates on this? I would like to use multicast groups with Zephyr and Openthread. Adding an ipmaddr seems to work fine however removing one is still a problem. It seems indeed that `ipv6_addr_event_handler` does still not handle `NET_EVENT_IPV6_CMD_MADDR_DEL`. Thanks